zigpy/zha-device-handlers #2577: [Device Support Request] TS0601 _TZE200_upagmta9

MStapelfeldt, 2023-09-08:
### Problem description

I got a Zigbee device (https://de.aliexpress.com/item/1005004666251263.html) which is regonized, but without Values. I tried to add signature to [ts0601_sensor.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py) but still no luck to get it working.

### Solution description

Help with integration

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0408",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_upagmta9",
  "model": "TS0601",
  "class": "ts0601_sensor.TuyaSoilSensor"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.08.3",
    "host_os": "Home Assistant OS 10.5",
[... 336 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

javicalle, 2023-09-09:
Attach the signature without the local quirk aoolied.
I would try to add it in this quirk:
https://github.com/zigpy/zha-device-handlers/blob/7745c838f98ac126d64ae4edad62556763b13d2e/zhaquirks/tuya/ts0601_sensor.py#L126

MattWestb, 2023-09-12:
@dawrut You shall open one issue at https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html as you is running Z2M and not ZHA.

AlexNoop, 2023-09-16:
hey ;)

Same issue for me ! with ZHA and signature in ts0601_sensor.py

Stevelande, 2023-10-04:
https://de.aliexpress.com/item/1005005627336725.html?spm=a2g0o.order_list.order_list_main.5.17da5c5fF6CUMQ&gatewayAdapt=glo2deu
Same issue here.
I tried.
`("_TZE200_upagmta9", "TS0601"),`
in
`class TuyaTempHumiditySensor_Square(CustomDevice):
    """Custom device representing tuya temp and humidity sensor with e-ink screen."""`

[image]

01fr, 2023-12-20:
Anyone managed to get the TS0601 _TZE204_upagmta9 working in ZHA?

this is the small tuya zigbee temperature and humidity sensor with screen.

ewoudwerkman, 2023-12-20:
For zigbee2mqtt they seem to have solved this issue. See https://github.com/Koenkk/zigbee2mqtt/issues/19137 
According to this comment https://github.com/Koenkk/zigbee2mqtt/issues/19137#issuecomment-1810843051 the Tuya device requires an extra command send from the controller after the interview:
-     if you issue a cluster 0xef00 command 0x03 the device send all values and goes to sleep.
-     send a temperature F° Command from frontend UI, which is confirmed on screen sensor switching to °F, then sending the cluster 0xef00 command 0x03 the sensor goes to sleep

Not sure how to implement this in zha.

01fr, 2023-12-23:
We need the help of the experts. Can someone help? :)

MStapelfeldt, 2023-12-23:
I wasn't able to get it working and the extra command after interview requires more knowledge then I have. (I'm just a regular C++/PHP/VB/C#/Ruby developer, no phyton 😅)
I brought some other devices without screen which was more easy and will look to buy known supported once if I need some with display.

RoboKaren, 2023-12-31:
Have the same little temp/humidity sensor. Can pair it but then it freezes up and no values come in. Sucks as it's inexpensive, I hope the next release of ZHA can get it working, it looks like there's a ton of work on TUYA devices over on the "other side" that ZHA can pull from.

JordanAco1, 2024-01-04:
It would be great to have support for this device given the fact this it has native support for Zigbee.

Adding in to ZHA is not an issue, no entities showing is the problem

[image]

dualmacops, 2024-01-25:
@javicalle The device signature:

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_upagmta9",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}`

TheJulianJES, 2024-01-30:
> For zigbee2mqtt they seem to have solved this issue. See [Koenkk/zigbee2mqtt#19137](https://github.com/Koenkk/zigbee2mqtt/issues/19137) According to this comment [Koenkk/zigbee2mqtt#19137 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/19137#issuecomment-1810843051) the Tuya device requires an extra command send from the controller after the interview:
> 
> * if you issue a cluster 0xef00 command 0x03 the device send all values and goes to sleep.
> * send a temperature F° Command from frontend UI, which is confirmed on screen sensor switching to °F, then sending the cluster 0xef00 command 0x03 the sensor goes to sleep
> 
> Not sure how to implement this in zha.

Haven't tested this at all, but linking my comment from the other thread (for the same exact device): https://github.com/zigpy/zha-device-handlers/issues/2694#issuecomment-1916001737

OZub, 2024-02-23:
same issue here.
is there any hope for a fix?
[image]

MStapelfeldt, 2024-02-23:
@dmulcahey should be one of the people responsible. But there is no one assigned to this issue. So I don't have much hope they try to change it...

01fr, 2024-04-21:
Given that there is solution to integrate these thermometers in HA, has anyone found any use for them? 
Is it possible to convert them from Zigbeee to bluetooth?

MacDada, 2024-04-21:
> has anyone found any use for them?

well… they are thermometers… they report temperature… what other use can they be?

01fr, 2024-07-28:
Just wondering if anyone managed to get these devices working on ZHA?

PR #3817 (2025-01-31): Fix Tuya temp/humidity sensor quirks (`_TZE200_upagmta9`, ...)
## Proposed change

Add Tuya `tze200 cirvgep4` and combine quirks to match z2m.

See: https://github.com/Koenkk/zigbee-herdsman-converters/blob/09cfce5153d3931c7cc009aa3ae3107d9b394f5c/src/devices/tuya.ts#L1073

## Additional information

Closes: #2577
Closes: #2854 
Closes: #2304 
Closes: #3260 
Closes: #2725 
Closes: #2929 

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-01-31 by TheJulianJES