zigpy/zha-device-handlers #2582: [Device Support Request] Aqara Water Leak Sensor T1 - lumi.flood.agl02

m67hoff, 2023-09-13:
### Problem description

Aqara Water Leak Sensor T1 SJCGQ12LM lumi.flood.agl02 
shows as "mains powered"  and does not show a battery  entity

It's supported by ZHA according to https://zigbee.blakadder.com/Aqara_SJCGQ12LM.html however it's not recognised by any quirk. 

[image]

### Solution description

I  tried a custom quirk (see below) - however I'm a new to HA and was not successful (this is my first week with HA and my first sensor. We had water in our cellar and the sensor was recommended to me. Seems I got an unsupported version. I have not expected to write/modify code already in my first week) 

Its gets applied:
```python
2023-09-13 16:57:16.502 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for LUMI lumi.flood.agl02 (54:ef:44:10:00:90:96:f3)
2023-09-13 16:57:16.502 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'wleak_agl02.WaterT1'>
2023-09-13 16:57:16.503 DEBUG (MainThread) [zigpy.quirks.registry] Found custom device replacement for 54:ef:44:10:00:90:96:f3: <class 'wleak_agl02.WaterT1'>
```  

However it does not change the nodeDescriptor (`*is_mains_powered=True`) and my guess is that this is the reason for the "mains"-power and with mains power ZHA will not show batterie.

### Screenshots/Video

<details><summary>zha toolkit scan</summary>

```json

{
  "ieee": "54:ef:44:10:00:90:96:f3",
  "nwk": "0x8422",
  "model": "lumi.flood.agl02",
  "manufacturer": "LUMI",
  "manufacturer_id": "0x4447",
  "endpoints": [
    {
      "id": 1,
      "device_type": "0x0402",
      "profile": "0x0104",
      "in_clusters": {
        "0x0000": {
          "cluster_id": "0x0000",
          "title": "Basic",
          "name": "basic",
          "attributes": {},
          "commands_received": {
            "0x00": {
              "command_id": "0x00",
              "command_name": "reset_fact_default",
              "command_arguments": "<class 'zigpy.zcl.foundation.reset_fact_default'>"
            }
          },
          "commands_generated": {}
        },
        "0x0001": {
          "cluster_id": "0x0001",
          "title": "Power Configuration",
          "name": "power",
          "attributes": {},
          "commands_received": {},
          "commands_generated": {}
        },
        "0x0003": {
          "cluster_id": "0x0003",
          "title": "Identify",
          "name": "identify",
          "attributes": {},
          "commands_received": {},
          "commands_generated": {}
        },
[... 858 lines cut ...]
                    IasZone.cluster_id                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    BasicCluster,
                    XiaomiPowerConfiguration,
                    Identify.cluster_id,
                    IasZone.cluster_id
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }

```

</details>

TheJulianJES, 2023-09-16:
Does the sensor functionality (for the water leak sensor entity) work correctly though?

Also, where did you find this sensor? I'm in Germany myself, but Amazon only seems to sell the old version.

TheJulianJES, 2023-09-16:
Please test the custom quirk from https://github.com/zigpy/zha-device-handlers/pull/2590 to see if that fixes the issue.

m67hoff, 2023-09-17:
Hi - I have got this sensor from https://www.alza.de/aqara-water-leak-sensor-t1-d7852553.htm  They had the Skyconnect  relative cheap and in stock what I ordered together.   (sorry for the additional work) 

I anyway will send back this "lumi.flood.agl02"  and order the better supported "lumi.sensor_wleak.aq1" on amazon.
So thanks for your help.  (I have not found any example to change the nodedescriptor to try this by my own. Also looks more complicated than I thought )

 I tested the  https://github.com/zigpy/zha-device-handlers/pull/2590 and it got applied and ZHA now show battery  entity. 
[image]
However it does not show a percentage.  So probably something in the  PowerConfiguration cluster is also different.
Attached a new zha_toolkit scan with your new quick applied
[lumi.flood.agl02_8.json.zip](https://github.com/zigpy/zha-device-handlers/files/12642392/lumi.flood.agl02_8.json.zip)

TheJulianJES, 2023-09-17:
Generally, the T1 sensor is more reliable than the older models.
So, thanks for the link! I'll probably buy one soon and make sure it's supported correctly.

Can you check in a few hours if the battery percentage shows up?
If there's a button on the device that you can press once, try to do a short click.

If battery doesn't show up in the next 24 hours, please try to re-pair the device.

m67hoff, 2023-09-18:
No chance - not even with button pressing, re-pairing or wetting. Battery remains "unknown".

Send me your address and I'll send you the part (free of charge, you can keep it).

TheJulianJES, 2023-09-18:
Ah, that’s unfortunate it doesn’t work yet.

It would be awesome if you could send me the sensor. Thanks for the offer! I’d pay the shipping if you want.
I didn’t find a private way to contact you, so you can just send me an email: julian.tjj@proton.me and I‘ll send you the address.

TheJulianJES, 2023-09-22:
Received the package yesterday. Thanks again (and also for the extra 😄)!

Apparently this seems to be a slightly newer revision of the T1 water leak sensor, as there's there hidden "Aqara Opple (E1)" cluster where it sends battery reports:

```python
# button click "on the top" causes it to report its model:
2023-09-22 02:42:18.867 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x91E3), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=36, profile_id=260, cluster_id=0, data=Serialized[b'\x18\r\n\x05\x00B\x10lumi.flood.agl02'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-32)
2023-09-22 02:42:18.869 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0x0000] Received ZCL frame: b'\x18\r\n\x05\x00B\x10lumi.flood.agl02'
2023-09-22 02:42:18.870 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=13, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-09-22 02:42:18.870 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0x0000] Decoded ZCL frame: BasicCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.flood.agl02'))])
2023-09-22 02:42:18.871 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0x0000] Received command 0x0A (TSN 13): Report_Attributes(attribute_reports=[Attribute(attrid=0x0005, value=TypeValue(type=CharacterString, value='lumi.flood.agl02'))])
2023-09-22 02:42:18.871 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0x0000] Attribute report received: model='lumi.flood.agl02'

# and also sends the Xiaomi attributes (including battery voltage):
2023-09-22 02:42:18.875 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x91E3), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=37, profile_id=260, cluster_id=64704, data=Serialized[b'\x1c_\x11\x0e\n\xf7\x00A$\x01!\xef\x0b\x03(\x19\x04!\xa8\x13\x05!Y\x00\x06$\x02\x00\x00\x00\x00\x08!\x1c\x01\n!\x00\x00\x0c \x01d\x10\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-32)
2023-09-22 02:42:18.876 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0xfcc0] Received ZCL frame: b'\x1c_\x11\x0e\n\xf7\x00A$\x01!\xef\x0b\x03(\x19\x04!\xa8\x13\x05!Y\x00\x06$\x02\x00\x00\x00\x00\x08!\x1c\x01\n!\x00\x00\x0c \x01d\x10\x00'
2023-09-22 02:42:18.876 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0xfcc0] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=True, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), manufacturer=4447, tsn=14, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-09-22 02:42:18.876 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0xfcc0] Decoded ZCL frame: XiaomiAqaraE1Cluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x00F7, value=TypeValue(type=LVBytes, value=b'\x01!\xef\x0b\x03(\x19\x04!\xa8\x13\x05!Y\x00\x06$\x02\x00\x00\x00\x00\x08!\x1c\x01\n!\x00\x00\x0c \x01d\x10\x00'))])
2023-09-22 02:42:18.876 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0xfcc0] Received command 0x0A (TSN 14): Report_Attributes(attribute_reports=[Attribute(attrid=0x00F7, value=TypeValue(type=LVBytes, value=b'\x01!\xef\x0b\x03(\x19\x04!\xa8\x13\x05!Y\x00\x06$\x02\x00\x00\x00\x00\x08!\x1c\x01\n!\x00\x00\x0c \x01d\x10\x00'))])
2023-09-22 02:42:18.877 DEBUG (MainThread) [zigpy.zcl] [0x91E3:1:0xfcc0] Attribute report received: 0x00F7=b'\x01!\xef\x0b\x03(\x19\x04!\xa8\x13\x05!Y\x00\x06$\x02\x00\x00\x00\x00\x08!\x1c\x01\n!\x00\x00\x0c \x01d\x10\x00'
2023-09-22 02:42:18.877 DEBUG (MainThread) [zhaquirks.xiaomi] 54:ef:44:10:00:90:96:f3 - Xiaomi attribute report. attribute_id: [247] value: [{'battery_voltage_mV': 3055, 'temperature': 25, 'X-attrib-4': 5032, 'X-attrib-5': 89, 'X-attrib-6': 2, '0xff01-8': 284, 'path': 0, '0xff01-12': 1, '0xff01-100': <Bool.false: 0>}]
2023-09-22 02:42:18.877 DEBUG (MainThread) [zhaquirks.xiaomi] Voltage mV: [Min]:2820 < [RAW]:3055 < [Max]:3100, Battery Percent: 84.0
```

I've looked at the implementation in Z2M and it also doesn't seem to catch reports from that cluster on this sensor. So I guess this should also be fixed in Z2M.
I'll update the PR for the quirk to catch "Xiaomi attribute reports" from both the Basic and custom XiaomiAqaraOpple cluster.
Will also try to get names for the custom attributes.

closed 2023-09-24 by TheJulianJES

vistvanv, 2023-09-25:
hey, I got 2  Aqara T1 door and window sensor( new version that had been launched in the past month, and it seems it is not supported either. neither the status or the battery. Any thoughts on how to fix this?

Device is initialized <Device model='lumi.magnet.agl02' manuf='LUMI' nwk=0x91F1 ieee=54:ef:44:10:00:92:52:06 is_initialized=True>
Checking quirks for LUMI lumi.magnet.agl02 (54:ef:44:10:00:92:52:06)
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace2'>
Fail because endpoint list mismatch: {1, 2, 242} {1}
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1}
Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because device_type mismatch on at least one endpoint
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1}

TekFan, 2023-11-16:
@TheJulianJES 
I purchased 4 Water leak sensor T1 about 10 days ago from Domadoo.
Unfortunately, they seem to present the same flaw as before, they stop reporting after pairing and the battery level stays "unknown".
Pressing the button, does update the status, however battery still stays "unknown" forever, until it timeout again and everything goes back to "unknown and the "last seen" timestamp never update again unless there is water detected or the button is pressed.
Fortunately, the iass zone is updated when water is detected so my alert automation at least works.
But in this situation, I would never know if the battery need to be changed either by it's status or the fact that the device becomes uavailable as it would always been unavailable...
This is why I need to report this.

The corresponding quirk is loaded.
I'm on 2023.11.2
The device reports firmware 28 by reading the cluster.

So I wonder if Aqara changed something...again.
I'm open to try to provide any info you would need.

TheJulianJES, 2023-11-16:
@TekFan Yes, newer Aqara devices have a minimum firmware version for Conbee and TI coordinators, as they need the Xiaomi manufacturer code during pairing.

After updating the firmware, you'll also need to re-pair the device.

TekFan, 2023-11-16:
@TheJulianJES
Ok, I understand, I'll try that tomorrow.
Just to be sure, when you say repair the device, you only mean this water leak sensor, not all my 68 devices, I hope ?

Thanks again for your incredible support !

TekFan, 2023-11-17:
@TheJulianJES Ok, you were right, of course...
Combee 2 update done, re-pair done, battery is finally reporting, yeah !
10 hours later, still alive and kicking, so I guess that was it.

That said, lot of warnings: "No response to 'Command.aps_data_indication' command with seq id '0xNN'" in the log (143 so far). I had a few before, but now it's much more.
Does not seem to affect operation, although I seem to experience som slow responses from motion detectors.

Thank you for your suggestion anyway, you pinpointed the exact culprit.