zigpy/zha-device-handlers #2583: [Device Support Request] ZLinky_TIC - add more sensors

Xalio, 2023-09-14:
### Problem description

Current device support doesn't show the entire list of attributes of the device in the sensor.
As per example in my case I need to get the electricity metering for three phases, not just phase 1.

When I try manually, the device gets the values, see screenshot example below:
[image]
 

### Solution description

I would need to get the information related to phase b & c as sensors so I can use them for graphs.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=200, server_mask=11264, maximum_outgoing_transfer_size=200, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0053",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0702",
        "0x0b01",
        "0x0b04",
        "0xff66"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LiXee",
  "model": "ZLinky_TIC",
  "class": "zhaquirks.lixee.zlinky.ZLinkyTICFWV12"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Side question, how can we build custom quirk from existing device signature? I can have a look at that if needed.

Thanks!

TheJulianJES, 2023-09-17:
This likely needs to be implemented as a sensor in [ZHA / HA Core](https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/sensor.py).

Hedda, 2023-12-20:
> Current device support doesn't show the entire list of attributes of the device in the sensor.

> This likely needs to be implemented as a sensor in [ZHA / HA Core](https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/sensor.py).

FYI, there is a related upstream request to the Zlinky TIC product manufacturer/developer here -> https://github.com/fairecasoimeme/Zlinky_TIC/issues/18

There Zlinky TIC owners are asking for and discussing better integration in Home Assistant's built-in ZHA integration component.

Xalio, 2024-07-30:
Not resolved, still a thing for those who want to use that device.

superlevure, 2024-09-17:
I have the same request to add the following attribute as an entity:

- Cluster: `ZLINKY_MANUFACTURER_CLUSTER_ID` (`0xff66`)
- Attribute: `std_current_supplier_price_description` (`0x0200`)

It's really important for those who subscribed to a TEMPO contract

closed 2025-08-17 by github-actions[bot]