zigpy/zha-device-handlers #2585: Unknown cluster Warning for TS011F plug

erkr, 2023-09-16:
### Bug description

My device TS011F by TZ3000_gvn91tmx is handled by quirk zhaquirks.tuya.ts011f_plug.Plug
That works but maybe no full support. 
I noticed an ‘ Unknown cluster ‘ warning in my log

### Steps to reproduce

Happens during startup

### Expected behavior

-

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python

Logger: zigpy.zcl
Source: runner.py:186 
First occurred: 00:51:37 (10 occurrences) 
Last logged: 00:51:38

[0xCCAF:1:0xe001] Unknown cluster command 225 b'\x02o\xaa\x97,\x8b\x00\x00\x00\x8b\x00\x00\x00\x00'

```

</details>

MattWestb, 2023-09-16:
Pleas post device signature and diagnostic so can see if your device is have the right cluster settings or not.

erkr, 2023-09-18:
> Pleas post device signature and diagnostic so can see if your device is have the right cluster settings or not.

@MattWestb I updated the initial post with the diagnostics and signature. Should have done that immediately. 
Best Eric

closed 2024-03-17 by erkr