zigpy/zha-device-handlers #2587: [BUG] Tuya TS0201 Temp / Humidity Sensors added but stop reporting data and go unresponsive shortly after

cvocvo, 2023-09-16:
### Bug description

I recently bought two Tuya temperature / humidity zigbee sensors and both of them are detected but and initially report data but then become unresponsive, stop reporting data, and are marked as unavailable in Home Assistant moments after adding.

The two sensors:
- TS0201 _TZ3000_dowj6gyi https://www.aliexpress.us/item/3256805267838648.html?spm=a2g0o.order_list.order_list_main.6.5aa81802De4DHg&gatewayAdapt=glo2usa
- TS0201 _TZ3000_xr3htd96 https://www.aliexpress.us/item/3256805415172258.html

### Steps to reproduce

1. Reset the devices and put them into pairing mode.
2. Add them to home assistant
3. Go to the device page and you see they become unknown shortly after and no further data is reported.

### Expected behavior

1. Reset the devices and put them into pairing mode.
2. Add them to home assistant
3. Go to the device page and you see them stay online and normally report humidity / temperature data.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "dyson_cloud": {
      "version": "0.15.0",
      "requirements": [
        "libdyson==0.8.7"
      ]
    },
[... 773 lines cut ...]
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

Not sure why but my log file is 200MB and I can't easily upload it.

TheJulianJES, 2023-09-17:
Hmm, maybe they need to have the "Tuya spell" cast on them. So a quirk changing the device to an `EnchantedDevice` and using an "Enchantable Tuya" cluster.

Can you also attach ZHA integration diagnostics? Also, what coordinator are you using?

cvocvo, 2023-09-23:
> Hmm, maybe they need to have the "Tuya spell" cast on them. So a quirk changing the device to an EnchantedDevice and using an "Enchantable Tuya" cluster.
How can I configure that?

The coordinator that I have is the Zooz 800 series ZST39 (https://www.getzooz.com/zooz-zst39-z-wave-long-range-usb-stick/)

Here are the diagnostics logs:

<details><summary>Zigbee Integration Diagnostic Logs</summary>
```python
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "dyson_cloud": {
      "version": "0.15.0",
      "requirements": [
        "libdyson==0.8.7"
      ]
    },
    "owlet": {
      "version": "1.0.0",
      "requirements": [
        "PyJWT",
        "gcloud",
        "sseclient",
        "requests_toolbelt"
      ]
    },
    "deepstack_face": {
      "version": "0.9.0",
      "requirements": [
        "pillow",
        "deepstack-python==0.8"
      ]
    },
    "weatheralerts": {
      "version": "0.1.5",
      "requirements": []
    },
    "dlink_hnap": {
      "version": "1.0.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "dyson_local": {
      "version": "0.16.4-4",
      "requirements": [
        "libdyson==0.8.11"
      ]
[... 494 lines cut ...]
      "16": 65.26028270288712,
      "17": 94.48255331375627,
      "18": 95.69133648577223,
      "19": 96.64469941013013,
      "20": 28.30261646762903,
      "21": 28.30261646762903,
      "22": 43.057636198227904,
      "23": 65.26028270288712,
      "24": 65.26028270288712,
      "25": 49.512515447068886,
      "26": 87.33047519856483
    },
    "versions": {
      "bellows": "0.36.3",
      "zigpy": "0.57.1",
      "zigpy_deconz": "0.21.1",
      "zigpy_xbee": "0.18.2",
      "zigpy_znp": "0.11.4",
      "zigpy_zigate": "0.11.0",
      "zhaquirks": "0.0.103"
    }
  }
}
```
</details>

cvocvo, 2023-09-24:
I mentioned it here in https://github.com/zigpy/zha-device-handlers/issues/2588 first, but today I updated the firmware on my SONOFF ZBDongle-E from 6.10.3 to 7.3.1. 
- In case anyone is wondering, I used this to update the firmware from the web easily; it only took me a few minutes to do. https://darkxst.github.io/silabs-firmware-builder/
- Then I removed both of these devices, reloaded the ZHA integration, and then re-added both of them. Then I stuck both of them under my leg for a bit to warm them up and we have data. 

If anyone stumbles across this, it seems https://www.aliexpress.us/item/3256805415172258.html is a bit slower to update for temperature than the other circular one. However, depending on what you're monitoring +/- 10ish minutes per reading may not be too bad.

[image]

closed 2023-09-24 by cvocvo