zigpy/zha-device-handlers #2588: [BUG] SNZB-02 Reports Humidity regularly but never reports a new temperature value

cvocvo, 2023-09-16:
### Bug description

I have two SNZB-02 sensors and both of them regularly report humidity values, however, both of them seem to be 'stuck' with the same temperature value. They never update the value; one is indoors and one is in my shed as an experiment. At minimum the shed temperature should be regularly fluctuating, so I'll post data about that below.

### Steps to reproduce

1. I added the devices and saw initial temperature and humidity values
2. After a while I can see changes in humidity are reported but no changes in temperature ever seem to be reported.

### Expected behavior

Both temperature and humidity values to be regularly reported.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "output_clusters": [
        "0x0003"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "TH01",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
[... 368 lines cut ...]
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>
My logs are 200MB so can't paste them here easily; not sure how to shrink them.

TheJulianJES, 2023-09-16:
Perhaps the device started to sleep during the pairing process, thus missing the setup of the attribute reporting for the temperature value.

If the device has a button, you can you try to "reconfigure the device" (shown on the HA device page for that sensor).
Whilst clicking "start reconfiguration", you can try to press the button (likely a reset button) once. Depending on the sensors firmware, that might cause the sensor to wake up (thus receiving the attribute reporting setup requests).

Otherwise, you'll likely have to allow joins again (click "Add Device") and then reset the sensor, so it can join the network again and hopefully finish configuration correctly that time.
If it still doesn't finish, logs would be helpful. You can paste the section from the time when you joined the device (it will show something like "Device ... joined the network". 

Also, you could try to reposition your coordinator. Maybe there's some interference. Make sure it's on a USB extension cable, away form metal, WiFi access points, bluetooth devices, ...

cvocvo, 2023-09-17:
(not the same device, but has the same problem and is the same model of device)

Reconfiguration on my 2nd SNZB-02 fails; I don't think the button wakes it up and I tried to insert the battery and then hit reconfigure but no luck.
[image]

However, I grabbed the logs from re-adding the device and this seems to be the problem:
`[0xB4F1:1:0x0003]: Configuring cluster attribute reporting
Error handling '_save_attribute' event with (00:12:4b:00:25:16:50:3a, 1, 0, 5, 'TH01', datetime.datetime(2023, 9, 17, 0, 46, 8, 994195, tzinfo=datetime.timezone.utc)) params: FOREIGN KEY constraint failed
[0xB4F1:1:0x0402]: Failed to bind 'temperature' cluster: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4F1), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=5, profile_id=0, cluster_id=0, data=Serialized[b'\x01\xdf_=\x0b\x00o\r\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-32)
[0xb4f1:zdo] ZDO request ZDOCmd.NWK_addr_req: [00:0d:6f:00:0b:3d:5f:df, <AddrRequestType.Single: 0>, 0]
[0xB4F1:1:0x0001]: Failed to bind 'power' cluster: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4F1), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=6, profile_id=260, cluster_id=1026, data=Serialized[b'\x185\x07\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-32)`

Full logs:
<details><summary>Logs</summary>

```python
Device 0xb4f1 (00:12:4b:00:25:16:50:3a) joined the network
[0xb4f1] Scheduling initialization
[0xb4f1] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4F1 ieee=00:12:4b:00:25:16:50:3a is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x00\xf1\xb4:P\x16%\x00K\x12\x00\x80'
[0xb4f1:zdo] ZDO request ZDOCmd.Device_annce: [0xB4F1, 00:12:4b:00:25:16:50:3a, 128]
Tries remaining: 5
[0xb4f1] Requesting 'Node Descriptor'
[0xb4f1] Extending timeout for 0x25 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xF3B8), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=195, profile_id=260, cluster_id=1029, data=Serialized[b'\x18 \n\x00\x00!4\x14'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=228, rssi=-43)
[0xF3B8:1:0x0405] Received ZCL frame: b'\x18 \n\x00\x00!4\x14'
[0xF3B8:1:0x0405] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=32, command_id=10, *direction=<Direction.Client_to_Server: 1>)
[0xF3B8:1:0x0405] Decoded ZCL frame: RelativeHumidity:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=5172))])
[0xF3B8:1:0x0405] Received command 0x0A (TSN 32): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=5172))])
[0xF3B8:1:0x0405] Attribute report received: measured_value=5172
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4F1), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=1, profile_id=0, cluster_id=32770, data=Serialized[b'%\x00\xf1\xb4\x02@\x80\x00\x00P\xa0\x00\x00\x00\xa0\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-32)
Received frame on uninitialized device <Device model=None manuf=None nwk=0xB4F1 ieee=00:12:4b:00:25:16:50:3a is_initialized=False> from ep 0 to ep 0, cluster 32770: b'%\x00\xf1\xb4\x02@\x80\x00\x00P\xa0\x00\x00\x00\xa0\x00\x00'
[0xb4f1] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xb4f1] Discovering endpoints
[0xb4f1] Extending timeout for 0x27 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xB4F1), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=2, profile_id=0, cluster_id=32773, data=Serialized[b"'\x00\xf1\xb4\x01\x01"], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-32)
[... cut ...]
[0xF1D6](S31 Lite zb): Update device availability -  device available: True - new availability: True - changed: False
[0x11B2](lumi.vibration.aq1): Device seen - marking the device available and resetting counter
[0x11B2](lumi.vibration.aq1): Update device availability -  device available: True - new availability: True - changed: False
[0xC315](DS01): Device seen - marking the device available and resetting counter
[0xC315](DS01): Update device availability -  device available: True - new availability: True - changed: False
```

</details>

TheJulianJES, 2023-09-17:
That looks like the sensor is asleep. Can you try to re-pair it? (maybe even try multiple times and/or with fresh batteries)

MattWestb, 2023-09-17:
If its going deep sleep and not being configured OK it can being needed removing the battery and putting it in and doing the reconfigure for it receiving it (Some IKEA controller with latest firmware is one normal problem).

cvocvo, 2023-09-23:
Thank you for your replies; I've been doing a bit of testing. I tried re-pairing a few times and most of them failed and got stuck on "Starting Interview". I wonder if this issue is related to this: https://github.com/home-assistant/core/issues/99497
I've tried repairing with new batteries, reloading the integration, and rebooting home assistant. 

In any case, when I repair either of these SNZB-02 devices the temperature doesn't update beyond the initial update. Humidity updates just fine.

Or perhaps it's an issue with the coordinator I have? Zooz 800 series ZST39 (https://www.getzooz.com/zooz-zst39-z-wave-long-range-usb-stick/)

It seems the only way I can get new devices to pair reliably is via "Add Devices Via This Device" on the Coordinator.

Zigbee Shed Temp

<details><summary>Re-Pairing Logs: Zigbee Shed Temp (Fails to Pair)</summary>

```python
New device 0x955f (00:12:4b:00:2a:52:26:bd) joined the network
[0x955f] Requesting 'Node Descriptor'
Device 0x955f (00:12:4b:00:2a:52:26:bd) joined the network
[0x955f] Requesting 'Node Descriptor'
[0x955f] Requesting 'Node Descriptor'
Device 0x955f (00:12:4b:00:2a:52:26:bd) left the network
Device 0x955f (00:12:4b:00:2a:52:26:bd) joined the network
[0x955f] Requesting 'Node Descriptor'
[0x955f] Requesting 'Node Descriptor'
[0x955f] Requesting 'Node Descriptor'
[0x955f] Requesting 'Node Descriptor'
[0x955f] Requesting 'Node Descriptor'
Device 0x955f (00:12:4b:00:2a:52:26:bd) joined the network
[0x955f] Requesting 'Node Descriptor'
Device 0x955f (00:12:4b:00:2a:52:26:bd) joined the network
[0x955f] Requesting 'Node Descriptor'
[0x955f] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0x955f] Discovering endpoints
[0x955f] Discovered endpoints: [1]
[0x955f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x955f:1] Discovering endpoint information
[0x955f] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7fb509fdb990>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[0x955f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x955f:1] Discovering endpoint information
[0x955f] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7fb509fdb990>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[0x955f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x955f:1] Discovering endpoint information
[0x955f] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7fb509fdb990>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[0x955f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x955f:1] Discovering endpoint information
[0x955f] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7fb509fdb990>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[0x955f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0x955f:1] Discovering endpoint information
```

</details>

<details><summary>Re-Pairing Logs: Zigbee Shed Temp (Fails to Pair again)</summary>

```python
New device 0xd73f (00:12:4b:00:2a:52:26:bd) joined the network
[0xd73f] Requesting 'Node Descriptor'
Device 0xd73f (00:12:4b:00:2a:52:26:bd) joined the network
[0xd73f] Requesting 'Node Descriptor'
[0xd73f] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xd73f] Discovering endpoints
[0xd73f] Discovered endpoints: [1]
[0xd73f] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xd73f:1] Discovering endpoint information
```

</details>

<details><summary>Re-Pairing Logs: Zigbee Shed Temp (Repaired using "Add Devices Via This Device")</summary>

```python
Device 0xf8c5 (00:12:4b:00:2a:52:26:bd) joined the network
[0xf8c5] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7fb5141c9890>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[0xf8c5] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xf8c5:1] Discovering endpoint information
Device 0xf8c5 (00:12:4b:00:2a:52:26:bd) joined the network
[0xf8c5] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0x7fb5141c9890>, 1: <Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
[... cut ...]
Creating cluster handler for cluster id: 3 class: <class 'homeassistant.components.zha.core.cluster_handlers.general.Identify'>
Creating cluster handler for cluster id: 1026 class: <class 'homeassistant.components.zha.core.cluster_handlers.measurement.TemperatureMeasurement'>
Creating cluster handler for cluster id: 1029 class: <class 'homeassistant.components.zha.core.cluster_handlers.measurement.RelativeHumidity'>
Creating cluster handler for cluster id: 1 class: <class 'homeassistant.components.zha.core.cluster_handlers.general.PowerConfigurationClusterHandler'>

```

</details>

TheJulianJES, 2023-09-23:
> Regarding this, can you also attach the diagnostic information from the ZHA integration itself? It can be downloaded on the integrations page.

^

and is that stick a Z-Wave and Zigbee stick?

cvocvo, 2023-09-23:
Can you toss me a screenshot for where to grab the logs you're looking for?

cvocvo, 2023-09-24:
Today I updated the firmware on my SONOFF ZBDongle-E from 6.10.3 to 7.3.1 and at first this didn't do much, however, after several pairing attempts I was able to re-pair the shed temp SNZB-02 with Home Assistant. I still don't think the temperature is regularly updating and humidity still is.
[home-assistant_zha_2023-09-24T19-42-19.274Z_RepairingWorked_zhaonly.log](https://github.com/zigpy/zha-device-handlers/files/12709537/home-assistant_zha_2023-09-24T19-42-19.274Z_RepairingWorked_zhaonly.log)

cvocvo, 2023-09-24:
Update: it seems that the temp is updating. I put the sensor under my leg and let it be for a few minutes and it is updating now! It seems updating the firmware and repairing a few times did the trick!
[image]

I'm re-testing https://github.com/zigpy/zha-device-handlers/issues/2587 as well and it seems that at least one of those is also fixed now.

closed 2023-09-24 by cvocvo