zigpy/zha-device-handlers #2589: [Device Support Request]  ""TZE204_lzriup1j_TS0601"" Avatto Zigbee thermostat panel [Electric heating]

darknessblade, 2023-09-16:
### Problem description

Device/item link:
aliexpress.com/item/1005002960679997.html?

Device is detected by ZHA but does not show any sensor/controls information.

### Solution description

Full climate support for the Thermostat panel. with all functions available under ZHA.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_lzriup1j",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 49661,
    "manufacturer": "_TZE204_lzriup1j",
    "model": "TS0601",
    "name": "_TZE204_lzriup1j TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 140,
    "rssi": -65,
    "last_seen": "2023-09-16T22:39:49",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE204_lzriup1j",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      },
      {
        "name": "PROXY_BASIC"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "ca3af7e440cc19b50985d7e113528db8",
    "area_id": null,
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 74
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE204_lzriup1j"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

System Logs for Cluster info
`[0xB58F:1:0xef00] Unknown cluster command 2 b'\x00\x16\x02\x02\x00\x04\x00\x00\x00\xe6'
[0xB58F:1:0xef00] Unknown cluster command 2 b'\x00\x17\x03\x02\x00\x04\x00\x00\x00\xff'
[0xB58F:1:0xef00] Unknown cluster command 2 b'\x00\x18\x02\x02\x00\x04\x00\x00\x00\xeb'
[0xB58F:1:0xef00] Unknown cluster command 2 b'\x00\x19\x03\x02\x00\x04\x00\x00\x00\xff'
[0xB58F:1:0xef00] Unknown cluster command 2 b'\x00\x1a\x03\x02\x00\x04\x00\x00\x00\xff'`

### Custom quirk

I have tried adding: 
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_electric_heating.py
As a custom quirk with the device ID/type of the Avatto thermostat.

This is the sole modification I did to the File

ShogunMan, 2023-09-17:
Try adding the device/ID to zhaquirks.tuya.ts0601_electric_heating.MoesBHT and see if this quirk works better.

darknessblade, 2023-09-17:
It is not loading any quirk at all.

I checked my custom quirks and I do not have the moes BHT custom one. do you have a link for it, or is it the same as the default one?

I do have a temp sensor [TS0601 type] that loads a different quirk, so its not a issue with the custom quirk settings.

Trying to add a different one {HAOZEE} does not help either.
It still gives the same type of error in the logs

`[0xC1FD:1:0xef00] Unknown cluster command 2 b'\x00\x02\x04\x04\x00\x01\x01'
[0xF153:1:0xef00] Unknown cluster command 2 b'\x00\x15\x03\x02\x00\x04\x00\x00\x00\xf3'
[0xF153:1:0xef00] Unknown cluster command 17 b'\x00\x0f@'
[0xF153:1:0xef00] Unknown cluster command 2 b'\x00\x17\x03\x02\x00\x04\x00\x00\x00\xf4'
[0xF153:1:0xef00] Unknown cluster command 2 b'\x00\x18\x03\x02\x00\x04\x00\x00\x00\xf4'
`

ShogunMan, 2023-09-17:
It is the default ts0601_electric_heating quirk. It just adds the MoesBHT class when loaded.
It is not loading the quirk because one of the input, and both output clusters does not match the quirk.

It has to be adjusted to support input cluster "0xef00" and output clusters "0x000a" and "0x0019".

darknessblade, 2023-09-17:
What changes do I need to make to support those input clusters?

darknessblade, 2023-09-20:
@ShogunMan I have received this file from the Manufacturer, My device does not look to be listed, but it does give quite a lot of "Tuya" data-points/cluster info for many devices.

It does list  "TS0601_thermostat" entries but these are for thermostat valves

https://github.com/darknessblade/-TZE204_lzriup1j_TS0601-Avatto-Zigbee-thermostat-panel-Electric-heating-/blob/main/tuya%20(1).ts

ShogunMan, 2023-09-20:
@darknessblade I've double checked and in fact all clusters on endpoint 1 are handled correctly by the current quirk.
However your device also lists endpoint 242 with a single output cluster that is not handled by the current quirk.
Can you try this modified quirk (rename extension to .py)

[ts0601_heating_avatto.txt](https://github.com/zigpy/zha-device-handlers/files/12671212/ts0601_heating_avatto.txt)

darknessblade, 2023-09-20:
I have checked but still nothing.

Custom quirks should be enabled
[image]
[image]

[image]

From the logs:
`Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 460, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_heating_avatto.py", line 180, in <module>
    class MoesAvatto(TuyaThermostat):
  File "/config/custom_zha_quirks/ts0601_heating_avatto.py", line 205, in MoesAvatto
    OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
                      ^^^^^^^^^^^^^^^
NameError: name 'GreenPowerProxy' is not defined`

ShogunMan, 2023-09-20:
Ok, there was a bug in the previous file. Try this:
[ts0601_heating_avatto.txt](https://github.com/zigpy/zha-device-handlers/files/12671461/ts0601_heating_avatto.txt)

darknessblade, 2023-09-20:
With that last change I can see a CLIMATE entity under controls
[image]

ShogunMan, 2023-09-20:
I'll make a bare minimum quirk, if that does not work we have to call on @TheJulianJES or @javicalle :-)

darknessblade, 2023-09-20:
Please do so.

The minimum quirk features, I would like to have

1 Climate [Manual temperature/ define temperature]
2 Child lock

ShogunMan, 2023-09-20:
Try this:
[ts0601_heating_minimum.txt](https://github.com/zigpy/zha-device-handlers/files/12671666/ts0601_heating_minimum.txt)

If this does not work, then i believe the manufacturer specific cluster is not compatible with Moes and more info is needed.

darknessblade, 2023-09-20:
I have checked and it had no effect on controls.

Will ask the manufacturer for the cluster information

ShogunMan, 2023-09-20:
But it loaded the minimum quirk? Then we can use that as a starting point for implementing the manufacturer specific cluster.

darknessblade, 2023-09-20:
It did load the minimum quirk. but I am unable to do anything else.

The thermal-runaway bug [bug where the temp keeps going up] keeps happening even with the minimal quirk.

{In order to properly load the quirk each time, I removed the device, and rebooted hass}

ShogunMan, 2023-09-20:
We could try the Haozee cluster for a test. It seems from your log that it is looking for this. Let me throw something together.

darknessblade, 2023-09-20:
OK. please send me some files for me to test, and I will report back

ShogunMan, 2023-09-20:
Give this a shot:
[ts0601_heating_avatto.txt](https://github.com/zigpy/zha-device-handlers/files/12672066/ts0601_heating_avatto.txt)

darknessblade, 2023-09-20:
this one does not look to work either

[image]

ShogunMan, 2023-09-20:
Ok, there is one last thing we could try, let me do a little research. I might be able to find the manufacturer cluster spec.

ShogunMan, 2023-09-20:
Found a bug, try this instead:
[ts0601_avatto2.txt](https://github.com/zigpy/zha-device-handlers/files/12672300/ts0601_avatto2.txt)

ShogunMan, 2023-09-20:
Can you remove/delete the device, then add it again?

darknessblade, 2023-09-20:
Tried removing/adding it back, and it did not change anything in the controls/sensors

ShogunMan, 2023-09-20:
For full functionality, I believe we need the full spec from the manufacturer.
But you are able to control the device from within HA? Child lock works?

darknessblade, 2023-09-20:
Child lock does not work. [To activate it on the panel its: HOLD temp-up/temp-down 5 seconds]
When activating it on the panel, it does not change the state of the "switch"

"Thermal-runaway still happens."

darknessblade, 2023-09-20:
Checking the logs:

`[0x3887:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xD52B:1:0x0102]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0300]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xD52B:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]`

darknessblade, 2023-09-20:
Instead of Thermal runaway, it now keeps triggering the "change time" button

darknessblade, 2023-09-20:
According to the seller the File I linked are the Zigbee2MQTT firmware details they also gave Zigbee2MQTT

darknessblade, 2023-09-20:
@ShogunMan The seller/manufacturer asked me what kind of Firmware file they should provide.

Should I ask for the FULL firmware file?

darknessblade, 2023-09-20:
They did say the file I linked was what they send to Zigbee2MQTT team.
I do not use Zigbee2MQTT myself, and do not have any way to test it out. [as I only have 1 zigbee adapter]

Not forgetting I have already a few dozen automations that rely on ZHA, so converting to Zigbee2MQTT is not something I can do within a few hours/days.

ShogunMan, 2023-09-20:
The only thermostats supported in Z2MQTT are TZE200_TS0601 devices. Yours is a TZE204.
Without proper documentation I am unable to progress. Maybe @javicalle or @TheJulianJES has any suggestions?

darknessblade, 2023-10-06:
@ShogunMan I have received this from the seller:
https://docs.google.com/spreadsheets/d/1ypoEE_izSjgS_P_WrWDGgSqzytljt_y6/edit?usp=sharing&ouid=107045649919621059641&rtpof=true&sd=true

Do you think it would be of any use

jacekk015, 2023-10-07:
Looks like another variation of Avatto
1. Remove TRV from HA

2. Use attached quirk - remove temporary other one that matches your TRV.
[ts0601_thermostat_avatto.py.zip](https://github.com/zigpy/zha-device-handlers/files/12838138/ts0601_thermostat_avatto.py.zip)

3. Enable debug level logging:
```
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
    zhaquirks: debug
```
4. Restart HA
5. Pair TRV like a brand new
6. Wait 10 minutes - IMPORTANT!!!
7. Post here FULL log from HA - DO NOT CUT ANYTHING!

Device will not work now, but for few minutes after pairing it will present all attributes from device.

darknessblade, 2023-10-07:
Zigbee logs [new to old] [1st attempt to add the  device, entities did not show up, as it did not correctly add the custom quirks]

`Logger: zigpy.zcl
Source: runner.py:186
First occurred: 6:01:22 PM (10 occurrences)
Last logged: 6:01:24 PM

[0xA6F6:1:0xef00] Unknown cluster command 2 b'\x00\t\x03\x02\x00\x04\x00\x00\x00\xe7'`

`
Logger: homeassistant.components.zha.core.cluster_handlers
Source: components/zha/core/cluster_handlers/__init__.py:537
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 6:01:18 PM (6 occurrences)
Last logged: 6:01:19 PM

[0xC163:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xC163:1:0x0b04]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0300]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
`

`Logger: zigpy.application
Source: runner.py:186
First occurred: 6:01:12 PM (2 occurrences)
Last logged: 6:01:12 PM

Unknown device AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xA6F6)`

`Logger: zhaquirks
Source: components/zha/__init__.py:122
First occurred: 6:00:34 PM (1 occurrences)
Last logged: 6:00:34 PM

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers`

darknessblade, 2023-10-07:
`Logger: zigpy.zcl
Source: runner.py:186
First occurred: 6:21:46 PM (32 occurrences)
Last logged: 6:22:02 PM

[0xA789:1:0xef00] Unknown cluster command 2 b'\x00\x15\x03\x02\x00\x04\x00\x00\x00\xf4'
[0xA789:1:0xef00] Unknown cluster command 2 b'\x00\x19\x03\x02\x00\x04\x00\x00\x00\xf4'
[0xA789:1:0xef00] Unknown cluster command 17 b'\x00\x0c@'`

`Logger: homeassistant.components.zha.core.cluster_handlers
Source: components/zha/core/cluster_handlers/__init__.py:537
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 6:18:53 PM (6 occurrences)
Last logged: 6:18:55 PM

[0xC163:1:0x0b04]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0300]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xC163:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]`

Logger: zhaquirks
Source: components/zha/__init__.py:122
First occurred: 6:18:08 PM (1 occurrences)
Last logged: 6:18:08 PM

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers

darknessblade, 2023-10-07:
From the dev tools:

`event_type: zha_event
data:
  device_ieee: a4:c1:38:58:cb:b7:8d:52
  unique_id: a4:c1:38:58:cb:b7:8d:52
  device_id: ca3af7e440cc19b50985d7e113528db8
  device_event_type: device_offline
origin: LOCAL
time_fired: "2023-10-07T16:21:50.801622+00:00"
context:
  id: 01HC5F3X2HYTJCACYWFZ9NA0QN
  parent_id: null
  user_id: null`

darknessblade, 2023-10-07:
[0x7383:1:0xef00] Unknown cluster command 2 b'\x00\x1b\x03\x02\x00\x04\x00\x00\x00\xf4'

Logger: zigpy.zcl
Source: runner.py:186
First occurred: 6:29:49 PM (118 occurrences)
Last logged: 6:31:02 PM

[0x7383:1:0xef00] Unknown cluster command 36 b'\x13\x00'
[0x7383:1:0xef00] Unknown cluster command 2 b'\x00\x14\x03\x02\x00\x04\x00\x00\x00\xf4'
[0x7383:1:0xef00] Unknown cluster command 2 b'\x00\x15\x03\x02\x00\x04\x00\x00\x00\xf4'
[0x7383:1:0xef00] Unknown cluster command 2 b'\x00\x16\x03\x02\x00\x04\x00\x00\x00\xf4'
[0x7383:1:0xef00] Unknown cluster command 2 b'\x00\x17\x03\x02\x00\x04\x00\x00\x00\xf4'

Logger: homeassistant.components.zha.core.cluster_handlers
Source: components/zha/core/cluster_handlers/__init__.py:537
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 6:28:48 PM (6 occurrences)
Last logged: 6:28:50 PM

[0xC163:1:0x0702]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0xC163:1:0x0b04]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0006]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0300]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]
[0x3887:1:0x0008]: async_initialize: all attempts have failed: [DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>'), DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>')]

Logger: zhaquirks
Source: components/zha/__init__.py:122
First occurred: 6:28:00 PM (1 occurrences)
Last logged: 6:28:00 PM

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers

darknessblade, 2023-10-07:
I have tried to add it multiple times [Removing/re-adding it+ restarting HASS each time]
but no attributes show up.

darknessblade, 2023-10-07:
If it doesn't work after all this time, I might just return the Panel, and buy one that is supported

Still have about a week left before I can't open a dispute  on aliexpress.

jacekk015, 2023-10-07:
@darknessblade You didn't do what I asked for.
Repeat the steps mentioned and post FULL HA log here.

HA logs are under:
Settings -> System -> Logs
You can click download and drag/drop that file into Github message window - it will add it as a text file

darknessblade, 2023-10-07:
[home-assistant_2023-10-07T17-13-48.505Z.log](https://github.com/zigpy/zha-device-handlers/files/12838351/home-assistant_2023-10-07T17-13-48.505Z.log)

jacekk015, 2023-10-08:
@darknessblade Your device doesn't follow standard of communication.
Replace my previous quirk file with the one below
[ts0601_thermostat_avatto.py.zip](https://github.com/zigpy/zha-device-handlers/files/12841071/ts0601_thermostat_avatto.py.zip)
and repeat steps above and drop again log file here.

darknessblade, 2023-10-08:
Hello, with this quirk file, there are still no entities/attributes visible

here are the logs:

[home-assistant_2023-10-08T16-34-12.523Z.log](https://github.com/zigpy/zha-device-handlers/files/12841187/home-assistant_2023-10-08T16-34-12.523Z.log)

darknessblade, 2023-10-08:
If I check the logs it looks to be a Endpoint failure
`
2023-10-08 18:22:47.947 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE204_lzriup1j TS0601 (a4:c1:38:58:cb:b7:8d:52)
2023-10-08 18:22:47.948 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_thermostat_avatto.Avatto'>
2023-10-08 18:22:47.948 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1} {1, 242}
`

jacekk015, 2023-10-08:
That's true. You have extra endpoint for GreenProxy
Attached two version. Please firstly remove device from HA so the initialization could be full, not re-join.
Firstly use _first quirk and repeat steps.
Then please use _second quirk with _first removed.
Changes made in second version may not be needed.

[quirks.zip](https://github.com/zigpy/zha-device-handlers/files/12841504/quirks.zip)

You will not see any entities for now. We are trying to find why ZHA have problems with internal communication between Zigpy and TRV.

darknessblade, 2023-10-08:
With the First Version after I paired it, I  saw entities in the pairing page. 
Also checked the entities in the device menu

Quirk: ts0601_thermostat_avatto.Beok

Logs [first]:

[home-assistant_2023-10-08T20-21-41.419Z.log](https://github.com/zigpy/zha-device-handlers/files/12841668/home-assistant_2023-10-08T20-21-41.419Z.log)

darknessblade, 2023-10-08:
The 2nd one also shows entities like:

Temperature
Child lock

But they do not look to be usable.

jacekk015, 2023-10-08:
Strangely device is not much communicative on start.
It's only sending room temp:
```
2023-10-08 22:10:42.051 DEBUG (MainThread) [zhaquirks.tuya] [0x9d22:1:0xef00] Received value [0, 0, 0, 224] for attribute 0x0203 (command 0x0002)
```
That's 22,4 degrees Celsius
That google doc your Chinese friend gave you is probably correct for attributes.
New quirk file has to be build. That Avatto quirk has already mixed code exceptions for Beok.

Which exact version you own? Gas, Water or Electric??

darknessblade, 2023-10-09:
@jacekk015 
The version I have is in the title: [electric heating]

jacekk015, 2023-10-14:
@darknessblade In my repo https://github.com/jacekk015/zha_quirks
use quirk file named:
ts0601_thermostat_avatto2.py

Since that's first version bugs are possible.
Yout thermostat isn't too chatty so remove previous test quirks and put that one.
Restart HA. Remove Thermostat from HA. And pair like a new.

Then play on the Thermostat with options you've mentioned above and look for results on HA panel.
In case of bugs or other things, try to observe the time you've used some function so I could check the logs on that time.
Best is to leave some time between using different function so there will be pause in the logs.
Of course attach logs for debugging in case of errors.

darknessblade, 2023-10-14:
@jacekk015 will test it.

currently updating HASS to the latest version, so will test it afterwards.

darknessblade, 2023-10-14:
Features working:

-Child lock
-Temp sensor
-Number temp [this is used for the offset in temperature]

Features not working:
-Temp control
-HVAC [on/off]

LOGS:

[home-assistant_2023-10-14T19-38-10.710Z.log](https://github.com/zigpy/zha-device-handlers/files/12907851/home-assistant_2023-10-14T19-38-10.710Z.log)

darknessblade, 2023-10-14:
at the moment I am able to turn/off the Relay trough the temp-offset.

It is currently not the best way to control the device, but it is in a Semi-working state now

jacekk015, 2023-10-14:
Need to be little more patience.
HVAC [on/off] should work now - Chinese friends didn't named it in the Google sheet properly.
Download code again.

> Features not working:
> -Temp control
You need to elaborate more, because I don't see in the logs any tries for HA->Thermostat communication for target temp.

You need to replace quirk and restart HA. You can re-pair just to be sure.
Change temp in HA few times, wait few seconds after each time.
Post the logs.

Post drag/drop image from HA device screen.

darknessblade, 2023-10-14:
> Need to be little more patience. HVAC [on/off] should work now - Chinese friends didn't named it in the Google sheet properly. Download code again.
>
>
> You need to replace quirk and restart HA. You [...]

It looks to be working now.
for the Hvac to show up I had to change the state/turn on/off the hvac trough the device page.

Will test it out the following days. to see if there are any bugs with the current custom quirks

jacekk015, 2023-10-14:
You need to wait for the Thermostat to sync with HA.
Remember - if Thermostat doesn't send data to HA, HA doesn't know anything.
HA cannot force Thermostat to talk - it just waits for data.
That's probably why HVAC state wasn't immediately visible in HA.

When you switch HVAC state, then HA sends data to Thermostat, and Thermostat sends confirmation to HA.
ONLY then HA changes the state in GUI. If the Thermostat won't respond properly HA will change state back to previous.

For child lock situation - can't tell anything without the logs.
It's possible that Child Lock state on, locks the changes at the Thermostat firmware level and HA has nothing to do here.
It sends data to Thermostat but Thermostat doesn't do changes.
I need to see the logs.

darknessblade, 2023-10-14:
Here are the logs for HVAC/childlock being changed from within HASS.

Where the childlock does not prevent HVAC from turning on the relay/heating

[home-assistant_2023-10-14T20-55-38.752Z.log](https://github.com/zigpy/zha-device-handlers/files/12908013/home-assistant_2023-10-14T20-55-38.752Z.log)

jacekk015, 2023-10-14:
```
2023-10-14 22:54:25.237 DEBUG (MainThread) [zhaquirks.tuya] [0xfac2:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2600 to custom {514: 260}
2023-10-14 22:54:25.240 DEBUG (MainThread) [zigpy.zcl] [0xFAC2:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=220, command_id=0, *direction=<Direction.Server_to_Client: 0>)
2023-10-14 22:54:25.244 DEBUG (MainThread) [zigpy.zcl] [0xFAC2:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=220, command_id=514, function=0, data=260))
2023-10-14 22:54:25.302 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xFAC2), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=135, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\xdc\x0b\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=120, rssi=-70)
2023-10-14 22:54:25.302 DEBUG (MainThread) [zigpy.zcl] [0xFAC2:1:0xef00] Received ZCL frame: b'\x18\xdc\x0b\x00\x00'
2023-10-14 22:54:25.303 DEBUG (MainThread) [zigpy.zcl] [0xFAC2:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=220, command_id=11, *direction=<Direction.Client_to_Server: 1>)
2023-10-14 22:54:25.304 DEBUG (MainThread) [zigpy.zcl] [0xFAC2:1:0xef00] Decoded ZCL frame: Avatto2ManufCluster:Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
2023-10-14 22:54:25.305 DEBUG (MainThread) [zigpy.zcl] [0xFAC2:1:0xef00] Received command 0x0B (TSN 220): Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
```

You've send request to change to 26 degrees
`
2023-10-14 22:54:25.237 DEBUG (MainThread) [zhaquirks.tuya] [0xfac2:1:0x0201] Mapping standard occupied_heating_setpoint (0x0012) with value 2600 to custom {514: 260}
`

Thermostat responded with Success. HA thinks it's all OK.
`
2023-10-14 22:54:25.305 DEBUG (MainThread) [zigpy.zcl] [0xFAC2:1:0xef00] Received command 0x0B (TSN 220): Default_Response(command_id=0, status=<Status.SUCCESS: 0>)
`

Moreover - Thermostat sent back updated temp data
```
2023-10-14 22:54:25.375 DEBUG (MainThread) [zhaquirks.tuya] [0xfac2:1:0xef00] Received value [0, 0, 1, 4] for attribute 0x0202 (command 0x0002)
2023-10-14 22:54:25.375 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xFAC2:1:0x0201]: Attribute report 'Avatto2Thermostat'[occupied_heating_setpoint] = 2600
2023-10-14 22:54:25.376 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xFAC2:1:0x0201]: Attribute report 'Avatto2Thermostat'[occupied_heating_setpoint] = 2600
```

Check it later again - but there are no errors there.

darknessblade, 2023-10-15:
I do not know what causes it but I sometimes still get the thermal-runaway bug

[home-assistant_2023-10-15T14-51-37.692Z.log](https://github.com/zigpy/zha-device-handlers/files/12909848/home-assistant_2023-10-15T14-51-37.692Z.log)

darknessblade, 2023-10-15:
As for the weekly programming feature, this isn't needed. as I have disabled it in the advanced settings in my thermostat.

[24 more replies]

PR #3748 (2025-01-25): Add Avatto thermostats (Tuya)
## Proposed change

Adds Tuya ZWT198/ZWT100-BH Avatto wall thermostats

- _TZE204_lzriup1j
- _TZE200_viy9ihs7
- _TZE204_xnbkhhdr
- _TZE284_xnbkhhdr

Closes: #3627
Closes: #3212
Closes: #2589
Closes: #3482
Closes: #3326

## Additional information

Based on: https://github.com/Koenkk/zigbee2mqtt/discussions/25752
and https://github.com/Koenkk/zigbee-herdsman-converters/blob/40e72bc2515deb67ef35dddab04437342b7b3eb9/src/devices/tuya.ts#L7027
## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using [... cut]
closed 2025-01-25 by TheJulianJES