zigpy/zha-device-handlers #2597: [BUG] MOES BHT-002-GCLZBWT alias _TZE200_aoclfnxz set temperature offset not possible

cwm77, 2023-09-18:
### Bug description

Within ZHA it is not possible to customize with the option "Manage zigbee device" the local temperature offset of the thermostat. So far, I used a Deconz stick, but would like to test on SkyConnect as well. With the Conbee II from Deconz, setting a temperature offset works flawlessly.

Home Assistant 2023.9.2 Supervisor 2023.09.2 Operating System 10.5 Frontend 20230911.0 - latest

"versions": {
  "bellows": "0.36.3",
  "zigpy": "0.57.1",
  "zigpy_deconz": "0.21.1",
  "zigpy_xbee": "0.18.2",
  "zigpy_znp": "0.11.4",
  "zigpy_zigate": "0.11.0",
  "zhaquirks": "0.0.103"
}

### Steps to reproduce

1. Go in settings of HA to Device & Services/Zigbee Home Automation
2. Click on the first entry in the list (your number) devices
3. Select _TZE200_aoclfnxz TS0601
4. Click next to RECONFIGURE on the three dots
5. Select "Manage Zigbee device"
6. Select under CLUSTERS "MoesBHTThermostat (Endpoint id: 1, Id: 0x0201, Type: in)"
7. Select under Attributes of the selceted cluster "local_temperature_calibration (id: 0x0010)"
8. Enter in Manufacture code to override the desired Value, e.g. 100 (this represents a shift of 1 degree Celsius upwards)
9. Click Write attribute
10. Click Read attribute
11. Value = None

### Expected behavior

In the case of the above example, the expectation is that value = 100 is specified.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

jacekk015, 2023-10-14:
Temp calibration is supported under custom quirk at:
https://github.com/jacekk015/zha_quirks

use quirk file:
ts0601_thermostat_avatto.py

cwm77, 2023-10-15:
> Temp calibration is supported under custom quirk at: https://github.com/jacekk015/zha_quirks
> 
> use quirk file: ts0601_thermostat_avatto.py

Thanks, I tested the Quirk file and yes, the calibration for _TZE200_aoclfnxz works. But the decimal point of the current temperature of the thermostat displayed in HA has unfortunately slipped one place to the right - instead of 22.5 degrees celsius 225 degrees celsius.

jacekk015, 2023-10-15:
There's no guessing here. Someone with that Thermostat id confirmed that all is OK.
So either he was lying or chineese friends made a change in the firmware using same id. Later I'll try to look at Z2M code for that Thermostat.
I can't change that back and forth because next guy will ask me to revert those changes back.
Sometimes they use same id with other device_version parameter.

Paste here full signature from device panel or better use 3-dots menu and put here Device diagnostics

cwm77, 2023-10-15:
I compared the previously used quirk with this one and found that the values for the DIRECT_MAPPED_ATTRS are different. 

I changed:

    DIRECT_MAPPED_ATTRS = {
        AVATTO_TEMPERATURE_ATTR: (
            "local_temperature",
            lambda value: value * 100,
            lambda value: value * 10,
        ),
        AVATTO_TARGET_TEMP_ATTR: (
            "occupied_heating_setpoint",
            lambda value: value * 100,
            lambda value: value * 10,
to

    DIRECT_MAPPED_ATTRS = {
        AVATTO_TEMPERATURE_ATTR: (
            "local_temperature",
            lambda value: value * 10,
        ),
        AVATTO_TARGET_TEMP_ATTR: (
            "occupied_heating_setpoint",
            lambda value: value * 100,

and now it works. 

Here is still the device diagnostic file.

[zha-fec7d948e227df4df792df21d0014c72-_TZE200_aoclfnxz TS0601-6ebe4206c1cf5134efa5cd13a6f0ce84.json.txt](https://github.com/zigpy/zha-device-handlers/files/12909647/zha-fec7d948e227df4df792df21d0014c72-_TZE200_aoclfnxz.TS0601-6ebe4206c1cf5134efa5cd13a6f0ce84.json.txt)

jacekk015, 2023-10-15:
Your TRV i Z2M is here
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/moes.ts#L100

and room temp code is here, in legacy file
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/legacy.ts#L3851

There are some exceptions for other TRV than: _TZE200_u9bfwha0, _TZE200_aoclfnxz
So probably those two behaves similar.
My quirk had exception for _TZE200_u9bfwha0 but I had no one with TRV _TZE200_aoclfnxz

Quirk in repo has been modified to work properly.
If you had gave me debug logs from HA I could check it myself.
Again - quirk modified.

https://github.com/jacekk015/zha_quirks
ts0601_thermostat_avatto.py

cwm77, 2023-10-20:
> Your TRV in Z2M is here https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/moes.ts#L100
> 
> and room temp code is here, in legacy file https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/legacy.ts#L3851
> 
> There are some exceptions for other TRV than: _TZE200_u9bfwha0, _TZE200_aoclfnxz So probably those two behaves similar. My quirk had exception for _TZE200_u9bfwha0 but I had no one with TRV _TZE200_aoclfnxz
> 
> Quirk in repo has been modified to work properly. If you had gave me debug logs from HA I could check it myself. Again - quirk modified.
> 
> https://github.com/jacekk015/zha_quirks ts0601_thermostat_avatto.py

Thanks, I tested the quirk, and it seems to work now with tze200_aoclfnxz. Here is the debug log to check. 
[home-assistant_zha_2023-10-20T09-45-20.000Z.log](https://github.com/zigpy/zha-device-handlers/files/13052837/home-assistant_zha_2023-10-20T09-45-20.000Z.log)

Can I ask one more thing. I also have brand new tze204_aoclfnxz thermostats from Moes (It looks just like tze200_aoclfnxz BHT-002, so it is a replacement for its predecessor). I found that there are https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto2.py for tze204_xxxxxxxx thermostats of this type. Unfortunately, an attempt at a quick fit failed. The thermostat was recognized, but he could not be turned on. Can we try to adjust this thermostat as well?

jacekk015, 2023-10-22:
> Can I ask one more thing. I also have brand new tze204_aoclfnxz thermostats from Moes (It looks just like tze200_aoclfnxz BHT-002, so it is a replacement for its predecessor). I found that there are https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto2.py for tze204_xxxxxxxx thermostats of this type. Unfortunately, an attempt at a quick fit failed. The thermostat was recognized, but it could not be turned on. Can we try to adjust this thermostat as well?
> 
> I can make the thermostat come to live with this quirk https://github.com/zigpy/zha-device-handlers/issues/2433#issuecomment-1690198060 , but it has no way to change the temperature offset.

For your _TZE204_aoclfnxz test updated file:
ts0601_thermostat_avatto.py
from: https://github.com/jacekk015/zha_quirks

Next time better don't mix different TRV's in one thread. Later people can get wrong quirk.

cwm77, 2023-10-23:
Thanks, I tested the quirk, and it works flawless as well with _TZE204_aoclfnxz.

closed 2023-10-24 by cwm77