zigpy/zha-device-handlers #2599: [Device Support Request] linptech Human Presence Sensor

seanhagen, 2023-09-20:
### Problem description

Request to add support for the linptech Human Presence Sensor ( aka Moes Human Presence Sensor -- might be a rebrand of the linptech )

It looks like [folks in this thread are getting close to having it working in zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt/issues/18637).

### Solution description

Integration into ZHA or zha_quirk to get the device working in HA.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0500",
        "0x4000",
        "0xe002",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3218_awarhusb",
  "model": "TS0225",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Vancouver",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
[... 417 lines cut ...]
                "value": 32771
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

betimmm, 2023-09-28:
Same need here, can anyone help?

judahrand, 2023-11-02:
This seems to be working in z2mqtt, I believe. I've had a play with trying to write a custom quirk but I'm not really sure what I'm doing! If anyone is willing/able to help I'd appreciate it!

felafel, 2023-11-04:
> This seems to be working in z2mqtt, I believe. I've had a play with trying to write a custom quirk but I'm not really sure what I'm doing! If anyone is willing/able to help I'd appreciate it!

I have only today learned about ZHA quirks.  If you would like someone to test it let me know.  Thanks

maelremrem, 2023-11-07:
Got mine today, and I only have the occupancy working. Can someone help me to write an ZHA quirk ?

felafel, 2023-11-07:
I tried to use a different quirk that is similar and change it to accept this model.  That did not work for me.  I am going to look into Zigbee2MQTT but not sure if it works with SkyConnect.

cbarth3, 2023-11-14:
Picked one of these up. If I can help in some way to get it working with zha let me know

felafel, 2023-11-14:
> Picked one of these up. If I can help in some way to get it working with zha let me know

I would say that writing a quirk for it would be great.

danielgomezrico, 2023-11-16:
@felafel can you open a PR with that code? Opening that may show what code did you add to get some advice

felafel, 2023-11-16:
I am not sure if this is helpful but here is what I have.  I do not know why it is not being picked up and I have not found a way to see if the quirk is being attempted.  I set up the settings I have seen in the YAML file but whether it is being used is a mystery.  It would be helpful if ZHA could confirm that it is looking for quirks in my folder, and also, whether it cannot use files that are there for whatever reason.  If there is a file there then ZHA should assume I want to use it, I think.

`"""Device handler for Tuya ZG-205Z-A Mini 24Ghz human presence sensor."""

import math
from typing import Dict, Optional, Tuple, Union
from zigpy.profiles import zgp, zha
from zigpy.profiles.zha import DeviceType
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation

from zigpy.zcl.clusters.general import Basic, Identify, Ota, GreenPowerProxy, AnalogOutput
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
)

from zhaquirks import MotionWithReset
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaManufCluster, TuyaNewManufCluster, TuyaZBE000Cluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class MotionCluster(MotionWithReset):
    """Motion cluster."""

    reset_s: int = 60

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaMmwRadarFadingTime(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for fading time."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Fading time")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 0)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 600)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)
        self._update_attribute(
            self.attributes_by_name["engineering_units"].id, 73
        )

class TuyaMmwRadarLargeMotionDetectionSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for Large motion detection sensitivity."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(self.attributes_by_name["description"].id, "Large motion sensitivity")
        self._update_attribute(self.attributes_by_name["min_present_value"].id, 0)
        self._update_attribute(self.attributes_by_name["max_present_value"].id, 10)
        self._update_attribute(self.attributes_by_name["resolution"].id, 1)

class TuyaMmwRadarLargeMotionDetectionDistance(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for Large motion detection distance."""
[... 308 lines cut ...]
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarStaticMotionDetectionSensitivity,
                ],
                OUTPUT_CLUSTERS: [],
            },
            8: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarStaticMotionDetectionDistance,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }`

felafel, 2023-11-21:
Does @danielgomezrico  or anyone else have constructive help with the code above?

danielgomezrico, 2023-11-22:
I was a bit out of clues about how to build this one, but I want to try

kaiserkiwi, 2023-12-02:
Bought some of these because they were recommended. Sadly the person that recommended it didn't know about ZHA support (he said it works out of the box, but he meant Z2MQTT. 

Any news on this?

mroony, 2023-12-06:
I did the same and too have no clue on how to make it better.

felafel, 2023-12-26:
> According to blakkadder the device seems to work already. I don't have the opportunity to test it right now, so any updates on this?
> 
> See https://zigbee.blakadder.com/Linptech_ES1.html

There is an attempted ZHA Quirk above to get the other features to work.  Presence itself appears to be in ZHA but the other features do not.  Do you have any suggestions on getting the quirk to work?

kaiserkiwi, 2023-12-26:
I sadly have no experience in Python or ZHA quirks. So I'm just here to ask. 😅

When I use it the way it works now and someone writes a quirk does it automatically work or do I have to re-pair it?

felafel, 2024-01-10:
I might just go with Zigbee2MQTT.  This is crazy that no-one can fix this.  ZHA should just deprecate if it cannot keep up with an alternative that seems more feature rich.  My concern was that SkyConnect would not work but I think it does now so hopefully it does and I can just get my devices to be fully utilised.

dmulcahey, 2024-01-10:
@felafel i understand you may be frustrated but everyone working on these things are volunteers. This isn’t the appropriate place to vent.

kaiserkiwi, 2024-01-10:
I think it's fair to point out that the official implementation is much worse in many points than a community alternative. The wrong place to vent? Definitely. But he is right in my opinion.

I'm really frustrated too. Tried to switch ti Zigbee2MQTT, failed and now are trapped in this. 

I really wish that ZHA would get much more attention. Not only for Device support.

dmulcahey, 2024-01-10:
> I think it's fair to point out that the official implementation is much worse in many points than a community alternative. The wrong place to vent? Definitely. But he is right in my opinion.
>
> I'm [...]

What am I missing? All of this is FREE and done by folks in their spare time… ZHA came from the community and is largely maintained by community members… no one forces anyone to use it. everything here in this repo is from the community. The problem is with device manufacturers… they basically do what they want. 

Tbh, being frustrated is perfectly normal and acceptable but dumping it on folks who put a lot of effort into this in their free time is just not right. Try to look at the situation from the opposite perspective and imagine how you would feel.

kineticscreen, 2024-01-10:
I think my frustration is certainly less about the work of the volunteers, which as you say is entirely supported by the community, and more that I listened to Home Assistant when they announced ZHA as the native support for Zigbee. I was at the time on Zigbee2MQTT and changed over thinking that I was moving to the more stable / supported platform. Since then it's consistently lagged in features and support. For example, at the time I already had OTA updates presented nicely in the UI for Z2M, years late ZHA still handles OTA update incredibly unintuitively. I kept expecting that such basic things were only around the corner, but like the experience of this presence sensor, there's no predictable outcome for ZHA based on what Z2M is managing to do.

Expressing frustration at the countless hours spent by volunteers on ZHA is definitely missing the point, and they don't owe anyone anything, but to someone starting a Home Assistant build afresh, they're still encouraged that ZHA is best option for them to set up their Zigbee network, and I just don't think that's proven to be accurate.

kaiserkiwi, 2024-01-24:
To come back to the initial topic: Has someone already implemented a custom quirk that we could test to get this running? 

I really would like to use the luminance sensor in this thing instead of trying to guess the current light based on time and weather. 😅

milobt, 2024-01-25:
I have 5 of these and they work pretty well, however, there is a lag from some of them some time, not sure if this is because of some interference or due to the sensors flooding the zigbee log.

Some good news of the LED indicator front, I found to that when pairing these sensors with a TUYA hub, instead of my Z2M network, it pops up saying there’s an update, specifically to allow you to turn off the indicator LED.
So, if you update the device and then set the LED to off in the TUYA app, remove the device from the TUYA hub and repair with you Z2M network, it keeps the off setting and so stays off when motion is detected - GREAT!

This does posisble mean that there is now a cluster exposed to be able to configure the LED state over zigbee so anyone that can, please look into whether the LED cluster is exposed and add it to the device configuration.

comperta, 2024-04-15:
Hello everyone, 

The _TZ3218_awarhusb module uses different clusters than those used by _TZE200_2aaelwxk so we cannot simply add the device, despite the fact that they are both TS0225 models.

I tried to implement a quirk that takes into account the signature of the _TZ3218_awarhusb module, without success so far. 
I simply managed to recover the "Fading Time" and only in the device management. 

I'll give you the code I use, if someone has better knowledge than me (first time I'm getting into this), I'm interested :-) 

I am blocked in particular on two points: 

- For the luminance the cluster is 0x4000 instead of 0x0400 and I cannot read it. 

- For the "custom" cluster 0xE002 I tried to implement it but my knowledge is limited. 

source from which I took inspiration: https://raw.githubusercontent.com/kkossev/Hubitat/development/Drivers/Tuya%20Multi%20Sensor%204%20In%201/Tuya%20Multi%20Sensor%204%20In%201.groovy 

search TS0225_LINPTECH_RADAR inside the file.

 "// LINPTECH / MOES are using a custom cluster 0xE002 for the settings (except for the fadingTime), ZCL cluster 0x0400 for illuminance (malformed reports!) and the IAS cluster 0x0500 for motion detection"

 and of course zigbee2mqtt: https://github.com/Koenkk/zigbee2mqtt/issues/18637
 there are screenshots and valuable information on the data to read :-)
[ts0225_presence.zip](https://github.com/zigpy/zha-device-handlers/files/14984184/ts0225_presence.zip)

rafmet, 2024-04-23:
Hi,

I don't have any experience writing zigpy quirks and at the beginning it seemed to be way above my head but as a complete beginner I was able to write a partially working quirk for TZ3218/TS0225.
First I tried to write V1 quirk but it was very difficult to expose sensor values as entities in Home Assistant. Then I tried V2 and they look a little simpler but documentation is poor and there are almost no examples since V2 quirks are quite a new idea.
The Zigbee2mqtt converter was helpful as a source of device details.
What is working:
- target_distance
- setting motion_detection_sensitivity, static_detection_sensitivity and motion_detection_distance (with problems)
What is not working:
- luminance
- setting sensitivity or detection distance returns error to home assistant but ... it works
- initial sensor names in home assistant are trashy

When setting sensitivity or detection distance I get UNSUPPORTED_ATTRIBUTE

[tz3218_ts0225.zip](https://github.com/zigpy/zha-device-handlers/files/15076168/tz3218_ts0225.zip)

I would be grateful for any help/feedback

mool, 2024-04-23:
Hi, I'm also totally new with quirks but yesterday I was able to get this kind of working.... at least I was able to set the `motion_detection_distance` to reduce the distance where the sensor was detecting presence.

https://gist.github.com/mool/d4fb96eaed3d04b9bdbf1d7c1517141c

kaiserkiwi, 2024-05-13:
As this would be my first quirk, how the heck do they work? 
I found many things online to use them but the last bit is always missing.

I created `/config/custom_zha_quirks`. 
I added the file from @mool.
I added the zha config to my `configuation.yaml`
```yaml
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks
```
I restarted and got the "Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers" log message.

But how do I apply the quirk? There's no option and it doesn't get applied automatically but with `TS0225` and `_TZ3218_awarhusb` the identifier should match.

What am I missing? 😅

rafmet, 2024-05-13:
> I restarted and got the "Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers" log message.

Good news - Home Assistant noticed the quirks :-)
 
> But how do I apply the quirk? There's no option and it doesn't get applied automatically but with `TS0225` and `_TZ3218_awarhusb` the identifier should match.

There is no need to apply quirk in any way. When you go to device settings you should see line "Quirk: ....." in Zigbee info.
If there is no quirk info then probably identifiers of quirk and your device differs.

felafel, 2024-05-13:
TL;DR I am using Zigbee2MQTT (Z2M) and it is working well including luminance.

For those that claimed my last post was "off-topic" or venting I think they should read the whole thread.  Reading the one post is not enough.  At that time I posted the most including these types of progressive posts [does anyone have a solution] [I tried this and maybe someone can help] [does anyone have suggestions on getting this to work].  The answers were "no idea" or crickets.

In addition, there was more missing context.  I am a real software developer and I can easily get HA to work without Nabu Casa.  I subscribe to Nabu Casa (in my second year) and I bought the SkyConnect.  I am not dumping on HA or any devs that are trying to make things better.

I think z2m works well.  I do not see the point in recreating the wheel.  I think the ZHA devs may be more effective making z2M better.  However, if they really want to duplicate something that works; one thing I think is very much missing is quirk functionality.  See above posts about this as well.  Is the quirks folder being used?  Who knows.  Is the device trying a quirk but it does not work? Who knows.

Have fun and I will unsubscribe since this thread is no longer applicable to me.

kaiserkiwi, 2024-05-13:
> There is no need to apply quirk in any way. When you go to device settings you should see line "Quirk: ....." in Zigbee info. If there is no quirk info then probably identifiers of quirk and your device differs.

That's so weird. It's right there.
The script:
[image]
The device:
[image]

Restarted multiple times now and even made an Home Assistant update just to be sure. But nope. Doesn't get detected. :/

rafmet, 2024-05-13:
> Restarted multiple times now and even made an Home Assistant update just to be sure. But nope. Doesn't get detected. :/

I don't know German but as I guess "Eigenart" means "Quirk" so the quirk is applied -  ts0225_linptech is the name of the file and TS0225Radar is the name of the class.

kaiserkiwi, 2024-05-14:
Oh well… I never thought someone would translate this as we use this word in this context here too, as no direct translation would be possible. 🙃

Then it's applied but doesn't give me anything extra. Thanks for your help. I guess future quirks won't be such a headache for me.

rafmet, 2024-05-14:
> Oh well… I never thought someone would translate this as we use this word in this context here too, as no direct translation would be possible. 🙃
> 
> Then it's applied but doesn't give me anything extra. Thanks for your help. I guess future quirks won't be such a headache for me.

It probably gives you possibility to set detection distance through 'Manage Zigbee Device' / selecting cluster / selecting attribute / write atrribute. Not very user friendly. As I understand it was one of the problems with "old" quirks. That's why I started to try with v2. You can try the quirk I've uploaded earlier in this thread.

comperta, 2024-05-14:
thank you @rafmet , I tried your file, and it works for me. However I have stability problems when I use it.

rafmet, 2024-05-17:
> thank you @rafmet , I tried your file, and it works for me. However I have stability problems when I use it.

I guess the instability you mentioned is related to the error when setting sensitivity or distance. Unfortunately still don't know how to fix it or at least somehow ignore error :-( v2 quirks are still new so I can't find many examples to learn from and documentation is poor :-(

thermseekr, 2024-08-14:
Hi there, any developments on this? I'm selecting sensors to buy for testing for a new project. One that combines mmWave and luminosity would be great.

cchobanIL, 2024-09-06:
OK all... I have the  _TZ3218_awarhusb TS0225.

I've successfully run a Tuya quark for a garage door opener for two years or so, so I know I have folders and process set up correctly.  I've tried every single quark that's posted in this conversation, and the only that even pics up and recognizes in the config is the one from Comperta.

Has anyone gotten one of these actually working in ZHA, or no?

Fantastic effort, I'll say that!

Thanks in advance!

sierralx, 2024-10-10:
Any update to make it work on ZHA? I only see IAS Zone

kaiserkiwi, 2024-11-22:
I wonder if this will ever work.

Is there a comparable presence sensor that works correctly with ZHA? Please don't recommend the Every Presence Lite. I have one and it's extremely unreliable.

closed 2025-02-24 by TheJulianJES

kaiserkiwi, 2025-02-24:
Is this possible? 
When will we see the change reflected in Home Assistant?

kaiserkiwi, 2025-03-06:
What does the quirk do then when it has no luminance? 😅

rafmet, 2025-03-06:
@kaiserkiwi it allows setting detection distance, both sensitivities and fading time. It also provides target distance

PR #3884 (2025-02-24): Add Tuya ZG-205Z/A human presence sensor `_TZE200_2aaelwxk`
## Proposed change

Adds Tuya ZG-205Z/A, 5.8Ghz/24Ghz Human presence sensor _TZE200_2aaelwxk, TS0225.

## Additional information

Closes: #2599
Closes: #2551

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works