zigpy/zha-device-handlers #2600: [Device Support Request] TS0601 Zigbee Motor blinds by HILADUO

bwrsandman, 2023-09-21:
### Problem description

Managed to pair with [HILADUO motorized blinds](https://www.amazon.ca/dp/B0976BVKVS?ref=ppx_yo2ov_dt_b_product_details&th=1), however they show up without any controls.

Normally they should be pairable with the Tuya app.

Unfortunately I do not have more information than that. I can however, probably look it up with instructions.

### Solution description

The device should give normal controls for bringing the blinds up, down and battery level.
If possible it should also have options for setting the height and tilt.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_9p5xmj5r",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.8.1",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "***",
    "os_name": "Linux",
    "os_version": "5.15.107+truenas",
[... 264 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I'm decent with python and electronics but not too much with ZHA. With some tips, I can go a long way towards contributing a fix.

cassidyjames, 2023-09-23:
I also just received a set of these and for some reason had thought I had seen people saying they worked ootb with Home Assistant. Alas, same issue here: they pair but the controls don't appear.

If there is any further debugging information I can provide, please let me know.

TheJulianJES, 2023-09-23:
Should be supported with:
- https://github.com/zigpy/zha-device-handlers/pull/2593

in Home Assistant 2023.10.0

You can probably use that file as a custom quirk for now.

cassidyjames, 2023-09-25:
That works great, thank you!

bwrsandman, 2023-10-05:
I updated to 2023.10.0 and it indeed now has the blind controls.
It doesn't report battery, do we know if the device does this?

closed 2024-04-03 by bwrsandman