zigpy/zha-device-handlers #2601: [Device Support Request] Tuya Thermostat (_TZE200_viy9ihs7)

Lesterpig, 2023-09-21:
### Problem description

Using this device: https://www.amazon.fr/gp/product/B0CCXRC7VK/

Pairing OK but no attribute available.

### Solution description

Expecting at least room temperature and set point temperature, ideally support for programs, temperature calibration, backlight configuration.

### Device signature

<details><summary>Device signature</summary>

```json
"signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_viy9ihs7",
      "model": "TS0601"
    },
```

</details>

### Additional information

I found the following work on this device, and I am currently trying to develop a quirk.

https://github.com/Koenkk/zigbee2mqtt/issues/18589

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L3821C5-L3888C7

Lesterpig, 2023-09-22:
I have started some work, inspired by https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto.py

I found the following attributes for this device:

- 0x0101: [0] if powered-off, [1] if powered-on
- 0x0202: [0, 0, 0, 210] target temperature
- 0x0203: [0, 0, 0, 215] room temperature
- 0x0404: [0] if mode is "schedule", [1] if mode is "manual"
- 0x0465: [0] if not heating, [1] if heating

jacekk015, 2023-10-14:
@Lesterpig Test from https://github.com/jacekk015/zha_quirks
quirk file:
ts0601_thermostat_zwt198.py

ranko-ha, 2023-10-16:
Hi @jacekk015,

received my ZWT198 last week and couldn't find a quirk, so I constructed one using attributes @Lesterpig published here.
I based it on your code (ts0601_thermostat_Avvato.py) and ended up with something very similar to the quirk you published on Saturday :)

I can confirm that basic thermostat functionality works fine!

However, CHILD_LOCK and TEMP_CALIBRATION don't seem to do anything.
(I used the attribute values from your Avvato quirk, tried also with the new ones you used for ZWT198.)

Not sure if these attribute values are correct… How do you figure these values out anyway?
(I have zero experience with ZHA clusters etc) 

Would be nice if we could somehow figure out the values for other functions too (like frost_protection or battery_percentage_remaining)

I found a Z2M description in zigbee-herdsman-converters with info on additional ZWT198 "Datapoints" but don't know how to use/interpret this info.
[https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L3916C1-L3984C1](url)

ranko-ha, 2023-10-16:
Hi @jacekk015,                  
Standard out-of-the-box functionality was zero entities for ZWT198 in HA, so great to have a quirk.
Thanks a lot for your work! I learned a lot and now have a much better understanding of ZHA.

Don't really care so much about the extra stuff, it is just that I saw that Z2M-Docu so thought it could somehow be usable for us in ZHA too. :)

BTW:
I based my code on your avatto quirk so noticed a difference in how you defined ENDPOINT 2.
Does the zwt version use the proper class in OUTPUT_CLUSTERS (Endpoint references just the class name)?
Avatto version uses a different name so there is no doubt there...

**ts0601_thermostat_zwt198.py:

class ZWT198UserInterface(TuyaUserInterfaceCluster):
...

class ZWT198UserInterface(CustomTuyaOnOff):
...

            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [ZWT198UserInterface],
                OUTPUT_CLUSTERS: [],
            },

**ts0601_thermostat_avvato.py:

class AvattoUserInterface(TuyaUserInterfaceCluster):
...

class AvattoChildLock(CustomTuyaOnOff):
...

            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [AvattoChildLock],
                OUTPUT_CLUSTERS: [],
            },

pascalwhoop, 2024-01-05:
Just wanted to chime in that there has been some progress on this device by zigbee2mqtt community
https://github.com/Koenkk/zigbee-herdsman-converters/pull/6155

some more discussion happening here

https://github.com/Koenkk/zigbee2mqtt/issues/19351

any chance we can take those insights and bring them in to HA directly?

closed 2024-08-25 by github-actions[bot]