zigpy/zha-device-handlers #2604: [Device Support Request] Yushun YS-MT750L Curtain motor (TS0601 _TZE200_gaj531w3)

jenko1080, 2023-09-23:
### Problem description

Request to integrate new Cover MODELS_INFO in `ts0601_cover.TuyaMoesCover0601` class for `YS-MT750L` curtain motor.

### Solution description

Expecting the following controls:

1. Open Cover
2. Stop Cover
3. Close Cover

Existing quirk appears compatible: `ts0601_cover.TuyaMoesCover0601` (Tested with custom quirk)

Model information available on Zigbee2MQTT: <https://www.zigbee2mqtt.io/devices/YS-MT750L.html>

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

(Before custom quirk)

```json
{
        "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
        "endpoints": {
            "1": {
                "profile_id": "260",
                "device_type": "0x0051",
                "input_clusters": ["0x0000", "0x0004", "0x0005", "0xef00"],
                "output_clusters": ["0x000a", "0x0019"]
            }
        },
        "manufacturer": "_TZE200_gaj531w3",
        "model": "TS0601",
        "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.8.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
[... 338 lines cut ...]
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

javicalle, 2023-09-24:
If you have tested the quirk in your device and it's working as expected, do you mind to create the PR?
You can do it just editing the file from your browser and follow the indications to create the pull request:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_cover.py

Thanks in advance.

jenko1080, 2023-09-25:
Tested in final install, took a bit to get around to the PR- Google Home integration appears to invert the direction requested but via the Home Assistant UI "open" means "open".

javicalle, 2023-09-25:
Are the covers connected to any physical device (by wires)? If they are, have you checked the motor direction from there?

PR #2605 (2023-09-26): Add Tuya curtain motor Yushun YS-MT750L `_TZE200_gaj531w3`
Include model info for Yushun YS-MT750L curtain motor

## Proposed change
Include `MODELS_INFO` for matching motor.

## Additional information

Fixes #2604 

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2023-09-26 by TheJulianJES