zigpy/zha-device-handlers #2606: Why lqi value in device info  is different from it in topology?

strawberryhcy, 2023-09-25:
[image]
[image]

I know what these two lqi values represent in the topology.But i want to know it in the device info.Why two device (coordinater and router)have 3 lqi value?

MattWestb, 2023-09-25:
The topology is done with requesting all routers neighbors and the links to them.
The device cars is made from the LQI that the coordinator is getting with the package revived from the network and its depends of the rout its taken and if end devices is direct connected to the coordinator with  EZSP you is very likely getting 255 in most firmware versions (best is blocking the coordinator having direct children for getting more stable network).

strawberryhcy, 2023-09-25:
> The topology is done with requesting all routers neighbors and the links to them. The device cars is made from the LQI that the coordinator is getting with the package revived from the network and [...]
I want to know why the values on the page are different from those in the topology diagram. As far as I know, the value of lqi comes from two places, but it has three values that are different from each other.

strawberryhcy, 2023-09-25:
I want to know why the values on the page are different from those in the topology diagram. As far as I know, the value of lqi comes from two places, but it has three values that are different from each other.

Shulyaka, 2023-10-05:
As @MattWestb already described, they come from 3 different sources. For the topology, the two LQI between the routers are requested from the routers themselves. For the device settings, the LQI is provided by the coordinator firmware and, depending on the firmware, could actually be the LQI of the last hop (i.e. between the coordinator and the nearest router)

strawberryhcy, 2023-10-05:
> As @MattWestb already described, they come from 3 different sources. For the topology, the two LQI between the routers are requested from those routers. For the device info, the LQI is provided by the coordinator firmware and, depending on the firmware, could actually be the LQI of the last hop (i.e. between the coordinator and the nearest router)

Thank you for your answer. But in the connection between the router and coordinator, the neighbor table only has one value, and to my knowledge, the other value is the connection quality of the last message determined by the coordinator firmware 
   https://github.com/dmulcahey/zha-network-visualization-card/issues/46        So, I still feel puzzled about this.

Shulyaka, 2023-10-05:
Every value in the topology graph comes from the neighbor tables. It could be the neighbor table of the coordinator itself if the coordinator supports loopback ZDO frames. Two devices would give two values for the link between them. See https://github.com/zigpy/zigpy/blob/50bb2171341bd44239898cfd3dd77b22ac49a03e/zigpy/appdb.py#L332

And the device info (and the zha network card) comes from the coordinator firmware with the last received packet, see
https://github.com/zigpy/zigpy/blob/50bb2171341bd44239898cfd3dd77b22ac49a03e/zigpy/application.py#L1008

I would not rely on the value in the device info because it is only for the last hop (and you would probably be more interested in the first hop).

MattWestb, 2023-10-05:
@Shulyaka What is the LQI / RSSI coming for the https://github.com/dmulcahey/zha-network-card ?
I think its its from the nigbur scan from all routers in the network then i have seen it being updated then clicking update and sniffing the network.

Shulyaka, 2023-10-05:
It comes from the device attributes (i.e. from the coordinator firmware), not the neighbor scan.

dmulcahey, 2023-10-05:
https://github.com/home-assistant/frontend/blob/f43171f91c51a354fb0d7f8d7f152ab95e8fc16a/src/panels/config/integrations/integration-panels/zha/zha-network-visualization-page.ts#L248

The topology in HA uses the neighbor tables.

dmulcahey, 2023-10-05:
The table is last hop set by zigpy on a per message basis. Neighbors are not used to build the table.

closed 2024-04-09 by github-actions[bot]