zigpy/zha-device-handlers #2607: [Device Support Request] Tuya Smoke detector TS0205

erkr, 2023-09-26:
### Problem description

I bought a few very affordable smoke detectors via AliExpress, They pair fine using ZHA (conbee2) and identify as a TS0205 by _TZ3210_up3pngle:
[image]

Stock, they pairs without a quirk and gets recognized as a motion detector, with two IAS sensors (opening and motion detection).
I tried to make a custom quirk, based on the smoke detector TS0601. This custom quirk now only changes the the input IasZone.cluster_id by TuyaIasZone what provides the minimum needed:
- Recognized as a smoke detector. 
- When I press the test button, the detection is received correctly.

According to the tuya description, it should be possible to mute the smoke detector via the app, but a mute button/switch for that is still missing.

### Solution description

Please have a look at my custom quirk and if possible, help me to add the mute button.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0500"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3210_up3pngle",
  "model": "TS0205",
  "class": "ts0205_smoke.TuyaSmokeDetector0205"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.3",
    "dev": false,
[... 505 lines cut ...]
            1: {
                PROFILE_ID: zigpy.profiles.zha.PROFILE_ID,
                DEVICE_TYPE: zigpy.profiles.zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaIasZone,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Time.cluster_id,
                    OnOff.cluster_id,
                    Ota.cluster_id,
                    LightLink.cluster_id, 
                ],
            },
        },
    }

```

</details>

Aleksandar896, 2023-11-21:
Also interested in this.

Krabalex, 2023-11-27:
I also bought this smoke sensor, and for it to work I had to change ZHA to zigbee2mqtt, you can try too

erkr, 2023-11-27:
> > I also bought this smoke sensor, and for it to work I had to change ZHA to zigbee2mqtt, you can try too
> 
> change to zigbee2mqqtt?? i have more devices that i can Trow a stick at it about 108 devices 680 entities and about 1000 automations...

Too much work ...

pabsi, 2023-12-11:
One of the biggest issues for me with this one is to not have the battery level being reported. Only whether smoke has been detected or not. Luckily I know that it battery is running out it'll just start playing a beep every few minutes...but still, would be nice to have this directly on HA.

jmuf, 2023-12-11:
Per [TUYA Smokedetector spec](https://developer.tuya.com/en/docs/iot/tuya-zigbee-smoke-sensor-access-standard?id=K9ik6zvm2tg0l#title-11-Special%20instructions), both battery status and "tamper" get sent along with the smoke detection attribute:
* Alarm1 + bit (2): tamper alarm.
* Alarm1 + bit (3): low voltage alarm.
and would need to be handled as such in the above quirk (i.e route to/update the corresponding generic attribute)

However, we cannot get battery percentages from this, just "ok" and "low". Not sure how to map this on ZHA ..(fixed "90%" and "5%" ?)

pabsi, 2023-12-11:
> Per [TUYA Smokedetector spec](https://developer.tuya.com/en/docs/iot/tuya-zigbee-smoke-sensor-access-standard?id=K9ik6zvm2tg0l#title-11-Special%20instructions), both battery status and "tamper" get [...]

Weird, as per https://developer.tuya.com/en/docs/iot/tuya-zigbee-smoke-sensor-access-standard?id=K9ik6zvm2tg0l#subtitle-7-Power%20configuration the device should report the exact battery level?

Sorry, perhaps my lack of knowledge about Zigbee and clusters and attributes is making this harder than it should.

Thanks

harrlih, 2023-12-20:
Yes I also would like to have this integrated...
I thought I would buy "_TZE204_ntcy3xu1" but had to learn I received "_TZ3210_up3pngle" instead (same casing, different content).
Quirk from here: https://github.com/zigpy/zha-device-handlers/issues/1838  which perfectly supports _TZE204_ntcy3xu1 does not work, when I add _TZ3210_up3pngle to the list. Well I don't know if this is allowed, as the base type? seems to be different (not sure what TS0205 vs TS0601 really means...)

fbonelle, 2023-12-26:
Hello,

I have received the same model. (TS0205 - _TZ3210_up3pngle- I have tried a lot of things without success. @harrlih, have you found any solution ?

Regards

renekuehl, 2023-12-29:
> ### Problem description
> 
> I bought a few very affordable smoke detectors via AliExpress ( https://a.aliexpress.com/_Eue8kl5), They pair fine using ZHA (conbee2) and identify as a TS0205 by _TZ3210_up3pngle: ![image](https://private-user-images.githubusercontent.com/4363599/270748907-cb4306fb-f041-458e-9837-ffe9f168d7f3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4NjI0NjUsIm5iZiI6MTcwMzg2MjE2NSwicGF0aCI6Ii80MzYzNTk5LzI3MDc0ODkwNy1jYjQzMDZmYi1mMDQxLTQ1OGUtOTgzNy1mZmU5ZjE2OGQ3ZjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjlUMTUwMjQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjQyMTkxNjMyMzU1OGFkNDU1MmJiODVhNTMyMWRiYzZjMGMxODU2NTIwNGU5YzJlZDMwNDdiMGQzMGU2NjRhOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.jGJ2hytgcf4CWpNplh7adPevCQg5tKslZ9AZP-9SQvc)
> 
> Stock, they pair without a quirk and gets recognized as a motion detector, with two IAS sensors (opening and motion detection). I tried to make a custom quirk, based on the smoke detector TS0601. This custom quirk now only changes the the input IasZone.cluster_id by TuyaIasZone what provides the minimum needed:
> 
>     * Recognized as a smoke detector.
> 
>     * When I press the test button, the detection is received correctly.
> 
> 
> According to the tuya description, it should be possible to mute the smoke detector via the app, but a mute button/switch for that is still missing.
> 
> Also the opening sensor, is not working, presuming that is the tampering switch
> ### Solution description
> 
> Please have a look at my custom quirk and if possible, help me to add the mute button.
> ### Screenshots/Video
> Screenshots/Video
> ### Device signature
> Device signature
> ### Diagnostic information
> Diagnostic information
> ### Logs
> Logs
> ### Custom quirk
> Custom quirk
> ### Additional information
> 
> _No response_

Just wanted to say "Thank you for the custom quirk". I received two of these for testing today and I was seeing that they were identified as Motion Sensors under ZHA. Thanks to your Code they are now seen as Smoke Detectors. Again a big Thank you ;)

[image]

jmuf, 2023-12-29:
Hm, I am not convinced the current "quirk" does much (except possibly to change the device type to `ZoneType.Fire_Sensor`).
In particular, the part that would "route" unknown attributes
```python
class TuyaSmokeDetectorCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the TS0601 smoke detector."""
    attributes = {
        TUYA_SMOKE_DETECTED_ATTR: ("smoke_detected", t.uint8_t, True),
    }
    def _update_attribute(self, attrid, value):
[..]
```
doesn't seem to be triggered at all, since the device already reports itself as a `IasZone` (and not a `TuyaManufCluster`)
```python
  INPUT_CLUSTERS: [
[..]
                    IasZone.cluster_id,
```

Instead, we directly get updated attributes into our replacement class.
With a few more debug statements:
```
2023-12-29 17:29:05.087 WARNING (MainThread) [ts0205_smoke] [0xbc9a:1:0x0500] smoke quirk TuyaIasZone received _update_attribute: 0x0002=0x0004
2023-12-29 17:29:08.523 WARNING (MainThread) [ts0205_smoke] [0xbc9a:1:0x0500] smoke quirk TuyaIasZone received _update_attribute: 0x0002=0x0000
2023-12-29 17:30:31.359 WARNING (MainThread) [ts0205_smoke] [0xbc9a:1:0x0500] smoke quirk TuyaIasZone received _update_attribute: 0x0001=0x0028
```
these would correspond to "tamper alarm" 0x4 (i.e me taking the thing off the wallmount), "normal" 0x0" (put it back..), "low voltage alarm"+"device fault alarm" 0x8 + 0x20 (no idea - pressing the test button?)

Derchristopher12, 2023-12-30:
> > ### Problem description
> > I bought a few very affordable smoke detectors via AliExpress ( https://a.aliexpress.com/_Eue8kl5), They pair fine using ZHA (conbee2) and identify as a TS0205 by _TZ3210_up3pngle: ![image](https://private-user-images.githubusercontent.com/4363599/270748907-cb4306fb-f041-458e-9837-ffe9f168d7f3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4NjI0NjUsIm5iZiI6MTcwMzg2MjE2NSwicGF0aCI6Ii80MzYzNTk5LzI3MDc0ODkwNy1jYjQzMDZmYi1mMDQxLTQ1OGUtOTgzNy1mZmU5ZjE2OGQ3ZjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjlUMTUwMjQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjQyMTkxNjMyMzU1OGFkNDU1MmJiODVhNTMyMWRiYzZjMGMxODU2NTIwNGU5YzJlZDMwNDdiMGQzMGU2NjRhOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.jGJ2hytgcf4CWpNplh7adPevCQg5tKslZ9AZP-9SQvc)
> > Stock, they pair without a quirk and gets recognized as a motion detector, with two IAS sensors (opening and motion detection). I tried to make a custom quirk, based on the smoke detector TS0601. This custom quirk now only changes the the input IasZone.cluster_id by TuyaIasZone what provides the minimum needed:
> > ```
> > * Recognized as a smoke detector.
> > 
> > * When I press the test button, the detection is received correctly.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > According to the tuya description, it should be possible to mute the smoke detector via the app, but a mute button/switch for that is still missing.
> > Also the opening sensor, is not working, presuming that is the tampering switch
> > ### Solution description
> > Please have a look at my custom quirk and if possible, help me to add the mute button.
> > ### Screenshots/Video
> > Screenshots/Video
> > ### Device signature
> > Device signature
> > ### Diagnostic information
> > Diagnostic information
> > ### Logs
> > Logs
> > ### Custom quirk
> > Custom quirk
> > ### Additional information
> > _No response_
> 
> Just wanted to say "Thank you for the custom quirk". I received two of these for testing today and I was seeing that they were identified as Motion Sensors under ZHA. Thanks to your Code they are now seen as Smoke Detectors. Again a big Thank you ;)
> 
> ![Bildschirmfoto vom 2023-12-29 16-01-25](https://private-user-images.githubusercontent.com/5886498/293385466-18af8a83-a690-4b5f-9f31-aa0ecd5ca18f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM5MzcxOTYsIm5iZiI6MTcwMzkzNjg5NiwicGF0aCI6Ii81ODg2NDk4LzI5MzM4NTQ2Ni0xOGFmOGE4My1hNjkwLTRiNWYtOWYzMS1hYTBlY2Q1Y2ExOGYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMzBUMTE0ODE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDJlMDM1ZDM0MjUwMzc1MWUxODlhNTZiM2JiZWZlMzk2N2I1MDViNGZiOWEyYjFkZTM3NTEwNDE4MzEzY2Y0NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.lwPYm8zz9rr9xhFR6mZLqpHFoKWiwdC2VV-9wcs0Rc0)

can you send the code from the ZHA Quirk? No one i tried worked for me. Thank you :)

renekuehl, 2023-12-30:
> > > ### Problem description
> > > I bought a few very affordable smoke detectors via AliExpress ( https://a.aliexpress.com/_Eue8kl5), They pair fine using ZHA (conbee2) and identify as a TS0205 by _TZ3210_up3pngle: ![image](https://private-user-images.githubusercontent.com/4363599/270748907-cb4306fb-f041-458e-9837-ffe9f168d7f3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4NjI0NjUsIm5iZiI6MTcwMzg2MjE2NSwicGF0aCI6Ii80MzYzNTk5LzI3MDc0ODkwNy1jYjQzMDZmYi1mMDQxLTQ1OGUtOTgzNy1mZmU5ZjE2OGQ3ZjMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjlUMTUwMjQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjQyMTkxNjMyMzU1OGFkNDU1MmJiODVhNTMyMWRiYzZjMGMxODU2NTIwNGU5YzJlZDMwNDdiMGQzMGU2NjRhOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.jGJ2hytgcf4CWpNplh7adPevCQg5tKslZ9AZP-9SQvc)
> > > Stock, they pair without a quirk and gets recognized as a motion detector, with two IAS sensors (opening and motion detection). I tried to make a custom quirk, based on the smoke detector TS0601. This custom quirk now only changes the the input IasZone.cluster_id by TuyaIasZone what provides the minimum needed:
> > > ```
> > > * Recognized as a smoke detector.
> > > 
> > > * When I press the test button, the detection is received correctly.
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > According to the tuya description, it should be possible to mute the smoke detector via the app, but a mute button/switch for that is still missing.
> > > Also the opening sensor, is not working, presuming that is the tampering switch
> > > ### Solution description
> > > Please have a look at my custom quirk and if possible, help me to add the mute button.
> > > ### Screenshots/Video
> > > Screenshots/Video
> > > ### Device signature
> > > Device signature
> > > ### Diagnostic information
> > > Diagnostic information
> > > ### Logs
> > > Logs
> > > ### Custom quirk
> > > Custom quirk
> > > ### Additional information
> > > _No response_
> > 
> > 
> > Just wanted to say "Thank you for the custom quirk". I received two of these for testing today and I was seeing that they were identified as Motion Sensors under ZHA. Thanks to your Code they are now seen as Smoke Detectors. Again a big Thank you ;)
> > ![Bildschirmfoto vom 2023-12-29 16-01-25](https://private-user-images.githubusercontent.com/5886498/293385466-18af8a83-a690-4b5f-9f31-aa0ecd5ca18f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM5MzcxOTYsIm5iZiI6MTcwMzkzNjg5NiwicGF0aCI6Ii81ODg2NDk4LzI5MzM4NTQ2Ni0xOGFmOGE4My1hNjkwLTRiNWYtOWYzMS1hYTBlY2Q1Y2ExOGYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMzBUMTE0ODE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDJlMDM1ZDM0MjUwMzc1MWUxODlhNTZiM2JiZWZlMzk2N2I1MDViNGZiOWEyYjFkZTM3NTEwNDE4MzEzY2Y0NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.lwPYm8zz9rr9xhFR6mZLqpHFoKWiwdC2VV-9wcs0Rc0)
> 
> can you send or link the code from the ZHA Quirk? No one i tried worked for me. Thank you :)

The Code is available under "Custom Quirk" in the first Post ;) That's the Code I used ...

erkr, 2024-01-10:
Please check if the quirk got loaded for the SD

kruegertom, 2024-01-10:
btw: is there meanwhile any solution for the other IAS-ZoneStatus attributes like Tamper or Battery?

Thorsten1982, 2024-02-08:
I would like to use the custom quirk and have understood most of what needs to be done. It's just not clear to me what the file with the custom quirk should be called, I think this doesn't matter so that it can be assigned to the device?

eyk107, 2024-02-08:
the name doesn't matter in my opinion. You only have to add the following line to your configuration:
zha:
  custom_quirks_path: config/custom_zha_quirks/

eyk107, 2024-02-08:
@erkr thank you very much, works great. One last question: Why is the battery level not listed in the sensors overview but in Diagnostics?
[image]

erkr, 2024-02-08:
>Why is the battery level not listed in the sensors overview but in Diagnostics?

ZHA always puts battery and identity under diagnostics. Nice it works for you!

Thorsten1982, 2024-02-11:
Somehow it doesn't work, I don't know why. Have

I created a directory and inserted a file there with and the Customk Quirk code from the post.
[image]
[image]

I added the following code to the Configuartion.yaml

`zha:
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
      ikea_provider: true
      inovelli_provider: true
      ledvance_provider: true
      salus_provider: true
      sonoff_provider: true
      thirdreality_provider: true
  custom_quirks_path: /config/zha_quirks/`

I then re-added the device but the custom quirk is not loaded.
[image]

erkr, 2024-02-11:
Should not make a difference, but I explicitly added `  enable_quirks: true`

eyk107, 2024-02-11:
I have to use the following to get it to work:
custom_quirks_path: config/zha_quirks/

erkr, 2024-02-11:
I have a slash before config. That works fine.

MikeInMaine, 2024-02-13:
Seems to work. Thanks!

Thorsten1982, 2024-02-13:
Strange, I've tried all the tips now, but it just doesn't work.

erkr, 2024-02-13:
> tips

Your configuration looks fine to me. You can try to remove and re- pair the smoke detector.

Thorsten1982, 2024-02-13:
I've already tried it several times

erkr, 2024-02-13:
@Thorsten1982 
- Nothing in your log file? 
- Is the quirk file in the right format (utf-8 only LF, no CR)?
- If you make a deliberate syntacs error in the quirk file, you should see errors logged. If that happens you know your config to load custom quirks is ok.

I'm just a user running out of ideas ;-)

Thorsten1982, 2024-02-13:
I have found this information in the log:

2024-02-13 06:36:44.985 DEBUG (MainThread) [zigpy.quirks.registry] Checking
quirks for _TZ3210_up3pngle TS0205 (a4:c1:38:72:4b:be:8f:ca)
2024-02-13 06:36:44.985 DEBUG (MainThread) [zigpy.quirks.registry]
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
2024-02-13 06:36:44.985 DEBUG (MainThread) [zigpy.quirks.registry] Fail
because endpoint list mismatch: {232, 230} {1}
2024-02-13 06:36:44.985 DEBUG (MainThread) [zigpy.quirks.registry]
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
2024-02-13 06:36:44.985 DEBUG (MainThread) [zigpy.quirks.registry] Fail
because endpoint list mismatch: {232, 230} {1}
2024-02-13 06:36:44.985 DEBUG (MainThread) [zigpy.quirks.registry]
Considering <class
'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry] Fail
because device_type mismatch on at least one endpoint
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry]
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry] Fail
because device_type mismatch on at least one endpoint
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry]
Considering <class
'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry] Fail
because input cluster mismatch on at least one endpoint
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry]
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry] Fail
because input cluster mismatch on at least one endpoint
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry]
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
2024-02-13 06:36:44.986 DEBUG (MainThread) [zigpy.quirks.registry] Fail
because endpoint list mismatch: {11, 13} {1}

erkr, 2024-02-13:
That looks like the matching of available quirks against devices. I don't see the custom quirk class  `TuyaSmokeDetector0205` mentioned. So no evidence your custom quirk was loaded.

DEVAXTATOR, 2024-02-13:
im just putting my two cents
first in configuration.yaml
you have to put
*zha:*
*  enable_quirks: true*
*  custom_quirks_path: /config/custom_zha_quirks/*
then create the folder and create the file
[image: image.png]
the code is in this thread
then reboot ha
then delete and add the device again
[image: image.png]
boom shakalaka!!

MikeInMaine, 2024-02-13:
In my HA docker install the leading "/" needs to be left off  of "
custom_quirks_path: /config/custom_zha_quirks/", hence it reads "
custom_quirks_path: config/custom_zha_quirks/. I expect this is because I
haven't done the necessary mapping between the docker instance and the OS.
BUt I prefer the config path to be inside the docker home directory.

Thorsten1982, 2024-02-14:
Thanks for the many tips, it works now. I recreated the Custom Quirk file again. Now it's working

cdalexndr, 2024-02-19:
Opened PR https://github.com/zigpy/zha-device-handlers/pull/2991

closed 2024-08-24 by github-actions[bot]

bjornwiberg, 2025-01-10:
mine is getting the wrong quirk and also missing the battery and have the opening sensor which was removed in the pr #2991, help appreciated :)

[image]
[image]

Dejwis218, 2025-02-02:
> [@erkr](https://github.com/erkr)dziękuję bardzo, działa świetnie. Ostatnie pytanie: dlaczego poziom naładowania baterii nie jest wyświetlany w przeglądzie czujników, ale w Diagnostyce? [image]

Please help me where should I create this Custom Quirkn? Where should I paste and save this code?

erkr, 2025-02-02:
Check the `Configuration Yaml` section in  [ZHA](https://www.home-assistant.io/integrations/zha/). To enable quirks you have to add something like this to your configuration.yaml:
``` 
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_quirks/
``` 
Create a custom_quirks sub-folder in your /config and put your quirks there

Dejwis218, 2025-02-03:
> For the device of this issue, no custom quirk is needed anymore (std supported today)

On the original zha integration I have it like in this post 
https://github.com/zigpy/zha-device-handlers/issues/2607#issuecomment-2581971870 
The battery is not available and the device status does not refresh.

[image]

erkr, 2025-02-03:
That looks ok. The released quirk is loaded. Looks exactly the same as mine. 
Battery updates can take a while!
Your LQI and RSSI are not updates as well.

markom84, 2025-02-21:
> > For the device of this issue, no custom quirk is needed anymore (std supported today)
> 
> On the original zha integration I have it like in this post [#2607 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2607#issuecomment-2581971870) The battery is not available and the device status does not refresh.
> 
> <img alt="Image" width="694" src="https://private-user-images.githubusercontent.com/123319042/408918871-cf2ef779-69ff-4480-a30e-e0e59527c16e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAxNjU1NjksIm5iZiI6MTc0MDE2NTI2OSwicGF0aCI6Ii8xMjMzMTkwNDIvNDA4OTE4ODcxLWNmMmVmNzc5LTY5ZmYtNDQ4MC1hMzBlLWUwZTU5NTI3YzE2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyMVQxOTE0MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02Y2Y1ZjE0YTlkZWRkNjQyMzRmMTNjMDExODcwZDZlZjRmODBmYjMwNWNhZGZmODg2Y2U1ZTRhNjk1MDQ5ZTgxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.vAP-yyWHlkAOvB51yBajJCGINJbAq6ymJv_U3vevwE0">

The same problem here, the device is properly recognised (opening, smoke, battery) but no status changes.

JaneX8, 2025-05-27:
Maybe a stupid question but what does "Opening" mean in this smoke detector? Also for me automatically Quirk CustomDeviceV2 got loaded, is that correct?

TS0205
by _TZ3210_up3pngle
Connected via Nabu Casa SkyConnect v1.0
Nwk: 0x7e50
Device Type: EndDevice
LQI: Unknown
RSSI: Unknown
Last seen: 2025-05-27T18:47:59
Power source: Battery or Unknown
Quirk: zigpy.quirks.v2.CustomDeviceV2

[image]

Possibly related: https://community.home-assistant.io/t/tuya-zigbee-smoke-detector-not-working-with-skyconnect-zbt-1-dongle/895584/2?u=00uzwz961

RcRaCk2k, 2025-11-20:
> The same problem here, the device is properly recognised (opening, smoke, battery) but no status changes.

The same for me.
How can we solve this issue?
I can see that the Transmit-LED blinking when i press the button on the back (installed to the wall, detached from the wall sensor) but on the device-page there is no change at all. Also trying to send the SMOKE DETECTED will not be refreshed in the HA-Instance.

Should i switch to Zigbee2MQTT?
Or is there a way to get it running in ZHA?