zigpy/zha-device-handlers #2615: [Device Support Request]  Nedis Motion Sensor with illuminance sensor _TZE200_s6hzw8g2

tehautanop, 2023-10-02:
### Problem description

Hi, 
I would like to have support for this PIR motion sensor/ Light sensor. 
Added it via ZHA and it does not show up much, as you can see from attached screenshot.
There are no entities created to this. System logs had something there but I cannot make anything useful of it.

Tried most of the TS0601 motion pir quirks I've found that would fit the bill, removed __pycache__ folders, restarted HA - but, with or without the quirks the device looks the same. There are no entities created etc.

Been doing this -"rinse & repeat" - for 2 solid days now.
Maybe someone can help me with this ?
Thx

-T

### Solution description

Support for this particular zigbee motion sensor / light sensor 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
"signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_s6hzw8g2",
      "model": "TS0601"
    },
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.9.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.5",
[... 308 lines cut ...]
2023-10-02 21:34:23.532 DEBUG (MainThread) [zigpy.zcl] [0xFCE5:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE200_s6hzw8g2')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
2023-10-02 21:34:23.532 INFO (MainThread) [zigpy.device] [0xfce5] Read model 'TS0601' and manufacturer '_TZE200_s6hzw8g2' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2023-10-02 21:34:23.533 INFO (MainThread) [zigpy.device] [0xfce5] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_s6hzw8g2' nwk=0xFCE5 ieee=a4:c1:38:e5:9e:72:d3:28 is_initialized=True>
2023-10-02 21:34:23.533 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_s6hzw8g2' nwk=0xFCE5 ieee=a4:c1:38:e5:9e:72:d3:28 is_initialized=True>
2023-10-02 21:34:23.533 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_s6hzw8g2 TS0601 (a4:c1:38:e5:9e:72:d3:28)
2023-10-02 21:34:23.533 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_pirmotion.PirMotion'>
2023-10-02 21:34:23.533 DEBUG (MainThread) [zigpy.quirks.registry] Fail because device_type mismatch on at least one endpoint

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

[Product link](https://nedis.com/en-us/product/safety-and-security/detectors/motion-detector/550783947/smartlife-motion-sensor-zigbee-30-battery-powered-usb-powered-detector-angle-100-0-detector-range-7-m-max-battery-life-24-months-white)

javicalle, 2023-10-03:
I would suggest to add your MODELS_INFO to the `MmwRadarMotion` quirk:
https://github.com/zigpy/zha-device-handlers/blob/74b9f043a996da6baca722d1e191d842a31ae198/zhaquirks/tuya/ts0601_motion.py#L342C7-L342C7

Once donde, repair your device and enable the debug logs. Interact with the device and attach the relevant logs.

tehautanop, 2023-10-03:
Thanks for replying !
Removed device from HA.
Added the models_info - like instructed, restarted HA.
Added the device back - all still looks exactly like earlier...
Tried the repair function, didn't do anything that would indicate "success"
Included the diagnostic info and relevant bits from system log.
I don't know where else I should look for debug logs
Thx

-T

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.9.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Helsinki",
    "os_name": "Linux",
    "os_version": "6.2.0-26-generic",
    "run_as_root": true
  },
  "custom_components": {
    "zha_toolkit": {
      "version": "v1.0.0",
      "requirements": [
        "pytz"
      ]
    },
    "ruuvi": {
      "version": "1.1.0",
      "requirements": [
        "simple-ruuvitag==0.0.8"
      ]
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
[... 677 lines cut ...]
2023-10-03 21:14:31.048 DEBUG (MainThread) [zigpy.zcl] [0xAFAE:1:0x0004] Decoded ZCL frame: Groups:Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2023-10-03 21:14:31.048 DEBUG (MainThread) [zigpy.device] [0xafae] Invalid state on future for 0x59 seq -- probably duplicate response
2023-10-03 21:14:31.048 DEBUG (MainThread) [zigpy.zcl] [0xAFAE:1:0x0004] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=91, command_id=2, *direction=<Direction.Server_to_Client: 0>)
2023-10-03 21:14:31.049 DEBUG (MainThread) [zigpy.zcl] [0xAFAE:1:0x0004] Sending request: get_membership(groups=[])
2023-10-03 21:14:31.049 DEBUG (MainThread) [zigpy.device] [0xafae] Extending timeout for 0x5b request
2023-10-03 21:14:31.049 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xAFAE), dst_ep=1, source_route=None, extended_timeout=True, tsn=91, profile_id=260, cluster_id=4, data=Serialized[b'\x01[\x02\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-10-03 21:14:31.049 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (a4:c1:38:e5:9e:72:d3:28, True)
2023-10-03 21:14:31.049 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xafae](TS0601): started initialization
2023-10-03 21:14:31.049 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:ZDO](TS0601): 'async_initialize' stage succeeded
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0x0000]: initializing cluster handler: from_cache: False
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0x0000]: finished cluster handler initialization
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0xef00]: initializing cluster handler: from_cache: False
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0xef00]: finished cluster handler initialization
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0x0019]: initializing cluster handler: from_cache: False
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0x0019]: finished cluster handler initialization
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0x0000]: 'async_initialize' stage succeeded
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0xef00]: 'async_initialize' stage succeeded
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xafae:1:0x0019]: 'async_initialize' stage succeeded
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xafae](TS0601): power source: Battery or Unknown
2023-10-03 21:14:31.050 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xafae](TS0601): completed initialization
2023-10-03 21:14:31.100 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received setExtendedTimeout: []
2023-10-03 21:14:31.100 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xafae, EmberApsFrame(profileId=260, clusterId=4, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=91), 92, b'\x01[\x02\x00')
2023-10-03 21:14:31.152 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendUnicast: [<EmberStatus.SUCCESS: 0>, 249]
```
</details>

javicalle, 2023-10-03:
From your logs I see that your HA instance is evaluating a `TZE200_s6hzw8g2.MmwRadarMotionGPP` class but the endpoints don't match:
```
2023-10-03 21:13:40.663 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'TZE200_s6hzw8g2.MmwRadarMotionGPP'>
2023-10-03 21:13:40.663 DEBUG (MainThread) [zigpy.quirks.registry] Fail because endpoint list mismatch: {1, 242} {1}
```

My suggestion was to try with the `MmwRadarMotion` (and not the GPP one).
Maybe you have taken the wrong one or maybe your HA instance is not updating the local quirks.

tehautanop, 2023-10-04:
Thx for your patience @javicalle  :1st_place_medal:    !
I paid more attention to editing this time and now re-adding, re-configuring the device - we can really see progress (see screenshots.)
Theres occupancy and illuminance entities etc.
So I guess this can be called 'done' :+1:  
Thx a lot, this was educational !

-T
[image]
[image]

javicalle, 2023-10-05:
Are the `Illuminance` and `Occupancy` sensors working?

tehautanop, 2023-10-07:
Hi,
Sorry I overlooked the other part of the functionality, as the **Occupancy** works fine. 

However, **Illuminance** default state is always in "_unknown_" - so its not working correctly.

I haven't had time to concentrate on this - but ofcourse hoping to get it working

[image]

jonasblomq, 2023-11-06:
Hi

Tried to apply this custom quirk for the exact same model, and i noticed that the occupancy sensor is the wrong way around. It will show clear when there is movement, and movement detected when there is no movement...

tehautanop, 2023-11-06:
Hi @jonasblomq - thanks for your comment !
I haven't had time to advance this any further, however looking forward having it working.

As a side note, something I overlooked was that were not getting battery level information

geleeroyale, 2024-04-21:
Thanks for all the work done here. I got some readings out of this sensor now. However, there is still:
- no Illuminance
- Occupancy is still switched

tehautanop, 2024-08-28:
Hi @geleeroyale did you ever get this fully functional ? Could you pls share the quirk

geleeroyale, 2024-08-28:
Sadly - no. I tried messing around with MQTT but that then broke my HA installation so I did not pursue it any further. The furthest I got with MQTT was that it recognized the device, but I could not read any sensor data.

Kylapaallikko, 2025-01-01:
Hi,

I was able to get the Nedis illuminance and motion sensor work with the quirk bellow. I'm not sure if it something to do Nedis sensor being (newer?) USB C variant. I used https://github.com/jatail/z2m-nedis-zbswm20wt-converter/blob/main/nedis-zbswm20wt.js as reference to write the quirk. Sensor settings doesn't work. Lux sensor isn't very sensitive so don't expect very good results. It works only on midday and if I'm pointing flashlight at it.

[image]

## The custom code:

```
import math

from zigpy.quirks.v2 import EntityType
from zigpy.quirks.v2.homeassistant import UnitOfLength, UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing

from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaIlluminanceCluster(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya Illuminance cluster."""

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

base_tuya_motion = (
    TuyaQuirkBuilder()
    .adds(TuyaOccupancySensing)
    # Common datapoints can be added here if applicable
    .adds(TuyaIlluminanceCluster)
    .skip_configuration()
)

# NEDIS-specific quirk 
# Reference https://github.com/jatail/z2m-nedis-zbswm20wt-converter/blob/main/nedis-zbswm20wt.js

(
    base_tuya_motion.clone()
    .applies_to("_TZE200_s6hzw8g2", "TS0601")
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaOccupancySensing.ep_attribute,
        attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
        converter=lambda x: x != 1,  # Inverted condition
    )
    .tuya_dp(
        dp_id=101,
        ep_attribute=TuyaIlluminanceCluster.ep_attribute,
        attribute_name=TuyaIlluminanceCluster.AttributeDefs.measured_value.name,
        converter=lambda x: 10000 * math.log10(x) + 1 if x != 0 else 0,
    )
    .tuya_battery(
        dp_id=4,
    )
    .add_to_registry()
)
```

closed 2025-08-16 by github-actions[bot]