zigpy/zha-device-handlers #2617: [Device Support Request]  Frient Electricity Meter Interface2 (EMIZB-141)

dalte-one, 2023-10-03:
### Problem description

in my setup ZHA is running with the Conbee II. I managed to pair the device well and a power sensor is created. But the values are completely wrong because my electricity meter works with 10000 imp/kWh. And I can not find the value _pulse configuration_ unfortunately.

### Solution description

it should be possible to change the pulse rate and see a normal power consumption.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4117, maximum_buffer_size=82, maximum_incoming_transfer_size=1500, server_mask=11264, maximum_outgoing_transfer_size=1500, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0xc0c9",
      "device_type": "0x0001",
      "input_clusters": [
        "0x0005",
        "0x0006"
      ],
      "output_clusters": []
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0053",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0702",
        "0x0b01",
        "0x0b05"
      ],
      "output_clusters": [
        "0x0003",
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "frient A/S",
  "model": "EMIZB-141",
  "class": "zigpy.device.Device"
}

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

"manufacturer": "frient A/S",
      "model": "EMIZB-141"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "sensor.powermeter1_battery",
        "name": "frient A/S EMIZB-141"
      },
      {
        "entity_id": "button.powermeter1_identify",
        "name": "frient A/S EMIZB-141"
      },
      {
        "entity_id": "sensor.powermeter1_instantaneous_demand",
        "name": "frient A/S EMIZB-141"
      },
      {
        "entity_id": "sensor.powermeter1_summation_delivered",
        "name": "frient A/S EMIZB-141"
      },
      {
        "entity_id": "switch.powermeter1_switch",
        "name": "frient A/S EMIZB-141"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "unknown 1 device_type of 0xc0c9 profile id"
      },
      {
        "name": "METER_INTERFACE"
      }
    ],
    "user_given_name": "Powermeter1",
    "device_reg_id": "*******************6088",
    "area_id": "keller",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "unknown",
          "id": 1
        },
        "profile_id": 49353,
        "in_clusters": {
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0006": {
            "endpoint_attribute": "on_off",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      },
      "2": {
        "device_type": {
          "name": "METER_INTERFACE",
          "id": 83
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "frient A/S"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "EMIZB-141"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0001": {
            "endpoint_attribute": "power",
            "attributes": {
              "0x003e": {
                "attribute_name": "battery_alarm_state",
                "value": 0
              },
              "0x0021": {
                "attribute_name": "battery_percentage_remaining",
                "value": 200
              },
              "0x0033": {
                "attribute_name": "battery_quantity",
                "value": 2
              },
              "0x0031": {
                "attribute_name": "battery_size",
                "value": 3
              },
              "0x0020": {
                "attribute_name": "battery_voltage",
                "value": 30
              }
            },
            "unsupported_attributes": {}
          },
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0020": {
            "endpoint_attribute": "poll_control",
            "attributes": {
              "0x0000": {
                "attribute_name": "checkin_interval",
                "value": 13200
              }
            },
            "unsupported_attributes": {}
          },
          "0x0702": {
            "endpoint_attribute": "smartenergy_metering",
            "attributes": {
              "0x0000": {
                "attribute_name": "current_summ_delivered",
                "value": 90956
              },
              "0x0304": {
                "attribute_name": "demand_formatting",
                "value": 248
              },
              "0x0302": {
                "attribute_name": "divisor",
                "value": 1000
              },
              "0x0400": {
                "attribute_name": "instantaneous_demand",
                "value": 4826
              },
              "0x0306": {
                "attribute_name": "metering_device_type",
                "value": 0
              },
              "0x0301": {
                "attribute_name": "multiplier",
                "value": 1
              },
              "0x0200": {
                "attribute_name": "status",
                "value": 0
              },
              "0x0303": {
                "attribute_name": "summation_formatting",
                "value": 248
              },
              "0x0300": {
                "attribute_name": "unit_of_measure",
                "value": 0
              }
            },
            "unsupported_attributes": {
              "0x0100": {
                "attribute_name": "current_tier1_summ_delivered"
              },
              "0x0102": {
                "attribute_name": "current_tier2_summ_delivered"
              },
              "0x0104": {
                "attribute_name": "current_tier3_summ_delivered"
              },
              "0x0106": {
                "attribute_name": "current_tier4_summ_delivered"
              },
              "0x0108": {
                "attribute_name": "current_tier5_summ_delivered"
              },
              "0x010a": {
                "attribute_name": "current_tier6_summ_delivered"
              }
            }
          },
          "0x0b01": {
            "endpoint_attribute": "meter_id",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0b05": {
            "endpoint_attribute": "diagnostic",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

KuntzeM, 2023-10-28:
Hi,
same setup and problem here.

Change the pulse configuration with ZHA Toolkit was not working, too.

themainframe, 2023-11-22:
I think this is down to Frient/Develco using the "Divisor" option to format the logged/received values rather than to influence the counting. The document describes the Divisor as being read-only, fixed at `1000`.

The pulse configuration I think is controlled by attribute `0x0300` of the metering cluster. You can write this with a ZHA Toolkit service invocation:

```yaml
service: zha_toolkit.attr_write
data:
  ieee: YOUR_IEEE_ADDRESS_HERE
  manf: 4117
  endpoint: 2
  cluster: 1794
  attribute: 768
  attr_val: 4000
```

For example to, to set the pulse rate to 4000 imp/kWh

arnili, 2023-12-21:
As soon as I even only read that endpoint using:
```
service: zha_toolkit.attr_read
data:
  ieee: <mac>
  cluster: 1794
  attribute: 768
  manf: 4117
```
The device completely switches its behaviour, reporting no units, just plain ints, usually just 0.

Something similar has also been reported here: https://community.home-assistant.io/t/frient-electricity-meter-interface2-emizb-141/619377/4

closed 2024-06-25 by github-actions[bot]