zigpy/zha-device-handlers #2619: [zigpy.zcl] [0xDE54:1:0xfeed] Unknown cluster command 42 b'

Tom-ahawk, 2023-10-03:
### Bug description

I am getting this issue spamming my log in HA:
2023-10-03 18:48:29.248 WARNING (MainThread) [zigpy.zcl] [0xDE54:1:0xfeed] Unknown cluster command 42 b''

Talked to the producer (Eva Smarthus ) of the smart plug (Apex / Datek Pop), and they say:
"It looks like a cluster which saying who is the parent, how many times restart and coverage. Its not a error, but must likely ZHA not supporting this."
"Det ser ut til at det er cluster som sender hvem som er parent, hvor mange ganger den har restartet og dekning. Det skal ikke være noe feil, men det er mulig ZHA ikke har støtte for det. "

### Steps to reproduce

Use a Apex/Eva smart plug in:
Home Assistant 2023.9.3
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest

### Expected behavior

Not getting a error.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Oslo",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "watchman": {
      "version": "0.5.1",
      "requirements": [
        "prettytable==3.0.0"
      ]
    },
    "metnowcast": {
      "version": "v2.3.0",
      "requirements": []
[... 557 lines cut ...]
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

ur7x, 2023-10-17:
Sorry I can't help you with this, but you are not alone, my system started doing this too...  about 2 weeks ago

  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 332, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 557, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message:
  Bad status line "Invalid method encountered:\n\n  b'SSTP_DUPLEX_POST /

henrikknowit, 2023-11-21:
Hi, I have the same problem. Would love to have someone look at this issue.

closed 2024-05-26 by github-actions[bot]