zigpy/zha-device-handlers #2621: [Device Support Request] ClickSmart+ CMA30035

Blackduke77, 2023-10-04:
### Problem description

This is a single UK in wall socket, https://click-smart.com/products/definity-smart-sockets
after in there is basic switch operation, but there is more available on this switch. I believe this is working under a generic driver. 
Under device info it is reporting as TS011F by TYZB01_mtunwanm
I am fairly certain this should be similar to TS011F

### Solution description

Please can we have device support for: on/off, lock, start as (on/off), light control

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0009",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x000a"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_mtunwanm",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.09.2",
    "host_os": "Home Assistant OS 10.5",
[... 306 lines cut ...]
  }
}]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

TheJulianJES, 2023-10-04:
Can you try to see if it works HA Core 2023.10.0?

Blackduke77, 2023-10-04:
Just trying to work out how I can install a beta build

Blackduke77, 2023-10-04:
Just upgraded and no different,

javicalle, 2023-10-05:
Maybe that quirk could do the job:

<details open>
<summary>Tuya_1G_Switch_var02</summary>

```python

"""Tuya TS011F Switches."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaSwitch,
    TuyaZBE000Cluster,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBOnOffAttributeCluster,
)

class Tuya_1G_Switch_var02(TuyaSwitch):
    """Tuya 1 gang light switch (variation 02)."""

    signature = {
        MODEL: "TS011F",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=9
            # input_clusters=[0, 4, 5, 6,10]
            # output_clusters=[25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Time.cluster_id,  
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    Time.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id],
            },
        },
    }
```

</details>

Blackduke77, 2023-10-09:
Thanks, I will give it a test

Blackduke77, 2023-10-09:
Thanks, the option has come up straight away, I will test the functionality tomorrow, but looks like the above adds the child lock.
I was trying to look at the quirk and reverse engineer how you surfaced the child lock, so I could work out how to surface some other the other features, like Power on state,  but for the life of keep I cant see how, any advice or help in surfacing some of the other settings like light and Power on State. the power on state, might be ok as reading the attribute it is defaulting to Last State.

Maybe a silly question, how do we add this to the default build so we don't have to deploy a quirk?

[image]

[image]

Blackduke77, 2023-10-09:
here is the diagnostics, I think it is working now

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.10.0",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "octopus_energy": {
      "version": "8.3.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp",
      "universal_silabs_flasher"
    ],
    "requirements": [
      "bellows==0.36.5",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.104",
      "zigpy-deconz==0.21.1",
[... 285 lines cut ...]
              "0x8004": {
                "attribute_name": "switch_mode"
              },
              "0xfffe": {
                "attribute_name": "reporting_status"
              }
            }
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

closed 2024-04-13 by github-actions[bot]