zigpy/zha-device-handlers #2623: Curtain Switch TS130F Modell SM-SW101-CZ no quirk loaded, no calibration, no Motor inversion.

jopeyyyy, 2023-10-05:
### Problem description

no quirk loaded, calibration attribute missing, no calibration possible, reverse output with HA command (physicaly OK)

### Solution description

correct quirk to load?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "output_clusters": [
        "0x000a"
      ]
    }
  },
  "manufacturer": "_TZ3000_ltiqubue",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.9.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.10.0",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "zha_toolkit": {
      "version": "v1.0.0",
      "requirements": [
        "pytz"
      ]
    },
[... 503 lines cut ...]
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

### Additional information

switches bought on amazon: https://www.amazon.de/dp/B0CCXZ96Z7/ref=pe_27091401_487027711_TE_SCE_dp_2

ZHA Toolkit installed and activated, custom quirk file and directory added to configuration.yaml

´´´
zha_toolkit:

zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks/
´´´

I'm new to Home Assistant and this is one of my first devices ...
Thanks for your help!

javicalle, 2023-10-05:
Remove any local quirk for that device and try this one:

<details open>
<summary>TuyaTS130F_var10</summary>

```python

"""Device handler TS130F smart curtain switch."""
from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import SwitchBackLight, TuyaZBExternalSwitchTypeCluster

ATTR_CURRENT_POSITION_LIFT_PERCENTAGE = 0x0008
CMD_GO_TO_LIFT_PERCENTAGE = 0x0005

class TuyaWithBacklightOnOffCluster(CustomCluster, OnOff):
    """Tuya Zigbee On Off cluster with extra attributes."""

    attributes = OnOff.attributes.copy()
    attributes.update({0x8001: ("backlight_mode", SwitchBackLight)})

class MotorMode(t.enum8):
    """Tuya motor mode enum."""

    STRONG_MOTOR = 0x00
    WEAK_MOTOR = 0x01

class TuyaCoveringCluster(CustomCluster, WindowCovering):
    """TuyaSmartCurtainWindowCoveringCluster: Allow to setup Window covering tuya devices."""

    attributes = WindowCovering.attributes.copy()
    attributes.update({0x8000: ("motor_mode", MotorMode)})
    attributes.update({0xF000: ("tuya_moving_state", t.enum8)})
    attributes.update({0xF001: ("calibration", t.enum8)})
    attributes.update({0xF002: ("motor_reversal", t.enum8)})
    attributes.update({0xF003: ("calibration_time", t.uint16_t)})

    def _update_attribute(self, attrid, value):
        if attrid == ATTR_CURRENT_POSITION_LIFT_PERCENTAGE:
            # Invert the percentage value (cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757)
            value = 100 - value
        super()._update_attribute(attrid, value)

    async def command(
        self, command_id, *args, manufacturer=None, expect_reply=True, tsn=None
    ):
        """Override default command to invert percent lift value."""
        if command_id == CMD_GO_TO_LIFT_PERCENTAGE:
            percent = args[0]
            # Invert the percentage value (cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757)
            percent = 100 - percent
            v = (percent,)
            return await super().command(command_id, *v)
        return await super().command(
            command_id,
            *args,
            manufacturer=manufacturer,
            expect_reply=expect_reply,
            tsn=tsn,
        )

class TuyaTS130F_var10(CustomDevice):
    """Tuya smart curtain roller shutter Time Out (variation 10)."""

    signature = {
        MODEL: "TS130F",
        ENDPOINTS: {
            1: {
				# "profile_id": "0x0104", "device_type": "0x0202", 
				# "input_clusters": [ "0x0000", "0x0003", "0x0004", "0x0005", "0x0006", "0x0102" ], 
				# "output_clusters": [ "0x000a" ]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
					Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
					Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaCoveringCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id],
            },
        },
    }

```
</details>

jopeyyyy, 2023-10-05:
Thanks for the feedback!

So I removed everything in my TS130F.py file and replaced it by that what you gave me. The result is unchanged, no quirk loaded.

I'm wondering if everything is correctly configured in my Home Assistant so the ZHA Toolkit and Custom quirks can even be loaded as it should? Once again, I'm a complete newbie to Home Assistant...

javicalle, 2023-10-05:
After changing and saving the file you need to restart HA.
Then remove the device, wait a minuto or so, and pair the device again.

Check for any error in your logs, on HA start up or while pairing the device.

You can enable the debug logs from the HA integration view and try to pair the device. After pairing, stop the debug logs from HA and attach the logs here.

MattWestb, 2023-10-05:
ZHA Toolkit is very likely (i have 3 loading it and 3 systems not) broke but its reported to the devevs but customer quirks is working well if its OK made.

jopeyyyy, 2023-10-05:
OK I made that (restart, removing, pairing again), I didn't wait a minute after removing the device....

Now I removed, restarted HA, waited 2 min and paired again. Here is the log:

[HA logfile.txt](https://github.com/zigpy/zha-device-handlers/files/12822786/HA.logfile.txt)

Can you find something wrong regarding to the switch/quirk?

jopeyyyy, 2023-10-07:
> After changing and saving the file you need to restart HA. Then remove the device, wait a minuto or so, and pair the device again.
>
> Check for any error in your logs, on HA start up or while pairing [...]

Could you find something in my log?

jopeyyyy, 2023-10-09:
Hi, I need to send the switched back if I don't get them working properly in time...

Would be fine if someone could help me, or give me feedback that it will not work , or if more Information are needed.

Thanks guys.

jopeyyyy, 2023-10-11:
Hello,
just wanted to inform that I bought another curtain switch (looks the same but with curtain pictures as button). That version has been identified by a Quirk directly and works properly. Calibration was possible!

As I sent the other version back, I cannot test anymore and close that issue.

closed 2023-10-11 by jopeyyyy