zigpy/zha-device-handlers #2628: [Device Support Request] MOES Star Ring 2 gang Zigbee smart dimmer switch

minkysteve, 2023-10-07:
### Problem description

Request for help with a handler for Moes Star Ring 2 gang switches (Also come in 1 and 3 gang versions) - They currently pair with ZHA via Skyconnect and work as routers, but their entities are not available in ZHA for use.

product here: https://moeshouse.com/products/star-ring-zigbee-dimmer-switch?variant=46040268276027

### Solution description

Hoping they can be integrated into ZHA. They're the only product so far that hasn't integrated seamlessly for me, so trying to avoid a move to MQTT.
I think there is a MQTT thread for the device here if that helps.
https://github.com/Koenkk/zigbee2mqtt/issues/18650

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_zenj4lxv",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
[... 367 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

danielgomezrico, 2023-10-19:
It would be wonderful to have it! I just bought it but got the same results 😢 

If anyone can guide me I can try to submit a PR for that

danielgomezrico, 2023-10-19:
They are already added on zigbee2mqtt https://github.com/Koenkk/zigbee2mqtt/issues/18650, but I am unable to test that currently

closed 2023-10-20 by TheJulianJES

gorstj, 2023-11-13:
There seems to be an issue with the Quirk for this device

Mainly works fine. However, when switching it on it comes on at the previously set value.

If useful, I can check the behavior with zigbee2mqtt

e.g.
Dimmed to 5%
Switched off
Service call to switch on with brightness 100%--> turns on to 5% (shows 100% in GUI)
or use GUI slider to turn on to 100% brightness-->turns on to 5% (shows 100% in GUI)

Subsequent changes (whilst the light is on) to brightness work fine.

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "6.1.56",
    "supervisor": "2023.11.0",
    "host_os": "Home Assistant OS 11.0",
    "docker_version": "24.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "heatmiserneo": {
      "version": "0.2.4",
      "requirements": [
        "neohubapi==1.0"
      ]
    },
    "smartthings_soundbar": {
      "version": "v1.0.2",
      "requirements": []
    },
    "hildebrandglow_dcc": {
      "version": "1.0.3",
      "requirements": [
        "pyglowmarkt==0.5.5"
      ]
    },
    "alexa_media": {
      "version": "4.6.5",
      "requirements": [
        "alexapy==1.27.6",
        "packaging>=20.3",
        "wrapt>=1.14.0"
      ]
    },
    "switch_manager": {
      "version": "1.3.1",
      "requirements": []
    },
    "scheduler": {
      "version": "v0.0.0",
      "requirements": []
    },
    "ocpp": {
      "version": "v0.4.42",
      "requirements": [
[... 592 lines cut ...]
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

danielgomezrico, 2024-01-08:
We should reopen this issue, it is indeed not reliable, I just got some dimmerizable lights and have not responsive behaviours 🤔

dales, 2025-02-11:
Was this ever solved? I just got a switch and doesn't work

[image]

Dan1jel, 2025-03-30:
Same here, also got this but do t see any buttons/switches in ZHA?