zigpy/zha-device-handlers #2633: [Device Support Request] Frankever Smart Water Valve (TS0601 | _TZE200_1n2zev06)

flaktivated, 2023-10-09:
### Problem description

Purchased a new Smart Water Valve Branded by Frankever Model Number FKS221992.
https://frankever.com/smart-watering-system

Pairs successfully to ZHA however switch is not exposed.

Functions of Device:
- On / Off

### Solution description

Add support for the simple valve using existing logic already built for the TS0601 type device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_1n2zev06",
  "model": "TS0601",
  "class": "ts0601_valve.BasicTuyaValve"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.8.4",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Australia/Sydney",
[... 346 lines cut ...]
                    TuyaValveManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOff,
                    TuyaValveManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

</details>

flaktivated, 2023-10-09:
Working Quirk Created, pull request raised.

### Test Evidence
Pairing the Device:
[image]
Seeing Device Controls:
[image]
Opening the Device from HA (See Logbook):
[image]
Closing the Device from Physical Button (See Logbook):
[image]

flaktivated, 2023-10-09:
<detail><summary>Detail Log Investigation</summary>
2023-10-10 07:18:55.260 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x4ECC, SrcIEEE=e0:79:8d:ff:fe:b1:df:5a, ParentNwk=0x00C2)

2023-10-10 07:18:55.261 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x4ECC, SrcIEEE=e0:79:8d:ff:fe:b1:df:5a, ParentNwk=0x00C2)

2023-10-10 07:18:55.262 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0x4ECC, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)

2023-10-10 07:18:55.316 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x4ECC, NWK=0x4ECC, IEEE=e0:79:8d:ff:fe:b1:df:5a, Capabilities=<MACCapabilities.Router|MainsPowered|RXWhenIdle|AllocateShortAddrDuringAssocNeeded: 142>)

2023-10-10 07:18:55.317 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x4ECC, IsBroadcast=<Bool.true: 1>, ClusterId=19, SecurityUse=0, TSN=131, MacDst=0xFFFF, Data=b'\xCC\x4E\x5A\xDF\xB1\xFE\xFF\x8D\x79\xE0\x8E')

2023-10-10 07:18:55.318 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4ECC), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=131, profile_id=0, cluster_id=19, data=Serialized[b'\x83\xccNZ\xdf\xb1\xfe\xff\x8dy\xe0\x8e'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)

2023-10-10 07:18:55.318 INFO (MainThread) [zigpy.application] New device 0x4ecc (e0:79:8d:ff:fe:b1:df:5a) joined the network

2023-10-10 07:18:55.319 DEBUG (MainThread) [zigpy.device] [0x4ecc] Scheduling initialization

2023-10-10 07:18:55.319 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x4ECC ieee=e0:79:8d:ff:fe:b1:df:5a is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x83\xccNZ\xdf\xb1\xfe\xff\x8dy\xe0\x8e'

2023-10-10 07:18:55.320 DEBUG (MainThread) [zigpy.zdo] [0x4ecc:zdo] ZDO request ZDOCmd.Device_annce: [0x4ECC, e0:79:8d:ff:fe:b1:df:5a, 142]

2023-10-10 07:18:55.320 INFO (MainThread) [zigpy.device] [0x4ecc] Requesting 'Node Descriptor'

2023-10-10 07:18:55.320 DEBUG (MainThread) [zigpy.device] [0x4ecc] Extending timeout for 0x38 request

2023-10-10 07:18:55.320 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4ECC), dst_ep=0, source_route=None, extended_timeout=True, tsn=56, profile_id=0, cluster_id=<ZDOCmd.Node_Desc_req: 0x0002>, data=Serialized[b'8\xccN'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)

2023-10-10 07:18:55.357 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteChk.Req(Dst=0x4ECC, RtStatus=<RouteStatus.ACTIVE: 1>, Options=<RouteOptions.MTO_ROUTE|NO_ROUTE_CACHE: 3>)

2023-10-10 07:18:56.659 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4ECC), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=2, TSN=56, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x38\xCC\x4E')

2023-10-10 07:18:56.963 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x4ECC, Relays=[0x00C2])

2023-10-10 07:18:56.964 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.NodeDescRsp.Callback(Src=0x4ECC, Status=<Status.SUCCESS: 0>, NWK=0x4ECC, NodeDescriptor=NullableNodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False))

2023-10-10 07:18:56.966 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x4ECC, IsBroadcast=<Bool.false: 0>, ClusterId=32770, SecurityUse=0, TSN=56, MacDst=0x0000, Data=b'\x00\xCC\x4E\x01\x40\x8E\x02\x10\x52\x52\x00\x00\x2C\x52\x00\x00')

2023-10-10 07:18:56.966 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4ECC), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=56, profile_id=0, cluster_id=32770, data=Serialized[b'8\x00\xccN\x01@\x8e\x02\x10RR\x00\x00,R\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)

2023-10-10 07:18:56.967 DEBUG (MainThread) [zigpy.application] Received frame on uninitialized device <Device model=None manuf=None nwk=0x4ECC ieee=e0:79:8d:ff:fe:b1:df:5a is_initialized=False> from ep 0 to ep 0, cluster 32770: b'8\x00\xccN\ [... cut ...]
2023-10-10 07:19:08.668 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4ECC), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=52, Options=<TransmitOptions.ACK_REQUEST|SUPPRESS_ROUTE_DISC_NETWORK: 48>, Radius=0, Data=b'\x18\x34\x0B\x02\x86')

2023-10-10 07:19:08.763 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x4ECC, Relays=[0x00C2])
</detail>

flaktivated, 2023-10-09:
Seeing if there is a DP for Valve Position:
<details><summary>Logs from Open to Close</summary>
2023-10-10 07:50:52.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 4 value: _TZE200_1n2zev06

2023-10-10 07:50:52.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 5 value: TS0601

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 4 value: _TZE200_1n2zev06

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 5 value: TS0601

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0006] Attribute id: 0 value: 1

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 1 value: 68

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 65506 value: 31

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 65508 value: 0

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 65503 value: ���,���,���,]��,^��,

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0xef00] Attribute id: 61184 value: 2.0.0

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0702] Attribute id: 774 value: 2

2023-10-10 07:50:52.185 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0702] Attribute id: 768 value: 7

2023-10-10 07:51:11.115 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 4 value: _TZE200_1n2zev06

2023-10-10 07:51:11.115 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 5 value: TS0601

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 4 value: _TZE200_1n2zev06

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 5 value: TS0601

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0006] Attribute id: 0 value: 1

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 1 value: 68

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 65506 value: 31

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 65508 value: 0

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0000] Attribute id: 65503 value: ���,���,���,]��,^��,

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0xef00] Attribute id: 61184 value: 2.0.0

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0702] Attribute id: 774 value: 2

2023-10-10 07:51:11.142 DEBUG (MainThread) [zigpy.appdb] [0x4ecc:1:0x0702] Attribute id: 768 value: 7

2023-10-10 07:51:23.549 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x4ECC, Relays=[])

2023-10-10 07:51:23.749 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x4ECC, Relays=[])

2023-10-10 07:51:24.134 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x4ECC, Relays=[])

2023-10-10 07:51:24.188 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=0, SrcAddr=0x4ECC, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=54, SecurityUse=<Bool.false: 0>, TimeStamp=347365, TSN=0, Data=b'\x08\x01\x0A\x01\x00\x20\x44\xE2\xFF\x20\x1F\xE4\xFF\x20\x00', MacSrcAddr=0x4ECC, MsgResultRadius=29)

2023-10-10 07:51:24.189 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x4ECC), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=0, data=Serialized[b'\x08\x01\n\x01\x00 D\xe2\xff \x1f\xe4\xff \x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=54, rssi=None)

2023-10-10 07:51:24.189 DEBUG (MainThread) [zigpy.zcl] [0x4ECC:1:0x0000] Received ZCL frame: b'\x08\x01\n\x01\x00 D\xe2\xff \x1f\xe4\xff \x00'

2023-10-10 07:51:24.190 DEBUG (MainThread) [zigpy.zcl] [0x4ECC:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=1, command_id=10, *direction=<Direction.Client_to_Server: 1>)

2023-10-10 07:51:24.192 DEBUG (MainThread) [zigpy.zcl] [0x4ECC:1:0x0000] Decoded ZCL frame: Basic:Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=68)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=31)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])

2023-10-10 07:51:24.192 DEBUG (MainThread) [zigpy.zcl] [0x4ECC:1:0x0000] Received command 0x0A (TSN 1): Report_Attributes(attribute_reports=[Attribute(attrid=0x0001, value=TypeValue(type=uint8_t, value=68)), Attribute(attrid=0xFFE2, value=TypeValue(type=uint8_t, value=31)), Attribute(attrid=0xFFE4, value=TypeValue(type=uint8_t, value=0))])

2023-10-10 07:51:24.193 DEBUG (MainThread) [zigpy.zcl] [0x4ECC:1:0x0000] Attribute report received: app_version=68, 0xFFE2=31, 0xFFE4=0

[... 437 lines cut ...]

[... cut ...]  data=Serialized[b'\x18\x0e\x0b\x01\x86'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)

2023-10-10 07:52:08.480 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x4ECC), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=61184, TSN=14, Options=<TransmitOptions.ACK_REQUEST|SUPPRESS_ROUTE_DISC_NETWORK: 48>, Radius=0, Data=b'\x18\x0E\x0B\x01\x86')
</details>

closed 2023-10-24 by TheJulianJES

derterr0r, 2024-04-26:
Hello,

i have the same valve mentioned above.
Opening and closing works well. Is it possible to add the valve position and automatic close time?
I tryed the valve in zigbee2mqtt with eclipse-mosquitto, there these settings are exposed and working.

inigortega, 2024-07-23:
Hello,

How can I install or add 6 May commit of [zha-device-handlers](https://github.com/zigpy/zha-device-handlers) to my home assistant in order to add this functionality of frankever zigbee valve?

Best regards,

> Hello,
> 
> i have the same valve mentioned above. Opening and closing works well. Is it possible to add the valve position and automatic close time? I tryed the valve in zigbee2mqtt with eclipse-mosquitto, there these settings are exposed and working.

PR #2634 (2023-10-24): Add Tuya Basic Valve TS0601 `_TZE200_1n2zev06` (#2633)
Frankever Smart Water Valve (https://frankever.com/smart-watering-system)

## Proposed change
Adds Support for a Basic Tuya Water Valve, in my instance Frankever Smart Water Valve.
This water Valve only supports the `OnOff` cluster and has no other configurable functions.

This quirk fixes the ability to trigger the valve on and off.

## Additional information
Fixes #2633 

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to ve [... cut]