zigpy/zha-device-handlers #2635: [Device Support Request] - ZigbeeModel: TS0601, Zigbee Manufacturer: _TZE204_nklqjk62

Lilos007, 2023-10-09:
### Problem description

Zigbee Manufacturer: _TZE204_nklqjk62 is different to the one currently supported - Zigbee Manufacturer: _TZE200_nklqjk62. Device doesn't recognise any of the entity attributes. Binding is successful, but reporting is blank using https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_garage.py.

It is the same device mentioned here - https://github.com/Koenkk/zigbee2mqtt/issues/18633

### Solution description

Device is currently unusable and without showing the attributes. You can't open the gate or see if the gate sensors are working.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

None

Danieldiazi, 2023-10-15:
I friend of mine had just bought this model and he has used this quirk file:

`       
       
        from typing import Dict
        from zigpy.profiles import zha
        from zigpy.quirks import CustomDevice
        import zigpy.types as t
        from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time
        from zigpy.zcl.clusters.security import IasZone

        from zhaquirks.const import (
            DEVICE_TYPE,
            ENDPOINTS,
            INPUT_CLUSTERS,
            MODELS_INFO,
            OUTPUT_CLUSTERS,
            PROFILE_ID,
        )
        from zhaquirks.tuya import TuyaLocalCluster
        from zhaquirks.tuya.mcu import (
            TuyaMCUCluster,
            TuyaOnOff,
            DPToAttributeMapping

        )

        #    TuyaDPType,

        from zhaquirks.tuya.mcu import ( TuyaMCUCluster
        )

        from zhaquirks.tuya.ts0601_dimmer import TuyaOnOffNM

        ZONE_TYPE = 0x0001

        class ContactSwitchCluster(TuyaLocalCluster, IasZone):
            """Tuya ContactSwitch Sensor."""

            _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

            def _update_attribute(self, attrid, value):
                self.debug("_update_attribute '%s': %s", attrid, value)
                super()._update_attribute(attrid, value)

        class TuyaGarageManufCluster(TuyaMCUCluster):
            """Tuya garage door opener."""

            attributes = TuyaMCUCluster.attributes.copy()
            attributes.update(
                {
                    # ramdom attribute IDs
                    0xEF02: ("dp_2", t.uint32_t, True),
                    0xEF04: ("dp_4", t.uint32_t, True),
                    0xEF05: ("dp_5", t.uint32_t, True),
                    0xEF0B: ("dp_11", t.Bool, True),
                    0xEF0C: ("dp_12", t.enum8, True),
                }
            )

            dp_to_attribute: Dict[int, DPToAttributeMapping] = {
                # garage door trigger ¿on movement, on open, on closed?
                1: DPToAttributeMapping(
                    TuyaOnOffNM.ep_attribute,
                    "on_off",
                    #dp_type=TuyaDPType.BOOL,
                ),
                2: DPToAttributeMapping(
                    TuyaMCUCluster.ep_attribute,
                    "dp_2",
[... 84 lines cut ...]
                            TuyaGarageManufCluster,
                            TuyaOnOffNM,
                        ],
                        OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
                    },
                    2: {
                        PROFILE_ID: zha.PROFILE_ID,
                        DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                        INPUT_CLUSTERS: [
                            ContactSwitchCluster
                        ],
                        OUTPUT_CLUSTERS: [],
                    },
                    242: {
                        PROFILE_ID: 0xA1E0,
                        DEVICE_TYPE: 0x0061,
                        INPUT_CLUSTERS: [],
                        OUTPUT_CLUSTERS: [0x0021],
                    },
                },
            }

`

It appears a light switch and a open/close sensor. He hasn't been able to test it on the garage door yet. If someone has the same model and wish to test it in a real environment  maybe we can have some interesting feedback

zdeneksvarc, 2023-10-29:
Thank you for the quirk @Danieldiazi

Please remove the \` before and after the code in your post as these characters will break the indentation.

Quirk works well, my friend said in return, but try to consider:

- Add automation to reset the switch after 1 s, since the event is more like touching a button
```
id: '...'
alias: Garage Door Switch Reset
description: ''
trigger:
  - platform: state
    entity_id:
      - switch.garage_door_controller_switch
    to: 'on'
    for:
      hours: 0
      minutes: 0
      seconds: 1
condition: []
action:
  - service: switch.turn_off
    data: {}
    target:
      entity_id: switch.garage_door_controller_switch
mode: single
```
- In the quirk code change

```
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
```

for

```
    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
```

tengotengo, 2023-10-29:
Same issue. Entities don't get created. 
Tried the .py file as /config/zhaquirks/ts0601_garage.py from the above comment. Including the
`DEVICE_TYPE: zha.DeviceType.SMART_PLUG` instead of `ON_OFF_LIGHT`  change

After .py file changes, I delete the device, restart HA and pair the device again.

Also added 
```
zha:
  custom_quirks_path: /config/zhaquirks
```
to /config/configuration.yaml

[image]

My diagnostics file:

```
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.10.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "5.10.0-21-amd64",
    "supervisor": "2023.10.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "24.0.6",
    "chassis": "laptop",
    "run_as_root": true
  },
  "custom_components": {
    "govee_lan": {
      "version": "1.0.0",
      "requirements": [
        "govee-led-wez>=0.0.14"
      ]
    },
    "meross_lan": {
      "version": "3.0.3",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "govee": {
      "version": "0.2.2",
      "requirements": [
        "govee-api-laggat==0.2.2",
        "dacite==1.6.0"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
[... 262 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

tengotengo, 2023-10-29:
Update: 
script posted here: https://github.com/zigpy/zha-device-handlers/issues/2533  
worked!

Danieldiazi, 2023-10-29:
> Thank you for the quirk @Danieldiazi
> 
> Please remove the ` before and after the code in your post as these characters seem to break the indentation.
> 
> Quirk works well, my friend said in return, but try to consider:
> 
>     * Add automation to reset the switch after 1 s, since the event is more like touching a button
> 
> 
> ```
> id: '...'
> alias: Garage Door Switch Reset
> description: ''
> trigger:
>   - platform: state
>     entity_id:
>       - switch.garage_door_controller_switch
>     to: 'on'
>     for:
>       hours: 0
>       minutes: 0
>       seconds: 1
> condition: []
> action:
>   - service: switch.turn_off
>     data: {}
>     target:
>       entity_id: switch.garage_door_controller_switch
> mode: single
> ```
> 
>     * In the quirk code change
> 
> 
> ```
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
> ```
> 
> for
> 
> ```
>     replacement = {
>         ENDPOINTS: {
>             1: {
>                 DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
> ```

I've removed ' before and after the code!
Thanks for the suggestions!

zdeneksvarc, 2023-10-29:
> Same issue. Entities don't get created.

I believe it is the incorrect formatting of the quirk code mentioned above. Python code must start without indentation. HA will point this out in the log when it tries to load the quirk file.

liouma, 2024-04-02:
Thanks for the quirk, it's working fine.
Turning on the switch activate the physical connection for a second, the it is immediately interrupted. This is probably the expected behaviour for a garage door but not for the device I use. I need a normal on/off switch.

Is this behaviour built in the hardware or is it programmed in the quirk? Do you know if there is a software way to modify it??
Thanks a lot!

closed 2024-10-06 by github-actions[bot]

Arean71, 2024-12-07:
With in the latets zha-handles-model update my garage opener (_TZE204_nklqjk62) does not recognized by builtin quirk. Diagnostik file is attached:
[zha-a8058e925082e356771c310c441b17fb-_TZE204_nklqjk62 TS0601-8f6ecb1358f469f5d7034e782cf36012.json](https://github.com/user-attachments/files/18048345/zha-a8058e925082e356771c310c441b17fb-_TZE204_nklqjk62.TS0601-8f6ecb1358f469f5d7034e782cf36012.json)