zigpy/zha-device-handlers #2636: [Device Support Request] Blind motor "_TZE200_libht6ua"

kineticscreen, 2023-10-10:
### Problem description

This appears to be similar to other blind motors - just another variation on the manufacturer name?

https://www.aliexpress.com/item/1005002954586484.html?spm=a2g0o.order_list.order_list_main.5.529e1802C56G4M

### Solution description

Hopefully is just a simple port of the other quirks for similar devices

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_libht6ua",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Sydney",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.10.0",
    "host_os": "Home Assistant OS 10.5",
    "docker_version": "23.0.6",
[... 331 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

superflyman74, 2023-11-02:
I have the same bind motor... How can this be added to HA?

kineticscreen, 2023-11-16:
> I have the same blind motor... How can this be added to HA?

I ended up just buying the cheapest Tuya wireless hub and then integrated via local Tuya. Doesn't bring in the temperature or its battery state, but it's a start...

[image]
[image]

hplar11111, 2023-11-20:
I would also like a quirk for this device.

javilopezalarcon, 2023-11-22:
I have the same motor for Tuya's blind that appears to me as unsupported if someone could help us I would appreciate it

javilopezalarcon, 2023-11-25:
I'm working on being able to add it to z2m, I would say it's ready but I need to learn how to upload it to the code... while I'm learning, if anyone wants to try it, I uploaded it here https://github.com/javilopezalarcon/Blind-motor-_TZE200_libht6ua-z2m  It works well, with position and battery control, the motor failure remains to be tested if it is real.

kineticscreen, 2023-12-01:
It sounds like your unit is in 'tilt' mode, which is just for adjusting the angle of venetian blinds.
From memory these instructions, which are actually for a different unit, also worked for this one, despite its very incomplete included manual.

[image]

fearless666, 2023-12-01:
> I'm working on being able to add it to z2m, I would say it's ready but I need to learn how to upload it to the code... while I'm learning, if anyone wants to try it, I uploaded it here [...]

Thanks for the converter @javilopezalarcon. I managed to figure out that datapoint 101 is for motor direction and 102 sets the upper limit and have made the following changes to your converter.

```js
    exposes: [
        ...,
        e.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
        e.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning')
    ],
    meta: {
        tuyaDatapoints: [
            ...,
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({ 'left': tuya.enum(0), 'right': tuya.enum(1) })],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({ 'start': tuya.enum(0), 'stop': tuya.enum(1) })],
        ],
    },
```

@env-dxb I had a similar problem to you until I found out about setting an upper limit. You can try the following to see if it helps.

1. Manually close your curtains
2. While the curtains are shut, connect them to the motor
3. Click "start" under "Set upper limit" to start learning the upper limit of your curtain. The motor should begin to open the curtain
4. Just before the curtain is fully open, click "stop" under "Set upper limit"
5. The controller should now remember the fully close and open positions

env-dxb, 2023-12-01:
> It sounds like your unit is in 'tilt' mode, which is just for adjusting the angle of venetian blinds. From memory these instructions, which are actually for a different unit, also worked for this [...]

Tilt mode exactly what I need, but it’s about half of required limit, and manual doesn’t work for me as well((
But I keep trying)

env-dxb, 2023-12-01:
> Thanks for the converter @javilopezalarcon. I managed to figure out that datapoint 101 is for motor direction and 102 sets the upper limit and have made the following changes to your converter. [...]

Seems I need a hub and app for it, I don’t)

kineticscreen, 2023-12-03:
Kind of weird the way this issue on the ZHA repository has turned into a Zigbee2MQTT thread, but hopefully that will lead to a quirk being available for ZHA I guess?

javilopezalarcon, 2023-12-18:
Thank you very much @env-dxb  for the contribution of the new sensors, I will add them before sending it to z2m (sorry for the delay, I was very busy) 👍🏻👍🏻

javilopezalarcon, 2023-12-18:
It's my fault for converting this from zha to z2m, I'm sorry @kineticscreen ... I was the same, I couldn't find anything that worked for me and I found this thread, where I saw that there were more people like me, in zha yes, but like me, without integration. .. I started to work to be able to add it in z2m and I wanted you to be the first to know, since you were like me... I imagine, that having the data from each sensor, passing it to zha will be much easier than not having anything, If it helps, I already fulfilled what I wanted, which was to help 😅

hplar11111, 2023-12-19:
But is it possible to make it work in ZHA?

oxcid3, 2023-12-23:
Also interested in this.

TeaMoe, 2024-02-15:
I've had some success by using the following quirk:

```python
"""Tuya based cover and blinds."""
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufacturerWindowCover,
    TuyaManufCluster,
    TuyaWindowCover,
    TuyaWindowCoverControl,
)

class TuyaMoesCover0601(TuyaWindowCover):
    """Tuya blind controller device."""

    tuya_cover_inverted_by_default = True

    signature = {
        # "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098,
        #                    maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264,
        #                    maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
        # "endpoints": {
        # "1": { "profile_id": 260, "device_type": "0x0051", "in_clusters": [ "0x0000", "0x0004","0x0005","0xef00"], "out_clusters": ["0x000a","0x0019"] }
        # },
        # "manufacturer": "_TZE200_zah67ekd",
        # "model": "TS0601",
        # "class": "zigpy.device.Device"
        # }
        MODELS_INFO: [
            ("_TZE200_libht6ua", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

Opening and closing works, which is what i need. I set the open and close positions by using the buttons on the device itself.

kineticscreen, 2024-02-17:
Does the quirk expose anything else, like Battery status etc? Or just basic up down. Does it support percentage adjustments?

TeaMoe, 2024-02-17:
It supports percentage adjustments but no battery status yet. I planed to
do more research in this device tomorrow. Maybe i'm able to figure out how
the battery status reporting works.

Aephir, 2024-02-20:
Taking the [tuya 0610 cover quirk](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_cover.py) and adding the device to the `MODELS_INFO` of the `TuyaZemismartSmartCover0601_3` class seems to at least do something.

<details>
<summary> modified MODELS_INFO, line 173 </summary>

```
        MODELS_INFO: [
            ("_TZE200_fzo2pocs", "TS0601"),
            ("_TZE200_iossyxra", "TS0601"),
            ("_TZE200_pw7mji0l", "TS0601"),
            ("_TZE200_libht6ua", "TS0601"),  # <-- This is the one I added
        ],
```

</details>

I get the controls in HA, and pressing them does make the motor turn (everything in the logbook was controlled from HA).

[image]

I haven't done thorough testing, but maybe sharing this first step will get more people testing it out, and we can get that done quicker?

helzarto, 2024-03-06:
Hey guys, converter by @javilopezalarcon with modifications by @fearless666 worked well, thanks a lot. However, I have a slight issue I imagine can be fixed in code but I am quite a newbie so struggling on how to do so.

When pressing up or down in HA, it does the opposite of what the buttons do in the remote and the device. 
Now, it doesn’t matter on whether it is set on right config or left config, they are always the opposite of each other.
Can anyone help with a fix perhaps?

[image]

benaclejames, 2024-04-11:
Using the quirk provided by @TeaMoe, I can now successfully press the up, down, and stop buttons using HA. Currently, though, I have an issue where when closing, the motor reports an open state when it's finished closing. HA therefore always thinks it's closing. Any solutions?

hannes2984, 2024-04-11:
@helzarto  I am really sorry to apparently ask a dumb question. How did you add the converter  _TZE200_libht6ua.js to HA and zigbee2mqqt? (i am really new to HA). Help is really appreciated and sorry for "pushing" the thread to HA Z2M again.

closed 2024-10-28 by github-actions[bot]

zenyagami, 2026-02-25:
> I created this quirk, based on the Z286-libht6ua quirk.
> 
> https://github.com/bartgabriels/HA-CodeStore/blob/main/Customer%20Quirk%20for%20Blind%20Motor. Works like a charm for me ...

AMAZING! I tried the whole weekend, it was so annoying, thank you so much, I just have an issue it seems that the controls in HA are inverted, UP goes down and Down goes up, but other than that works great!!
do you know by change how to fix this?