zigpy/zha-device-handlers #2637: [Device Support Request] TS0502B | _TZ3210_xwqng7ol | Rexlight Zigbee 3.0 Dual White LED controller

rubdos, 2023-10-10:
### Problem description

Pairing a TS0502B does not work correctly: the device pairs, gets recognised in Home Assistant, is then correctly controllable for a few seconds, after which the device seems to disconnect again.

### Solution description

The device should remain controllable.

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010c",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_xwqng7ol",
  "model": "TS0502B",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

No logs, let me know if they are still necessary.

<details><summary>Logs</summary>

```python
```

</details>

### Custom quirk

I started, but it doesn't seem to do anything yet.

<details><summary>Custom quirk</summary>

```python

"""Tuya dimmable dual white led controller."""
from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya.mcu import (
    TuyaLevelControlManufCluster,
)

class DimmableLedController(CustomDevice):
    """Tuya dimmable dual white led controller."""

    # {
    #   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
    #   "endpoints": {
    #     "1": {
    #       "profile_id": "0x0104",
    #       "device_type": "0x010c",
    #       "input_clusters": [
    #         "0x0000",
    #         "0x0003",
    #         "0x0004",
    #         "0x0005",
    #         "0x0006",
    #         "0x0008",
    #         "0x0300",
    #         "0x1000",
    #         "0xef00"
    #       ],
    #       "output_clusters": [
    #         "0x000a",
    #         "0x0019"
    #       ]
    #     },
    #     "242": {
    #       "profile_id": "0xa1e0",
    #       "device_type": "0x0061",
    #       "input_clusters": [],
    #       "output_clusters": [
    #         "0x0021"
    #       ]
    #     }
    #   },
    #   "manufacturer": "_TZ3210_xwqng7ol",
    #   "model": "TS0502B",
    #   "class": "zigpy.device.Device"
    # }
    signature = {
        MODELS_INFO: [
            ("_TZ3210_xwqng7ol", "TS0502B"),
            ("_TZ3210_frm6149r", "TS0502B"),
            ("_TZ3210_jtifm80b", "TS0502B"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=257
            # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    TuyaLevelControlManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
                # input_clusters=[]
                # output_clusters=[33]
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    LightLink.cluster_id,
                    TuyaLevelControlManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

### Additional information

The device seems available in z2m: https://github.com/Koenkk/zigbee2mqtt/discussions/12419 / https://github.com/Koenkk/zigbee-herdsman-converters/blob/c2b34bc8e26b3224187765f1b2088bad043cd677/src/devices/tuya.ts#L1790

```ts
{
        zigbeeModel: ['TS0502B'],
        model: 'TS0502B',
        vendor: 'TuYa',
        description: 'Light controller',
        whiteLabel: [
            tuya.whitelabel('Mercator Ikuü', 'SMI7040', 'Ford Batten Light', ['_TZ3000_zw7wr5uo']),
            {vendor: 'Mercator Ikuü', model: 'SMD9300', description: 'Donovan Panel Light'},
            tuya.whitelabel('Aldi', 'F122SB62H22A4.5W', 'LIGHTWAY smart home LED-lamp - filament', ['_TZ3000_g1glzzfk']),
            tuya.whitelabel('MiBoxer', 'FUT035Z', 'Dual white LED controller', ['_TZ3210_frm6149r', '_TZ3210_jtifm80b', '_TZ3210_xwqng7ol']),
            tuya.whitelabel('Lidl', '14156408L', 'Livarno Lux smart LED ceiling light', ['_TZ3210_c2iwpxf1']),
        ],
        extend: tuya.extend.light_onoff_brightness_colortemp({colorTempRange: [153, 500], noConfigure: true}),
        configure: async (device, coordinatorEndpoint, logger) => {
            device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 16});
        },
    },
```

rubdos, 2023-10-10:
https://github.com/zigpy/zha-device-handlers/pull/2639 seems to at least get detected correctly now. I will try pairing again tonight/tomorrow morning, when I have physical access to the pair button :-)

rubdos, 2023-10-10:
Same symptoms still appear, even when setting `SKIP_CONFIGURATION` to `True`. Moreover, I don't have the color temperature control anymore, so I probably miss an `input`.

javicalle, 2023-10-11:
> I don't have the color temperature control anymore, so I probably miss an `input`.

You have changed the `DEVICE_TYPE`. Is this intentional?

```python
    signature = {
        .../...
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=268
            # input_clusters=[0, 3, 4, 5, 6, 8, 768, 4096, 61184]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
.../...
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.DIMMABLE_LIGHT,
```

rubdos, 2023-10-11:
> You have changed the `DEVICE_TYPE`. Is this intentional?

No, and my pull-request actually has the correct one already.  But I definitely tested with `DIMMABLE_LIGHT`, I'll switch them around again.

TheJulianJES, 2023-11-01:
> Pairing a TS0502B does not work correctly: the device pairs, gets recognised in Home Assistant, is then correctly controllable for a few seconds, after which the device seems to disconnect again.

I'm pretty sure that's not an issue a quirk can fix.
It's possible that something is wrong with your network. It looks like you're using an EZSP coordinator. Did you migrate your network from another coordinator to this one? There might be "security issues" (with hashed link keys) which causes your device to leave.
Also, please try Home Assistant Core 2023.11.x (beta).

rubdos, 2023-11-02:
> I'm pretty sure that's not an issue a quirk can fix.

Then I don't think a quirk will actually be needed here.

> It's possible that something is wrong with your network. It looks like you're using an EZSP coordinator. Did you migrate your network from another coordinator to this one? There might be "security issues" (with hashed link keys) which causes your device to leave.

I migrated from Z2M to a Skyconnect, I followed [Everything Smart Home's video for that](https://www.youtube.com/watch?v=iUfJb4Vi6OE). Would you suggest I create a clean network? I have a whole lot of devices, and some of them are really awful to pair :'-(

rubdos, 2023-11-04:
> > Also, please try Home Assistant Core 2023.11.x (beta).
> 
> I will try that first! Thanks!

I have tried 2023.11.0 (not beta), which has the same symptoms (still with my quirk).

Maybe worth noting: I'm the same person as the one in #2639, so better to deduplicate the support effort and keep everything in this issue, I suppose?

I'm not very much looking forward to re-pairing (and renaming) 40 devices; that will probably take me half a day.

Redawoi, 2023-11-26:
Hi,

I'm also using the FUT035Z. It has two options, one for single colour with white balance and one for two led strips that you could control independently but only brightness. Can this be added as well?

If you need any information please let me know.
Nodes can be changed by double clicking the set button. In z2m it also worked, so maybe this can be done here as well :)

49jan, 2024-01-18:
I have the same problem with MiBoxer FUT035Z+ (_TZB210_lmqquxus, TS0502B).

Can this be added as well?

GenuineJakob, 2024-01-24:
Hi, I am also facing issues with my TS0502B. It is recognised and can be turned on/off, but neither the brightness nor the color temperature controls appear.
[image]

fhornrudd, 2024-01-28:
Just added a TS0502B to zha using skyconnect, working flawlessly (temperature and dimming) for two weeks. Using on 24v COB cw/ww strip.

Puma7, 2024-03-17:
> Hi,
>
> I'm also using the FUT035Z. It has two options, one for single colour with white balance and one for two led strips that you could control independently but only brightness. Can this be added [...]

> I have the same problem with MiBoxer FUT035Z+ (_TZB210_lmqquxus, TS0502B).
> 
> Can this be added as well?

> Hi, I am also facing issues with my TS0502B. It is recognised and can be turned on/off, but neither the brightness nor the color temperature controls appear. <img alt="TS0502B" width="648" src="https://private-user-images.githubusercontent.com/37959554/299467081-99dc0078-c827-4b6c-9017-37645f06cb8f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2MTIzMjAsIm5iZiI6MTcxMDYxMjAyMCwicGF0aCI6Ii8zNzk1OTU1NC8yOTk0NjcwODEtOTlkYzAwNzgtYzgyNy00YjZjLTkwMTctMzc2NDVmMDZjYjhmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE2VDE4MDAyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRhN2VjZGVjNjY4YTY1ZDc3NDkyMTcxNjRhZGEyM2JmYWIzNzkxN2Q3MzM3MDJkYThjYjk4NDFiMTlkN2IyZGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.eDRmwn5yk9OoUUgJdc1FbMB27YpF7BYwL2sJf-ylqE8">

Ensure that you have read the details on the controller box. My Miboxer FUT035Z+ states on the controller, "If the red light is on, only one color is available. If the green light is on, you can control the light temperature." In my case, I needed to press the set button on the MiBoxer twice, and the red light turned to green, allowing me to control not only the dimming and on/off functions but also the light temperature.

closed 2024-09-20 by github-actions[bot]