zigpy/zha-device-handlers #2643: Get occasional error in the logs

def1149, 2023-10-11:
### Bug description

I get the following in the logs occasionally but everything seems to be working fine

Logger: homeassistant.components.zha.core.cluster_handlers
Source: components/zha/core/cluster_handlers/init.py:537
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 3:12:18 PM (2 occurrences)
Last logged: 3:12:18 PM

[0xF9C9:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

### Steps to reproduce

Just appears in the logs 

### Expected behavior

No errors, LOL

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

TheJulianJES, 2023-10-12:
Can you check what device `0xF9C9` is? You can open up the ZHA network map visualization and enter that in the search to see a (barely) highlighted device.

def1149, 2023-10-12:
It's a Sonoff USB Dongle E

def1149, 2023-10-12:
> What exact firmware is the Sonoff USB Dongle E running ?

Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl <https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/blob/master/Dongle-E/Router/Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl> 

 

 

 

 

Regards,

 

Darrell

MattWestb, 2023-10-13:
itead have cooking one standard firmware from Silabs demo file nd have nit fixing all small things that the manufacture shall do.
Make one issue in there guthub and see if they can fixing it.
Alternative i think one x-employee wad doing one good working router firmware but i dont remember where it is.

MattWestb, 2023-10-13:
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle

def1149, 2023-10-13:
Looks like older version? 6.7.10 vs 6.10.3 that I'm using.

closed 2024-04-17 by github-actions[bot]