zigpy/zha-device-handlers #2647: [BUG]

accessiblepixel, 2023-10-12:
### Bug description

After updating from an older home assistant to latest home assistant, the quirk wouldn't run and trying to find an updated ts0601_trv_siterwell.py file from the thread (from Jan 3rd), it gives the following error.

Now I could just be being a donut and getting something wrong here (quite possible!) but it would be very appreciated if someone could point me in the right direction (or if it is indeed a bug a potential fix or workaround)?

The temperature still reports and updates, but all the other sensors in the valve are now greyed out.

Thank you muchly.

### Steps to reproduce

Install quirk, fails to load with exception "Unexpected exception importing custom quirk 'ts0601_trv_siterwell'"

### Expected behavior

It to load the quirk.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=0, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYST11_czk78ptr",
  "model": "zk78ptr",
  "class": "zhaquirks.tuya.ts0601_trv.SiterwellGS361_Type1"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.0",
    "dev": false,
    "hassio": true,
[... 416 lines cut ...]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 460, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_trv_siterwell.py", line 61, in <module>
    class SiterwellManufCluster(TuyaManufClusterAttributes):
  File "/usr/local/lib/python3.11/site-packages/zigpy/zcl/__init__.py", line 167, in __init_subclass__
    raise TypeError(f"Duplicate definitions exist for {duplicates}")
TypeError: Duplicate definitions exist for ['window_alarm']
```

</details>

### Additional information

Thank you :)

javicalle, 2023-10-12:
Have you tried to remove the quirk file and test the device?

MattWestb, 2023-10-13:
ZHA / ziggpy have make some internal changes and the quirk need being updated.
The father for tuya TRVs have posting updated working ones here https://github.com/jacekk015/zha_quirks.

accessiblepixel, 2023-10-24:
@MattWestb Ah, thank you. I hadn't seen that your comment had been modified since you wrote it.

I first tried downloading the new quirk from the github you linked and at first I got...

```Unexpected exception importing custom quirk 'ts0601_trv_siterwell'

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 460, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/ts0601_trv_siterwell.py", line 95
    <title>zha_quirks/ts0601_trv_siterwell.py at main · jacekk015/zha_quirks · GitHub</title>
                                                      ^
SyntaxError: invalid character '·' (U+00B7)
```
But then I realised it was a PEBKAC (problem exists between keyboard and chair) since I wget'ed the page showing the code, rather than using the github raw link to download the actual file. G'oh!

Anyway, this new version is working for me now and I can finally start valve calibration remotely - a couple of my thermostats are in awkward places so after battery replacements it can be difficult to start a valve calibration once they're back on the radiator.

Thanks to all involved for the help with this, it is much appreciated! Will these updated quirks be included in the mainline quirks at any point, so we don't have to load them as custom ones?

MattWestb, 2023-10-24:
Its complaning on the string: 
`<title>zha_quirks/ts0601_trv_siterwell.py at main · jacekk015/zha_quirks · GitHub</title>`
That shall not being in the quirk file so likely you have downloaded the HTTP code and not the quirk code (easy to do i have doing it some times), 
Use "RAW" mode and copy the text and putting it in your quirk file or download the repro as zip and copy the file you need and it shall working OK.

accessiblepixel, 2023-10-24:
@MattWestb 
Yes, I'd realised I'd done that, and wrote it later on in my last message, since apparently my brain was absent minded ;) We're all good now! <3

Are jacekk015's quirks ever going to be mainlined into the "main" quirks database for ZHA? Sorry if I don't know the right terminology - I wish I knew more of this stuff other than my clumsy bludgeoning to help more than reporting bugs, but here we are!

Hope life is treating you well, my friend.

MattWestb, 2023-10-24:
Great you was getting it  working then the "winter is coming" (GTO) and it needed for all of us !!
@jacekk015 and i was doing one try for nearly 2 years ago but was fling getting it thru so both have giving it up and we trying have support on his github that is very good but the best shall being that the maintainer is helping getting it in the ZHA for alls best.

accessiblepixel, 2024-01-01:
Reporting back after a couple of months (kinda forgot to come back and close this!).

@MattWestb  Using @jacekk015 's quirks from https://github.com/jacekk015/zha_quirks have worked perfectly fine since you pointed me in that direction, even when updating today to HA 12.05. 

Thanks @jacekk015 for working on them :)

Hope you both had a great new year. All the best.

closed 2024-01-01 by accessiblepixel