zigpy/zha-device-handlers #2648: [BUG] Quirk Issue / Question

ThiSaadeh, 2023-10-14:
### Bug description

I've recently bought a smart plug from Tuya which is labeled as: "_TZ3000_lotmgthb TS011F", it loads the standard "zhaquirks.tuya.ts011f_plug.Plug", I've noticed that the ENDPOINT and INPUT_CLUSTERS are pretty similar, except that the one in the quirk includes an INPUT as [9] but reading the remainder of the quirks within the same file I couldn't figure out what it stands for, I can write and load custom quirks, but writing my own custom quirk by using others as an example I land at the standard quirk mentioned above.

The issue is, the pug itself only switches on/off but everything else from turning off the led, child lock, etc doesn't work at all.

P.s.: I re-wrote the quirk using the skeleton of a different one that works at home, but honestly it looks exactly the same as the quirk it's already loading on it.

### Steps to reproduce

Adding the plug and going through the device settings page trying to change params on it doesnt work, but it doesnt throw errors on the log, it just doesnt affect the unit.

### Expected behavior

Led turn-off/turn-on, last state setup or child lock doesn't work on it.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
"signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x010a",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0702",
            "0x0b04",
            "0xe000",
            "0xe001"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZ3000_lotmgthb",
      "model": "TS011F"
    },
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.3",
    "dev": false,
    "hassio": true,
[... 578 lines cut ...]
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
```

TheJulianJES, 2023-10-14:
Change `CustomDevice` to `EnchantedDevice` (might need to import that). Then, restart HA with the new custom quirk and reconfigure the plug using the UI (or re-pair it).
Then, check if it works. Otherwise, I think your plug doesn't support those attributes for some reason.

ThiSaadeh, 2023-10-14:
Out of curiosity, what's the difference between Custom and Enchanted?

I just did what you recommended, I can see the quirk loaded into the system on the log:

Logger: zhaquirks
Source: components/zha/__init__.py:122
First occurred: 20:35:32 (1 occurrences)
Last logged: 20:35:32

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers

But removing/adding the plug still defaults to the standard Plug implementation, so looking at the re-added plug lists:

Quirk: zhaquirks.tuya.ts011f_plug.Plug

Instead of the one I wrote-up. Did I miss anything? Thanks!

ThiSaadeh, 2023-10-15:
It was missing Endpoint 242, I got it re-added using the custom quirk, but still couldnt get it to work with the Led Status / Last State / Child lock.

TheJulianJES, 2023-11-23:
> Custom and Enchanted?

Enchanted "casts a Tuya spell" when reconfiguring clusters (so during pairing) that Tuya devices need in order to work properly. (They try to detect if they're connected to a Tuya hub)

> But removing/adding the plug still defaults to the standard Plug implementation

Well, then you don't need a custom quirk, because an existing one already matches.
If child lock and so on don't work, then it's possible that the plug just doesn't support them in the first place.

How are you trying to configure those options? Using the clusters menu (Manage Zigbee device)?
If so, make sure to only change the upper text field. Always leave the lower text field completely empty.
Can you read the attributes on the Tuya OnOff clusters using the clusters UI?

closed 2024-05-28 by github-actions[bot]