zigpy/zha-device-handlers #2650: [Device Support Request] PC311-Z-TY Single-Phase Power Clamp Zigbee model 'TS0601' and manufacturer name '_TZE200_rks0sgb7

lepalom, 2023-10-15:
### Problem description

I would like to kindly ask to support this device:

[PC311-Z-TY Single-Phase Power Clamp Zigbee] (https://es.aliexpress.com/item/1005005931365212.html)

This device is the same commented here, in the [zigbee2mqtt project](https://github.com/Koenkk/zigbee2mqtt/issues/18734).

### Solution description

Fully integrated in HA.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00",
        "0xff66"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_rks0sgb7",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.10.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Andorra",
    "os_name": "Linux",
    "os_version": "5.10.0-25-armmp",
    "supervisor": "2023.10.0",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "24.0.6",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
    "sunspec": {
[... 335 lines cut ...]
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Custom quirk

In https://github.com/Koenkk/zigbee2mqtt/issues/18734 they have configured it.

vladii07, 2023-10-16:
I have this device and it is detected as PJ-1203A, which seems to look the same but does not have the same exposes, or at least not in the right order. Is there an easy way to over-ride a detected device with an external converter?

vladii07, 2023-10-16:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
divideBy1000: tuya.valueConverterBasic.divideBy(1000),
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE200_rks0sgb7',
},
],
model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
vendor: 'TuYa',
description: 'Bidirectional Energy Meter with 80A Current Clamp',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
whiteLabel: [ tuya.whitelabel('Tuya', 'PC311-Z-TY', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE200_rks0sgb7']), ],
// whiteLabel: [{vendor: 'EARU', model: 'PC311-Z-TY'}],
exposes: [ //only report data
e.ac_frequency(),
exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
exposes.numeric('current_AB', ea.STATE).withUnit('A').withDescription('Total Current'),
exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'),
exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'),
exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'),
],

meta:{
    tuyaDatapoints: [//only report data
      [104, 'ac_frequency', tuya.valueConverter.raw],//known ok
      [108, 'total_power_A', tuya.valueConverter.divideBy10],//known ok
      [111, 'total_power_B', tuya.valueConverter.divideBy10],//known ok
      [102, 'total_power_AB', tuya.valueConverter.divideBy10],
      [106, 'voltage', tuya.valueConverter.divideBy10],//known ok
      [107, 'current_A', tuya.valueConverter.divideBy1000],//known ok
      [110, 'current_B', tuya.valueConverter.divideBy1000],//known ok
      [105, 'current_AB', tuya.valueConverter.divideBy1000],//known ok
      [109, 'power_factor_A', tuya.valueConverter.raw],//known ok
      [112, 'power_factor_B', tuya.valueConverter.raw],//known ok
      [113, 'power_direction A', tuya.valueConverter.raw],
      [114, 'power_direction B', tuya.valueConverter.raw],
      [101, 'energy_forward_A', tuya.valueConverter.divideBy100],//known ok
      [103, 'energy_forward_B', tuya.valueConverter.divideBy100],//known ok
      [113, 'update_frequency' , tuya.valueConverter.raw],//might be ok?
    ],
},
};

module.exports = definition;

I took a stab at finding where the values are stored. I don't have reverse energy, so you would have to look for those (as I can't test that number). I am also not sure if it has power direction. Again, this is very rough, based on existing PJ-1203A, but it does what I need. I added it as an external converter, and it works ok.

KPfff, 2023-10-29:
Got in the email from vladii07 an attempt at mapping values with external converter. Not so proficient in github yet so apologies for double posting (answering to email too).

Tested it as I have exactly the device as linked above. 

My comments are added to initial  comments like this _"- comment"_

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
divideBy1000: tuya.valueConverterBasic.divideBy(1000),
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE200_rks0sgb7',
},
],
model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
vendor: 'TuYa',
description: 'Bidirectional Energy Meter with 80A Current Clamp',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
whiteLabel: [ tuya.whitelabel('Tuya', 'PC311-Z-TY', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE200_rks0sgb7']), ],
// whiteLabel: [{vendor: 'EARU', model: 'PC311-Z-TY'}],
exposes: [ //only report data
e.ac_frequency(),
exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power AB'),// - description corrected
exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
exposes.numeric('current_AB', ea.STATE).withUnit('A').withDescription('Total Current'),
exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
exposes.numeric('power_direction A', ea.STATE).withDescription('Power direction A 0/1 for forward/reverse'),
exposes.numeric('power_direction B', ea.STATE).withDescription('Power direction B 0/1 for forward/reverse'),
exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
exposes.numeric('update_frequency', ea.STATE).withUnit('sec').withDescription('Update frequency'),
],

meta:{
tuyaDatapoints: [//only report data
[104, 'ac_frequency', tuya.valueConverter.raw],//known ok - not ok. always 0
[108, 'total_power_A', tuya.valueConverter.raw],//known ok - no divide needed, corrected to raw
[111, 'total_power_B', tuya.valueConverter.raw],//known ok - no divide needed, corrected to raw
[102, 'total_power_AB', tuya.valueConverter.divideBy10],//not ok, always 0
[106, 'voltage', tuya.valueConverter.divideBy10],//known ok - ok
[107, 'current_A', tuya.valueConverter.divideBy1000],//known ok - ok
[110, 'current_B', tuya.valueConverter.divideBy1000],//known ok- ok
[105, 'current_AB', tuya.valueConverter.divideBy1000],//known ok - ok with a ~second delay
[109, 'power_factor_A', tuya.valueConverter.raw],//known ok - ok
[112, 'power_factor_B', tuya.valueConverter.raw],//known ok - ok
[113, 'power_direction A', tuya.valueConverter.raw],//known ok - not ok. seems to show frequency as value is always 50
[114, 'power_direction B', tuya.valueConverter.raw],//known ok - not ok. always null
[101, 'energy_forward_A', tuya.valueConverter.divideBy100],//known ok - seems ok
[103, 'energy_forward_B', tuya.valueConverter.divideBy100],//known ok - seems ok
[113, 'update_frequency' , tuya.valueConverter.raw],//might be ok? - not ok. shows null
],
},
};

module.exports = definition;`

KPfff, 2023-10-29:
Posted in github too. Apologies don't know how it works exactly with coms
in gihub.

Tested it as I have exactly the device -
https://es.aliexpress.com/item/1005005931365212.html

My comments are added to initial  comments like this _"- comment"_

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
divideBy1000: tuya.valueConverterBasic.divideBy(1000),
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE200_rks0sgb7',
},
],
model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
vendor: 'TuYa',
description: 'Bidirectional Energy Meter with 80A Current Clamp',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
whiteLabel: [ tuya.whitelabel('Tuya', 'PC311-Z-TY', 'Bidirectional Energy
Meter with 80A Current Clamp', ['_TZE200_rks0sgb7']), ],
// whiteLabel: [{vendor: 'EARU', model: 'PC311-Z-TY'}],
exposes: [ //only report data
e.ac_frequency(),
exposes.numeric('total_power_A',
ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B',
ea.STATE).withUnit('W').withDescription('Total power B'),
exposes.numeric('total_power_AB',
ea.STATE).withUnit('W').withDescription('Total power AB'),// - description
corrected
exposes.numeric('voltage',
ea.STATE).withUnit('V').withDescription('Voltage'),
exposes.numeric('current_A',
ea.STATE).withUnit('A').withDescription('Current A'),
exposes.numeric('current_B',
ea.STATE).withUnit('A').withDescription('Current B'),
exposes.numeric('current_AB',
ea.STATE).withUnit('A').withDescription('Total Current'),
exposes.numeric('power_factor_A',
ea.STATE).withUnit('%').withDescription('Instantaneous measured power
factor A'),
exposes.numeric('power_factor_B',
ea.STATE).withUnit('%').withDescription('Instantaneous measured power
factor B'),
exposes.numeric('power_direction A', ea.STATE).withDescription('Power
direction A 0/1 for forward/reverse'),
exposes.numeric('power_direction B', ea.STATE).withDescription('Power
direction B 0/1 for forward/reverse'),
exposes.numeric('energy_forward_A',
ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
exposes.numeric('energy_forward_B',
ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
exposes.numeric('update_frequency',
ea.STATE).withUnit('sec').withDescription('Update frequency'),
],

meta:{
tuyaDatapoints: [//only report data
[104, 'ac_frequency', tuya.valueConverter.raw],//known ok - not ok. always 0
[108, 'total_power_A', tuya.valueConverter.raw],//known ok - no divide
needed, corrected to raw
[111, 'total_power_B', tuya.valueConverter.raw],//known ok - no divide
needed, corrected to raw
[102, 'total_power_AB', tuya.valueConverter.divideBy10],//not ok, always 0
[106, 'voltage', tuya.valueConverter.divideBy10],//known ok - ok
[107, 'current_A', tuya.valueConverter.divideBy1000],//known ok - ok
[110, 'current_B', tuya.valueConverter.divideBy1000],//known ok- ok
[105, 'current_AB', tuya.valueConverter.divideBy1000],//known ok - ok with
a ~second delay
[109, 'power_factor_A', tuya.valueConverter.raw],//known ok - ok
[112, 'power_factor_B', tuya.valueConverter.raw],//known ok - ok
[113, 'power_direction A', tuya.valueConverter.raw],//known ok - not ok.
seems to show frequency as value is always 50
[114, 'power_direction B', tuya.valueConverter.raw],//known ok - not ok.
always null
[101, 'energy_forward_A', tuya.valueConverter.divideBy100],//known ok -
seems ok
[103, 'energy_forward_B', tuya.valueConverter.divideBy100],//known ok -
seems ok
[113, 'update_frequency' , tuya.valueConverter.raw],//might be ok? - not
ok. shows null
],
},
};

module.exports = definition;`

KPfff, 2023-10-29:
[103, 'energy_forward_B', tuya.valueConverter.divideBy100], - tested by comparing to another device, OK

KPfff, 2023-10-31:
**Details:**
- I own double clamp Zigbee version from here https://es.aliexpress.com/item/1005005931365212.html
**- Reporting tested and working on my local mains!** 
- Checked multiple DPs to find right ones, didn't find any more DPs with any values reported
- Reverse energy tested by installing clamps backwards
- Power readings goes negative with reverse load
- Summed "AB" values are coming with a delay and are not always 1+1 accurate
- Reporting frequency seems to be fixed at once per second

**NOTE:** Readings accuracy is directly impacted by distance from other wires!!! One clamp was left without wire passing through, just wire next to it and it was showing 7W consumption.

**Working converter below:**
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;
const {Buffer} = require('buffer');

const valueConverter = {
divideBy10: tuya.valueConverterBasic.divideBy(10),
divideBy100: tuya.valueConverterBasic.divideBy(100),
divideBy1000: tuya.valueConverterBasic.divideBy(1000),
};

const definition = {
fingerprint: [
{
modelID: 'TS0601',
manufacturerName: '_TZE200_rks0sgb7',
},
],
model: 'TS0601_Bidirectional_Energy_Meter_with_80A_Current_Clamp',
vendor: 'TuYa',
description: 'Bidirectional Energy Meter with 80A Current Clamp',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime,
configure: tuya.configureMagicPacket,
whiteLabel: [ tuya.whitelabel('Tuya', 'PC311-Z-TY', 'Bidirectional Energy Meter with 80A Current Clamp', ['_TZE200_rks0sgb7']), ],
// whiteLabel: [{vendor: 'EARU', model: 'PC311-Z-TY'}],
exposes: [ //only report data
e.ac_frequency(),
exposes.numeric('total_power_A', ea.STATE).withUnit('W').withDescription('Total power A'),
exposes.numeric('total_power_B', ea.STATE).withUnit('W').withDescription('Total power B'),
exposes.numeric('total_power_AB', ea.STATE).withUnit('W').withDescription('Total power AB'),
exposes.numeric('voltage', ea.STATE).withUnit('V').withDescription('Voltage'),
exposes.numeric('current_A', ea.STATE).withUnit('A').withDescription('Current A'),
exposes.numeric('current_B', ea.STATE).withUnit('A').withDescription('Current B'),
exposes.numeric('current_AB', ea.STATE).withUnit('A').withDescription('Total Current'),
exposes.numeric('power_factor_A', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor A'),
exposes.numeric('power_factor_B', ea.STATE).withUnit('%').withDescription('Instantaneous measured power factor B'),
exposes.numeric('energy_reverse_A', ea.STATE).withUnit('kWh').withDescription('Total energy A reverse'),
exposes.numeric('energy_reverse_B', ea.STATE).withUnit('kWh').withDescription('Total energy B reverse'),
exposes.numeric('energy_reverse_AB', ea.STATE).withUnit('kWh').withDescription('Total energy AB reverse'),
exposes.numeric('energy_forward_A', ea.STATE).withUnit('kWh').withDescription('Total energy A forward'),
exposes.numeric('energy_forward_B', ea.STATE).withUnit('kWh').withDescription('Total energy B forward'),
exposes.numeric('energy_forward_AB', ea.STATE).withUnit('kWh').withDescription('Total energy AB forward'),
],

meta:{
tuyaDatapoints: [//only report data
[113, 'ac_frequency', tuya.valueConverter.raw],
[108, 'total_power_A', tuya.valueConverter.raw],
[111, 'total_power_B', tuya.valueConverter.raw],
[9, 'total_power_AB', tuya.valueConverter.raw],
[106, 'voltage', tuya.valueConverter.divideBy10],
[107, 'current_A', tuya.valueConverter.divideBy1000],
[110, 'current_B', tuya.valueConverter.divideBy1000],
[105, 'current_AB', tuya.valueConverter.divideBy1000],
[109, 'power_factor_A', tuya.valueConverter.raw],
[112, 'power_factor_B', tuya.valueConverter.raw],
[102, 'energy_reverse_A', tuya.valueConverter.divideBy100], 
[104, 'energy_reverse_B', tuya.valueConverter.divideBy100],
[2, 'energy_reverse_AB', tuya.valueConverter.divideBy100],
[101, 'energy_forward_A', tuya.valueConverter.divideBy100],
[103, 'energy_forward_B', tuya.valueConverter.divideBy100],
[1, 'energy_forward_AB' , tuya.valueConverter.divideBy100],
],
},
};

module.exports = definition;`

[image]

lepalom, 2023-11-01:
@KPfff Can you guide me to test this configuration? I have updated HA but I still have the old configuration.

KPfff, 2023-11-03:
@lepalom, Are you familiar with setting up external converters? 

1) You need to File editor - I use the "File editor" from Add-on Store
2) Open file editor and navigate to /config/zigbee2mqtt folder
3) Create file with .js extension. E.g., I named mine "earu-pc311-z-ty.js" (IMPORTANT - don't use capital letters in the file name!)
4) Once file is created click on it in editor and copy paste converter from above, save the file.
5) Go to Zigbee2MQTT add-on => Settings => External converters => Add the file with the same name, e.g. "earu-pc311-z-ty.js"
6) Restart Zigbee2MQTT add-on

If the device is still reporting old values then try removing it and adding it again. (Worst case force remove and pair once again)

Note: Mind you that any historical data will not be brought over - it is like adding new device.

Hope it helps :)

jmuf, 2023-11-03:
@KPfff: many thanks for this investigation. I guess that part of the issue is that not everybody is using `zigbee2mqtt`. To convert your findings to a `zha-device-handlers` "quirk" means writing a small python module similar to [ts0601_din_power.py](zhaquirks/tuya/ts0601_din_power.py). 
While the TUYA datapoint identifiers hopefully are the same (and can be copied from your code), the structure is quite different.

jmuf, 2023-11-04:
Work-in-progress "quirk":
* gets loaded,
* Zigbee "manage device" sees the custom cluster,
* **but** does not load any data ( all values are null)
   * even for stuff like 'frequency' (0x71) which does not get translated
* `_update_attribute()` is not even called (no logging)

[ts0601_sp_clamp_power.py.txt](https://github.com/zigpy/zha-device-handlers/files/13257870/ts0601_sp_clamp_power.py.txt)

robertstumberger, 2023-11-05:
Hi guys,
I have the same meter, PC311-Z-TY Single-Phase Power Clamp Zigbee model 'TS0601' and manufacturer name '_TZE200_rks0sgb7 but with two Clamps...
[image]
And of course I would ask for your help ;-)
And I use ZHA.
i need help how "quirk"

lepalom, 2023-11-05:
@KPfff thanks a lot. Well, my problem is that I'm not using zigbee2mqtt. Now I have understood the problem. I have seen that there are like two big families to manage a zigbee devices: ZHA integration and zigbee2mptt.

I have been using ZHA and I would like to test zigbee2mqtt, but I have to migrate all the devices. I will see.

robertstumberger, 2023-11-05:
Ok,
I managed to give a "quirk" to work, but incorrectly, I assume
[image]
What am I doing wrong? Please help :-(

robertstumberger, 2023-11-05:
> Work-in-progress "quirk":
>
>     * gets loaded,
>
>     * Zigbee "manage device" sees the custom cluster,
>
>     * **but** does not load any data ( all values are null)
>
>       * even for stuff like [...]

@jmuf Are you working on  "quirk" for ZHA?

jmuf, 2023-11-06:
I had a look but am stuck, no idea whether the "attributes" from @KPfff have been correctly put in place in this code, or whether something basic is missing (this is my first attempt at a "quirk").

robertstumberger, 2023-11-07:
Do we have someone who could check - validate this?
Is there any other way I can help?
As you can see from the picture above, the "quirk" was accepted, but unfortunately nothing happened, even after a reboot.

jmuf, 2023-11-11:
Attached is a new version of the "quirk" (apparently need some offset to "translate" attributes numbers between zigbee2mqtt and ZHA), as usual need to rename to `something.py` (but cannot upload as such).
* this more-or-less works for me (YMMV), although I haven't checked all values for plausibility.
* it (ab?)uses the "phase B" attributes to report the values for the "Circuit 2" clamp (but this is 
[ts0601_sp_clamp_power.py.txt](https://github.com/zigpy/zha-device-handlers/files/13325411/ts0601_sp_clamp_power.py.txt)
a single-phase meter)
* it does **not** report the energy per circuit (which arguably is the most interesting measurement..), just the total

If somebody has an idea/example on how to address the last point, I'd be happy for hints (instantiate a new "endpoint"?). This might also allow to not use "Phase B".

robertstumberger, 2023-11-13:
Hi,
I tested a new script, Reconfiguring device,
[image]
But again it doesn't show me the entities...

[image]

KPfff, 2023-11-13:
@robertstumberger, Apologies I see I have added to the confusion - I think I thought that this is one of the threads on zigbee2mqtt project when in fact this is for ZHA. :( 

As far as mappings go I have been using this device with custom converter ever since with no issues. 
I still miss configuration options (like setting intervals, accuracy etc.) but at least reported values are mapped correctly and are usable :) 

I haven't used ZHA integration but apparently custom quirks are formatted completely differently - so unless you can translate my mappings to custom quirk then I can't be of any help here.

didius62, 2023-11-17:
Hi all ,
I have just integrated this module (PC311) into Z2M/Jeedom with the external converter.
[image]

I only connect the circuit1 , i don't understand the value of energy reverse (displayed 5.72 KWH) , this value is wrong for me!

Didier

jmuf, 2023-11-17:
> Hi, I tested a new script, Reconfiguring device, [image] But again it doesn't show me the entities...

@robertstumberger : this is not so bad, the device has been recognized - please make sure you take the actual latest version of [ts0601_sp_clamp_power.py](https://github.com/zigpy/zha-device-handlers/files/13327868/ts0601_sp_clamp_power.py.txt) , an earlier version indeed did not report any standard attribs back to HA.

MattWestb, 2023-11-17:
Reconfigure of "virtual  cluster" is not working then the commands / attributes is not sent by the endpoint / cluster they is sent by the tuya MCU on the tuya cluster.

robertstumberger, 2023-11-17:
Hej @jmuf , 
Thank you for your help, it worked... you're the best :-)
That's a huge improvement
[image]

Maybe we will manage to add both Clamps? :-)

didius62, 2023-11-18:
Hi , 
I am not under HA !
Didier

robertstumberger, 2023-11-19:
Hej @jmuf ,
I checked a little what I get and so on:

and if i go:
[image]
i get correct values:
[image]
[image]
[image]
[image]
and so on...
for all of them
And the Power (on the right side) it's the same as power_A (on the left side) 
[image]

this means, that the declarations for these entities are defined somewhere else and not in this Quirk (ts0601_sp_clamp_power.py)
I can't find these declarations in the entire script
how to get these entities?

jmuf, 2023-11-19:
As you've seen, the `Tuya_EnergyMeter_2Clamp` class "knows" how to read the interesting attributes from the device (based on the work by @KPfff ). It then stuffs data into two helper classes:
* `PowerMeasurement_2Clamp` - this is derived from `ElectricalMeasurement`, and ZHA seems to look at the attributes of that class for "Voltage", "AC frequency", "Current" etc reported in the right panel. And it indeed seem to ignore the measurements for "Phase B" (which is where the data from the second clamp goes.. for now). Ideally we'd find a way to create a second instance of  `ElectricalMeasurement` and report the second clamp data there.
* `Metering_2Clamp` - derived from `Metering`, and ZHA looks at that for "Summation delivered" (which should have the sum over both clamps)

benoitbarotin2, 2023-12-07:
Hi,

I bought the same module and install the quirks [ts0601_sp_clamp_power.py](https://github.com/zigpy/zha-device-handlers/files/13327868/ts0601_sp_clamp_power.py.txt).

But it doesn't show me the entities : 
[image]

@MattWestb : you wrote : _Reconfigure of "virtual cluster" is not working then the commands / attributes is not sent by the endpoint / cluster they is sent by the tuya MCU on the tuya cluster._

I don't undestand what i have to do ?

OwlBawl, 2023-12-09:
> @robertstumberger : this is not so bad, the device has been recognized - please make sure you take the actual latest version of [...]

The quirk is partly functional...
Only 2nd clamp is showing. Half success 
Would be great if someone could fix it to show full info for both clamps

jmuf, 2023-12-10:
Here is a new version that has both clamps individually (as well as the total)
[ts0601_sp_clamp_power.py.v2.txt](https://github.com/zigpy/zha-device-handlers/files/13626919/ts0601_sp_clamp_power.py.v2.txt)

However, I haven't yet found a way to give them different entity names (so you'll get 3x "xyz Summation delivered" and so on - you can always rename them yourself, of course). Pointers to some similar device than can do this would be welcome...

robertstumberger, 2023-12-14:
Hi @jmuf 
I I used your new version and I'm impressed, fantastic work. I'll do some testing and report back, but it looks promising
As you mentioned, you just need to rename the entities and you're done
[image]

robertstumberger, 2023-12-14:
After the first tests, I can say that it does not measure bidirectionally 
[image]

jmuf, 2023-12-14:
@robertstumberger - thanks for the test. Could you edit the "quirk" and change ` t.uint16_t` to `t.int16s` in the following?

```
     62         TOTAL_POWER_AB_ATTR: ("power_both", t.uint16_t, True),
     63         TOTAL_POWER_A_ATTR: ("power_A", t.uint16_t, True),
     64         TOTAL_POWER_B_ATTR: ("power_B", t.uint16_t, True),
```
(per [above](https://github.com/zigpy/zha-device-handlers/issues/2650#issuecomment-1787932773) "Power readings goes negative with reverse load"..).

There also should be some "Sum received" attributes somewhere that sum up outgoing kWh...

jmuf, 2023-12-17:
Latest version (with the above "negative" power for outgoing load, also some cleanups): [ts0601_sp_clamp_power.py.v3.txt](https://github.com/zigpy/zha-device-handlers/files/13695705/ts0601_sp_clamp_power.py.v3.txt)

The issue of metric names (3x "Power") is still open..

DEDEit, 2024-03-16:
Hi everyone, I tried with various versions of the quirk (including V3) but once I restarted Home Assistant all the ZHA integration no longer works.
Below are the logs, can you help me?
Thank you

[image]
[image]

lebeck69, 2024-07-02:
Originally I was in same situation than robertstumberger : using ZHA, PC311-Z-TY .. and only had 2 values (LQI and RSSI)
I learned about Quirks .. which I knew nothing .. already thx ;)
I directly used V3.

But numbers does seems good on my side.

I've put one on a working oven and the other one on a fridge.
Either values are null.. or equals and very smalls (not over 7W)

Questions : 

1. I have 120A clamps .. they seems huge compare to the oven electrical oven, is that normal ?
2. All infos are in tripple .. what are there for ? clamp 1, clamp 2 .. and device itself ?

[image]

didius62, 2024-10-02:
Hi all ,
i have a question about this device about the traffic , This device communicates constantly, I wonder if it doesn't load the Zigbee network too much!
do you know a parameter to read the values ​​every minute for example (like a linkee module)
[image]

many thanks
DD

SabatoArdolino, 2024-10-07:
Hi to all
I've configure today ZHA and I try to setup the tuya clamp but I can understand where and how I cam install them with quirks. I've configure in my configuration.yaml 

#### ZHA ###
zha:
  custom_quirks_path: /config/custom_zha_quirks/
  enable_quirks: true

in the path I've uploaded the .py file, repairing the devide and restart HA but I see only 2 entities. 
Can explain How can i solve my issue ? 
Thanks

rabievdm, 2024-12-23:
@jmuf Was there any plans to merge [https://github.com/zigpy/zha-device-handlers/files/13695705/ts0601_sp_clamp_power.py.v3.txt](url) back into the main dev branch and also any luck in fixing the naming, specifically the A+B portion?

jmuf, 2024-12-24:
> @jmuf Was there any plans to merge [https://github.com/zigpy/zha-device-handlers/files/13695705/ts0601_sp_clamp_power.py.v3.txt](url) back into the main dev branch

There are currently two "competing" pull requests open
* "mine" at https://github.com/zigpy/zha-device-handlers/pull/2870 (lacking some tests? would need to read up on these.. If sb knows how to add these, please feel free to contribute some testcases)
* @jeverley 's at https://github.com/zigpy/zha-device-handlers/pull/2961 (looked more promising but appears to be stuck in some discussion on "negative" values..)

> and also any luck in fixing the naming, specifically the A+B portion?

looks like that would need support in the base code (not happening?), or could possibly unclear how) be addressed if we go from "quirk" to "integrated in ZHA" (per  https://github.com/zigpy/zha-device-handlers/issues/2840 )

jeverley, 2024-12-24:
We should be able to resolve the naming issue using the new quirks v2 format.

I'm planning to reimplement mine using quirks v2 in the new year, exposing config options in the UI that should allow flipping the negative values for people who have that issue.

jmuf, 2025-01-26:
[..]
> There are currently two "competing" pull requests open
> 
> * "mine" at [Add Tuya 2-clamp power meter `_TZE200_rks0sgb7` #2870](https://github.com/zigpy/zha-device-handlers/pull/2870) (lacking some tests? would need to read up on these.. If sb knows how to add these, please feel free to contribute some testcases)

FWIW, the update to homeassistant 2025.1 broke "my" quirk, will close the pull request. Hope that @jeverley 's version can move forward.

closed 2025-08-16 by github-actions[bot]