zigpy/zha-device-handlers #2651: [Device Support Request] TS0202 by _TZ3000_mg4dy6z6 - Motion Sensor

iRapsody, 2023-10-16:
### Problem description

I bought [this](https://www.aliexpress.com/item/1005005893945826.html) motion sensor from Aliexpress. It seems to be detected and "successfully" configured by ZHA but it does not detect any movement. I don't know why there are two sensors detected ( Motion and Opening ) because in product description says that it only has motion detection.
[image]

### Solution description

Make sensor available for automations.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "output_clusters": [
        "0x0006",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TZ3000_mg4dy6z6",
  "model": "TS0202",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Bucharest",
    "os_name": "Linux",
    "os_version": "6.1.45",
    "supervisor": "2023.10.0",
    "host_os": "Home Assistant OS 10.5",
[... 397 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

iRapsody, 2023-10-20:
Found [this](https://github.com/Koenkk/zigbee2mqtt/issues/9757#issuecomment-1242909728) in zigbee2mqtt git, maybe will help someone to figure it out

Nemesis24, 2023-11-24:
the same for me

jonathh21, 2023-12-31:
I have a few of these devices and they are all over the place.

Sometimes they work ok for ages, then they just stop.

Today I had several stop and had to restart just the zigbee2mqtt service and they all started working again.

Sometimes I see the light flash to indicate it has seen up me no packet arrives and the last seen isn't updated.

The only thing that is consistent is how unreliable these devices are.

MacDada, 2024-01-25:
I've just installed it yesterday -> detection works for me with no problems.

I also have the `Opening` sensor entity, that is constantly `Closed`.
And the `Identify` diagnostic entity, that does nothing upon pressing.
Those two I think are a mistake to be present, but not a big issue -> I just disabled them.

[image]

[image]

itsolon, 2024-02-07:
Im facing similar issues with TS0202 "group" for me i try to write a quirk at the moment and first success was that i could remove that unneccessary sensor. FYI i will enclose the link where i reported everything
if you want you can try 
[https://community.home-assistant.io/t/tz3000-kmh5qpmb-ts0202-information-on-how-to-tell-what-each-attribute-of-entity-does/670243](url)

zeppu, 2024-02-28:
Has anyone managed to successfully change the off wait time for this? I've tried several quirks with no luck.

wsegatto, 2024-04-14:
Has anyone been able to make it work with ZHA?

MacDada, 2024-04-14:
> Has anyone been able to make it work with ZHA?

https://github.com/zigpy/zha-device-handlers/issues/2651#issuecomment-1909931515 ??

wsegatto, 2024-04-14:
> > Has anyone been able to make it work with ZHA?
> 
> 
> 
> https://github.com/zigpy/zha-device-handlers/issues/2651#issuecomment-1909931515 ??

Yeah, I got confused. So your "Opening" is at a fixed value but your detection does work? Mine is always stuck at "Detected". And this happens for two units I own. 😕

MacDada, 2024-04-14:
> your detection does work?

It used to work back when I was using ZHA. It's still working, now that I'm using Zigbee2MQTT. It does not have the `opening` entity anymore, looks like this:
[image]

i-segura, 2024-09-01:
Hello, I have two of these same devices. Also reports as described [here](https://github.com/zigpy/zha-device-handlers/issues/2651#issuecomment-1909931515).

Is there a way to change the "keep time" or "wait time" at least manually via device ZHA administration? I've tried reading several zigbee attributes but they all give me an error.

closed 2025-08-16 by github-actions[bot]