zigpy/zha-device-handlers #2652: [Device Support Request] Circuit Breaker 63A with Power Monitoring and Threshold TS011F | _TZ3000_lepzuhto

giball, 2023-10-16:
### Problem description

Electrical measurements in this device not working properly - current and summation delivered (kWh) are too high.

### Solution description

Please correct the current multiplier

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your
[image]
 media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_lepzuhto",
  "model": "TS011F",
  "class": "zigpy.device.Device"
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.3",
[... 635 lines cut ...]
  }
}]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Pajuris, 2024-01-17:
Tongou TO-Q-SY2-163JZT - TS011F _TZ3000_cayepv1a.

Wrong amperes (always show 62-63), wrong measurements (kWh), no threshold.

apolloathome, 2024-01-20:
Same problem with Tongou TO-Q-SY2-163JZT - TS011F _TZ3000_cayepv1a

[image]

Ampere is x 1000

Cumulated power is x 1000

dartvader1234, 2024-02-18:
TS011F by _TZ3000_cayepv1a 
Same problem but different values:

- Current in ampere is x1000 so it indicates 10000 Amperes with 10 Amperes load
- The cumulated "summation delivered" in kWh is x100 (not x1000), so 200 kWh indicated with 2 kWh.
- Power in Watt is ok.

MnM001, 2024-03-16:
If we are talking abut the same device (TO-Q-SY2-163JZT) , and I think we are, the integration into ZHA needs a lot of work.
The wrong measurements needs to be fixed, however, there are lot of things missing from this device in ZHA. Zigbee2mqtt has all of them covered. 

As it is right now this device function as TO-Q-SY1 device, which is just a normal breaker with an ON/OFF button and displaying some power values.
The  TO-Q-SY2-163JZT is a different product in the sense that it had all the features of the TO-Q-SY1 device plus more:

- temperature protection
- over current protection
- over voltage protection
- under voltage protection
- high power protection
- timing

This is the zigbee2mqtt page for this device:

https://www.zigbee2mqtt.io/devices/TO-Q-SY2-163JZT.html

Will be good to see the same kind of functionality in ZHA too because with out that it just acts like a dumb TO-Q-SY1 device offering no protection at all.

liouma, 2024-03-29:
After some more testing, here are more info.
1/ about the custom quirk from this thread: https://github.com/zigpy/zha-device-handlers/issues/2364 I can say that Current, Voltage and Power values are correct. I measured current and voltage. Then P=UI rule is verified with the displayed power value. (did not test yet summation delivered). -> the custom quirk is good, that would be great if someone could adapt it here!

2/ about the values measured by _TZ3000_cayepv1a:
Values displayed in HA are:
- Current: 194A
- voltage 237v
- power 23W

Value measured with an external tool (simple AC clamp meter) = 0.130A

Let's simplify for now and says that there is a x1000 factor. There will still be a huge difference between 130 and 194. Can it be explained by the relatively low current and the method (AC clamp)?? Later I'll connect an amperemeter to get a more precise value.
But already we can say that the info displayed by HA are:
- current: mA and not A -> need to do a value = value/1000
- power should be 46 (0.194x237). Correction needed: value=value x 2

Unfortunately I'm not able to write a custom quirk. I hope those info will help someone write one on the basis of this one: [AC clamp meter](https://github.com/zigpy/zha-device-handlers/issues/2364#issuecomment-1537199889)

For all others: you can make a simple correction using templates

liouma, 2024-03-29:
> * The cumulated "summation delivered" in kWh is x100 (not x1000), so 200 kWh indicated with 2 kWh.
> * Power in Watt is ok.
@dartvader1234: can you please double check your power value? are your sure you do not read a value/2 ??
Are you using 110v or 220v ? -> if you are in 110v that could explain why your value is correct and mine /2

apolloathome, 2024-03-29:
> TS011F by _TZ3000_cayepv1a Same problem but different values:
> 
> * Current in ampere is x1000 so it indicates 10000 Amperes with 10 Amperes load
> * The cumulated "summation delivered" in kWh is x100 (not x1000), so 200 kWh indicated with 2 kWh.
> * Power in Watt is ok.

You are rigth, me wrong, cumulated power is  x 100

mortiy, 2024-04-02:
Hi,
Just want to add that there are two more fingerprints for this device:
`_TZ3000_cayepv1a` and  `_TZ3000_qystbcjg`

Taken from here:
https://github.com/Vorta/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L4805

apolloathome, 2024-04-04:
Thank you mortiy. 👍

Milhauz666, 2024-04-21:
I have _TO-Q-SY2-JZT - TS011F_ __TZ3000_cayepv1a_ with `SimpleDescriptor(endpoint=1, profile=260, device_type=266, device_version=1, input_clusters=[0, 3, 4, 5, 6, 1794, 2820, 1026, 57344, 57345], output_clusters=[25, 10])`, which is using the default quirk "_zigpy.device.Device_", 

There is a quirk for similar device [_TZ3000_qeuvnohg](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_plug.py#L1246) having the same descriptor, but with slightly reordered clusters:
```
# <SimpleDescriptor endpoint=1 profile=260 device_type=266
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 1026, 1794, 2820, 57344, 57345]
            # output_clusters=[10, 25]>
```

I suppose tweaking this quirk should help to identify the __TZ3000_cayepv1a_ device properly?

neildsb, 2024-05-23:
Hi, please excuse the simple question, I too have this device (TO-Q-SY2-163JZT) and am using ZHA, and have noticed the lack of exposed sensors compared to Zigbee2mqtt (within this thread), will this be updated/rectified overtime, are there manual interventions/configuration changes needed to identify the current product? I am primarily using the switching function (noted the wrong current values) the other sensors are nice to have. Thanks in advance…

meremortals70, 2024-05-27:
Another vote for fixing this please.

apolloathome, 2024-05-31:
I can send to the developper team, for free to resolve this problem. If your are interested, please contact me

franortiz, 2024-08-16:
I create a custom quirk for this device [here](https://github.com/zigpy/zha-device-handlers/issues/3044#issuecomment-2294139989)

franortiz, 2024-09-20:
I updated the file to make use of the quirk v2 features to expose the attributes to HA entities
It works great except for some reason the entity names are not generated correctly and everything is suffixed with _none_xxx instead of the declared attribute name.
Also, the number entity generator is not reading correctly the mode attribute and its showing a slider instead of an input box [I already created an issue](https://github.com/zigpy/zha/issues/202)

franortiz, 2024-10-03:
Updated version with [quirks v2 support](https://gist.github.com/franortiz/3fd91c6fa0884f2e33fa7e9d57ad0930)

franortiz, 2024-11-06:
[Updated](https://gist.github.com/franortiz/3fd91c6fa0884f2e33fa7e9d57ad0930) support for HA 2024.11 and zigpy 0.71 quirks v2 key requirements

uvNikita, 2024-11-06:
> entity names are not generated correctly and everything is suffixed with _none_xxx instead of the declared attribute name.

@franortiz did you figure out why .number entity ids are generated with _none suffix and how to fix it? I have the dame issue here.

franortiz, 2024-11-06:
I had to remove the device and re-add it then the attributes were repopulated correctly. My guess is some kind of id that is generated with the old version.
My problem was with all the created entities, not only with .number, but as i said, fixed once i remove it and add it again.

[image]

The .number entities issue is related to the display of a slider instead of a box
[image]

daurnimator, 2025-08-08:
👍

davidswelt, 2026-10-31:
Can you please try [PR 5645](https://github.com/zigpy/zha-device-handlers/pull/5645), adding the device code to the list of supported devices, and report back if it works with your contactor?