zigpy/zha-device-handlers #2655: [BUG] AduroSmart scene switch not detecting scene button clicks

Daandeve, 2023-10-17:
### Bug description

When integrating the AduroSmart scene switch in zha the scene switches do not work.
The on/off switch works fine, but the other 3 (scene) switches do not work. Well sort of when one of the 3 scene buttons is pressed the battery percentage changes.

### Steps to reproduce

1. Integrate the device into zha
2. Click one of the 3 scene buttons

### Expected behavior

The switch should be detected as the scene switch and not the dimmer switch

### Screenshots/Video

[image]

[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4653, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0xc05e",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfccc"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfccc"
      ]
    },
    "2": {
      "profile_id": "0xc05e",
      "device_type": "0x03f2",
      "input_clusters": [
        "0x1000"
      ],
      "output_clusters": [
        "0x1000"
      ]
    }
  },
  "manufacturer": "AduroSmart Eria",
  "model": "ADUROLIGHT_CSC",
  "class": "zigpy.device.Device"
}
[... 529 lines cut ...]
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
2023-10-17 19:40:36.609 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0xF994, ClusterId=5, SrcAddr=0x489C, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=76, SecurityUse=<Bool.false: 0>, TimeStamp=8667961, TSN=0, Data=b'\x11\x98\x05\x94\xF9\xFF', MacSrcAddr=0x489C, MsgResultRadius=29)
2023-10-17 19:40:36.610 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x489C), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.Group: 1>, address=0xF994), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=5, data=Serialized[b'\x11\x98\x05\x94\xf9\xff'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=76, rssi=None)
2023-10-17 19:40:36.610 DEBUG (MainThread) [zigpy.zcl] [0x489C:1:0x0005] Received ZCL frame: b'\x11\x98\x05\x94\xf9\xff'
2023-10-17 19:40:36.610 DEBUG (MainThread) [zigpy.zcl] [0x489C:1:0x0005] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=152, command_id=5, *direction=<Direction.Server_to_Client: 0>)
2023-10-17 19:40:36.610 DEBUG (MainThread) [zigpy.zcl] [0x489C:1:0x0005] Decoded ZCL frame: Scenes:recall(group_id=0xF994, scene_id=255, transition_time=None)
2023-10-17 19:40:36.610 DEBUG (MainThread) [zigpy.zcl] [0x489C:1:0x0005] Received command 0x05 (TSN 152): recall(group_id=0xF994, scene_id=255, transition_time=None)
2023-10-17 19:40:36.611 DEBUG (MainThread) [zigpy.zcl] [0x489C:1:0x0005] No explicit handler for cluster command 0x05: recall(group_id=0xF994, scene_id=255, transition_time=None)
```

</details>

Daandeve, 2024-05-10:
This issue is still relevant, the switch does not work properly with HA: 2024.5.2

Vinno97, 2025-04-06:
@TheJulianJES  Sorry for reviving such an old issue, but I also believe this issue is incorrectly closed by github-actions. I'm running into the same issues with this remote. It appears that the remote is "compliant" in reporting a scene change. However, there is handler for "recalling scenes", which is what the "scene" buttons trigger. As far as I'm aware, HA doesn't actually use Zigbee scenes (right?) so would the best solution be to to remap this to a button action? 

I'd be willing to help implement this if desired.

fabio-s-franco, 2025-05-17:
@TheJulianJES @Vinno97 
I created a new quirk (under the aduro folder). The exiting quirk is for the dimmer switch, which although apparently almost identical, behaves very differently than the scene switch.

This dimmer model is `Adurolight_NCC`, the new quirk is for model `Adurolight_CSC`

Vinno97, 2025-05-23:
Many thanks, Fabio! I just tried the new quirk and it works great :)
For me, my troubles are completely fixed by the PR.

Out of curiosity: you write that there are other events that these buttons send, what are those? Are those button press/release events?

Vinno97, 2025-05-23:
Straight after commenting that, I was spamming some buttons and the devices appeared to stop working.

The only thing I can see is that the log is spammed with messages like this:
```
Event is not JSON serializable: <Event zha_event[L]: device_ieee=00:15:8d:00:05:8c:ef:99, device_id=bb43b573c27708aad1be05b4cfa803ce, unique_id=00:15:8d:00:05:8c:ef:99:1:0xfccc, endpoint_id=1, cluster_id=64716, command=button_4_long_press, args=button=button_4, press_type=remote_button_long_press, args=b'\x01\x03\x00', params=, params=>: Type is not JSON serializable: bytes
```
I'll check again in the morning

Vinno97, 2025-05-27:
Debouce feels somewhat long when playing around with the remote but after placing it on the wall I don't think I've ever noticed it. I can check this weekend (sorry, barely home this week) if other values feel nicer without any false clicks.

And yeah, think I accidentally unpaired it haha.

Any thoughts on that log line btw?

Vinno97, 2025-08-11:
Hi Fabio, I completely forgot I was running an unmerged quirk until it suddenly stopped working after upgrading HA to 2025.8.

When I click on a button, this warning shows up in my logs.

```
Logger: zigpy.listeners
Source: runner.py:156
First occurred: 22:00:22 (1 occurrence)
Last logged: 22:00:22

Matcher checkin() and command ZCLHeader(frame_control=FrameControl<0x11>(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 0>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=68, command_id=0, *direction=<Direction.Client_to_Server: 0>) b'\x00\x00\x00' are incompatible
```

Can I be of any help getting this quirk properly merged into the repo?

fabio-s-franco, 2025-08-14:
I am not sure now, but I think this device sends a command with value "0" that is not really supported, alongside the button command. It basically triggers two events one is like any button clicked, and button X was clicked.

It's safe to ignore it (assuming I recall it correctly), though I am not sure there is a way to suppress this, given my previously inexistent experience writing ZHA drivers.

I don't really know what's the hold up to get it merged. But it's still waiting for review.