zigpy/zha-device-handlers #2657: Warning on HA restart

hageson, 2023-10-17:
### Bug description

Logger: homeassistant.components.zha.core.cluster_handlers
Source: components/zha/core/cluster_handlers/__init__.py:537
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 2:18:19 PM (2 occurrences)
Last logged: 2:18:19 PM

[0x71DE:1:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

### Steps to reproduce

restart HA

### Expected behavior

No warning

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2023-10-17:
Post the diagnostic fro the device `0x71DE` from its device card (likely one SonOff repeater).

def1149, 2023-12-29:
I  see this too for SonOff Dongle Es used as routers (with router firmware). I have two, one at each end of the house powered by USB wall PS w/  Li-Po rechargeable batteries as power fail backups

2023-12-29 07:39:08.165 WARNING (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xF9C9:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected
2023-12-29 07:39:08.166 WARNING (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x57D6:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.0b2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "6.1.63-haos-raspi",
    "supervisor": "2023.12.1",
    "host_os": "Home Assistant OS 11.3.rc1",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "keymaster": {
      "version": "v0.0.87",
      "requirements": []
    },
    "alexa_media": {
      "version": "4.8.1",
      "requirements": [
        "alexapy==1.27.10",
        "packaging>=20.3",
        "wrapt>=1.14.0"
      ]
    },
    "chime_tts": {
      "version": "v0.10.2",
      "requirements": [
        "pydub"
      ]
    },
    "frigate": {
      "version": "5.0.0-beta.3",
      "requirements": [
        "pytz==2022.7"
      ]
    },
    "nws_alerts": {
      "version": "4.1",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "awnet_local": {
      "version": "1.2.0",
      "requirements": []
    },
    "mail_and_packages": {
      "version": "0.3.19",
      "requirements": [
        "beautifulsoup4",
        "Pillow>=9.0"
      ]
    },
[... 773 lines cut ...]
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

MattWestb, 2023-12-29:
@def1149 Your routers is having bad configured firmware:
```
2023-12-29 07:39:08.165 WARNING (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xF9C9:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected
2023-12-29 07:39:08.166 WARNING (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0x57D6:2:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected
```
The system is reading the color temperature and the device is replaying 0 = not valid in Zigbee standard.
Live with it or complain to the maker of the router firmware.

closed 2024-07-03 by github-actions[bot]