zigpy/zha-device-handlers #2662: [BUG] Paulmann 501.34 Wall Switch (sw_build_id (id: 0x4000) 2.7.6_r25) cannot be used in HASSIO automations

drueppela, 2023-10-20:
### Bug description

This is my first bug report. Please be lenient

---

References:

https://github.com/zigpy/zha-device-handlers/pull/1678#issuecomment-1771660070
https://community.home-assistant.io/t/paulmann-501-34-wall-switch/418414/7?u=ajd_ht
https://de.paulmann.com/p/wandschalter-smart-home-zigbee-3.0-on-off-dimm-weiss/50134

---

I paired

Paulmann 501.34 Wall Switch ([Wandschalter Smart Home Zigbee 3.0 On/Off/Dimm Weiß](https://de.paulmann.com/p/wandschalter-smart-home-zigbee-3.0-on-off-dimm-weiss/50134))

[image]

with

Home Assistant 2023.10.3
Supervisor 2023.10.0
Operating System 11.0
Frontend 20231005.0 - latest

and

ZHA

I cannot find any possibility to integrate the Paulmann 501.34 Wall Switch in any automation.

I only find 1 device

[image]

with 4 entities

https://user-images.githubusercontent.com/9786634/276741930-42356fd5-4b57-41b8-85bf-226f1bc7469d.png

### Steps to reproduce

1. Pair Paulmann 501.34 Wall Switch (sw_build_id (id: 0x4000) 2.7.6_r25)
2. Try to automate.

### Expected behavior

1. It should be possible to see two devices / switches
2. The following events (a1, a0, b1, b0) should be exposed and should be used in automations
[image]

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4644, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
[... 559 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

drueppela, 2023-10-26:
I tried to implement

https://github.com/zigpy/zha-device-handlers/tree/dev/zhaquirks/paulmann)](https://github.com/zigpy/zha-device-handlers/tree/dev/zhaquirks/paulmann

### **Step 1:**
I created the directory:

`/config/zha_custom_quirks`

[image]

### **Step 2:**
Added

```
zha:
database_path: /config/zigbee.db
enable_quirks: true
custom_quirks_path: /config/zha_custom_quirks/
```

to

`/config/configuration.yaml`

### **Step 3:**
Added

`__init__.py`

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/__init__.py

to the directory:

`/config/zha_custom_quirks`

[image]

### **Step 4:**
Added

`const.py`

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/const.py

to the directory:

`/config/zha_custom_quirks`

[image]

### **Step 5:**
Added directory

`paulmann`

https://github.com/zigpy/zha-device-handlers/tree/dev/zhaquirks/paulmann

to the directory:

`/config/zha_custom_quirks`

[image]

### **Step 6:**
Added

`__init__.py`

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/paulmann/__init__.py

to the directory:

`/config/zha_custom_quirks/paulmann`

[image]

### **Step 7:**
Added

`fourbtnremote.py`

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/paulmann/fourbtnremote.py

to the directory:

`/config/zha_custom_quirks/paulmann`

https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/paulmann/fourbtnremote.py

[image]

### **Step 8:**
Disconnected Paulmann Licht GmbH Wall Switch 501.34

### **Step 9:**
Factory reset Paulmann Licht GmbH Wall Switch 501.34

### **Step 10:**
Restart Home Assistant

### **Step 11:**
Pairing Paulmann Licht GmbH Wall Switch 501.34

### **Result:**

As before I can see events fired, but Paulmann 501.34 Wall Switch (sw_build_id (id: 0x4000) 2.7.6_r25) cannot be used in HASSIO automations.

[image]

---

Where can I see if the quirk is running?

[image]

```
Logger: zhaquirks
Source: components/zha/__init__.py:122
First occurred: 19:34:30 (1 occurrences)
Last logged: 19:34:30

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers
```

---

[image]

---

I cannot find a hint at

https://github.com/zigpy/zha-device-handlers/pull/2419
https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

TheJulianJES, 2023-10-28:
Can you listen to `zha_event` in the Home Assistant "developer tools" -> "events" section and see if you get any events when pressing the buttons?
Also, see if there's anything in the logbook when pressing buttons.

drueppela, 2023-11-01:
@TheJulianJES & @dmulcahey & @MattFromGer & @javicalle

[image]

-------------------
###  GROUP A - START
-------------------

### **GROUP A - A1 - ON [SHORT CLICK & RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP A - A0 - OFF [SHORT CLICK & RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP A - A1 - DIMMING + [LONG PRESS/HOLD]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP A - A1 - DIMMING + [RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP A - A1 - DIMMING - [LONG PRESS/HOLD]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP A - A1 - DIMMING - [RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

-------------------
###  GROUP B - START
-------------------

### **GROUP B - B1 - ON [SHORT CLICK & RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP B - B0 - OFF [SHORT CLICK & RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP B - B1 - DIMMING + [LONG PRESS/HOLD]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP B - B1 - DIMMING + [RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP B - B1 - DIMMING - [LONG PRESS/HOLD]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

### **GROUP B - B1 - DIMMING - [RELEASE]**
### **Listen to `zha_event`**
[image]

### **Logbook**
[image]

javicalle, 2023-11-01:
Just noticed that your device signature is:
```
  "manufacturer": "Paulmann Licht GmbH",
  "model": "501.34",
```

While the quirk is using the value:
https://github.com/zigpy/zha-device-handlers/blob/d1a1b599da44c471b52a616c320aab1d5491e163/zhaquirks/paulmann/__init__.py#L3

Have you tried to add your signature to quirk? it would be something like:
```python
class PaulmannRemote4Btn(CustomDevice):
    """Custom device representing Paulmann 4-button 501.34 remote control."""

    signature = {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=1
        # device_version=0
        # input_clusters=[0, 1, 3, 2821, 4096]
        # output_clusters=[3, 4, 5, 6, 8, 25, 768, 4096]>
        MODELS_INFO: [
            (PAULMANN, "501.34"),
            ("Paulmann Licht GmbH", "501.34"),
        ],
        ENDPOINTS: {
            1: {
                .../...

```

closed 2023-11-01 by TheJulianJES

drueppela, 2023-11-01:
@javicalle : 🥇 

All Device Triggers are available now. MANY THANKS.

---

I would suggest to change

[zha-device-handlers/zhaquirks/paulmann/__init__.py](https://github.com/zigpy/zha-device-handlers/blob/d1a1b599da44c471b52a616c320aab1d5491e163/zhaquirks/paulmann/__init__.py#L3)

Line 3
from
`"PAULMANN = "Paulmann LichtGmbH"`

to
`"PAULMANN = "Paulmann Licht GmbH"`

---

Is there a chance to make the Device Triggers available as Entities?

---

Is there a chance to change the Device Trigger Names

`Button 1` ==> `Button a1`

and

"First button"2 pressed to "Button a1" pressed?

 like the Paulmann description?

[image]

redfusion83, 2023-11-23:
@drueppela Thanks for this very well written guide. It's great that you put a lot of work into your posts, it really contributes to the community! Also, I had the same problem and now I can use the newer switches too!

I think it's still too much work to create the automations for this switch and I'd like to create a blueprint to make it easier. I've never done this before and need to learn, but that's the fun of Home Assistant!

itsmicash, 2024-09-17:
@redfusion83 @drueppela @MattFromGer @TheJulianJES 

Hi guys,
I have issues with button A0 and B0 (or button 2 and 4)
all button presses work except long press and press and realease on these 2 buttons. hope someone can support me.

Button 2 long press & release
[image]

Button 2 long press
[image]

Button 4 long press & release
[image]

Button 4 long press
[image]

I'm really struggling and don't know what to do.

PR #2695 (2023-11-01): Add support for different Paulmann manufacturer name
## Proposed change

Add support for slightly different Paulmann manufacturer name. Some switches have `Paulmann LichtGmbH`, newer models have `Paulmann Licht GmbH` as manufacturer name.

## Additional information

Fixes #2662 

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works