zigpy/zha-device-handlers #2663: [Device Support Request] Moes STAR RING Curtain Switch ZS-SR-EUC   TS0601 _TZE204_Srmahpwl No Control

GuyAzria, 2023-10-20:
### Problem description

I bought [this](https://moeshouse.com/products/zigbee-star-ring-curtain-switch?variant=46106364117307) Curtain Switch from Aliexpress. It seems to be detected and "successfully" configured by ZHA but it does not find the conrol only as router  

### Solution description

This device will be supported?

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_srmahpwl",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Asia/Jerusalem",
    "os_name": "Linux",
    "os_version": "6.1.56",
    "supervisor": "2023.10.0",
[... 354 lines cut ...]
  }
}]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

GuyAzria, 2023-10-23:
I want to try to write Quirk myself, I would be very grateful for help on how to get started or is there a quirk Tuya for Moes Star Ring that can be adjusted to help me started.  I'm a programmer but I don't know much about Zigbee program

Batzi1967, 2024-01-05:
Hello everyone! I use the same device and have the same problem. Is there already a solution to this problem?

dompagoj, 2024-01-05:
Im also still waiting on support for this device 🥲

Nico8854, 2024-01-31:
j'attend aussi l'interrupteur volet se prend pour un thermomètre sous ZHA Il n'est donc pas reconnu sous ca bonne forme et model (il faudrais une bonne config ZHA pour lui redonner vie a ses entitées vraies

rogerlancien, 2024-03-05:
toujours pas de solutions ?

Nico8854, 2024-03-05:
J’ai fini par le passez par zigbeetomqtt :/ rien à faire de mon côté désolé

rogerlancien, 2024-03-05:
c'est ce que je viens de faire, mais je n'arrive pas a le remettre en appairage, c'est comme sil il était toujours connecté a ZHA

nrcdt, 2024-04-04:
Is there a way i can be of help by providing logs or something to get this going forward?

rogerlancien, 2024-04-04:
> Is there a way i can be of help by providing logs or something to get this going forward?
> 
> ```
> 2024-04-04 09:41:03.719 INFO (MainThread) [zigpy.device] [0x2550] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)
> 2024-04-04 09:41:03.720 INFO (MainThread) [zigpy.device] [0x2550] Discovering endpoints
> 
> 
> 2024-04-04 09:49:31.344 INFO (MainThread) [zigpy.endpoint] [0xc3bc:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=81, device_version=1, input_clusters=[4, 5, 61184, 0], output_clusters=[25, 10])
> ```

I think the best option is to use Mqtt+, it's working out of the box and it's much faster than HTTP requests

erkin-o, 2024-08-27:
I had the same issue with ZHA today. Does Zigbee2MQTT works? Or requires a custom quirk or even exists?

dompagoj, 2024-08-28:
> I had the same issue with ZHA today. Does Zigbee2MQTT works? Or requires a custom quirk or even exists?
> 
> Edit: I found my answer. It merged to Z2M. I’m leaving the link here and will update after trying it myself. [Koenkk/zigbee-herdsman-converters#6985](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6985)

i made it work bit Z2M, can confirm that it does not work with ZHA, same for their dimmer light switch, the dimmer switch works out of the box with Z2M, but the curtain switch required some custom JS i took from somewhere else

closed 2025-08-16 by github-actions[bot]