zigpy/zha-device-handlers #2664: [Device Support Request] Tuya/Aviatto TS0601  _TZE200_rtrmfadk

dassiegfried, 2023-10-20:
### Problem description

https://de.aliexpress.com/item/1005005883444420.html  
https://web.archive.org/web/20231020151002/https://www.aliexpress.us/item/3256805697129668.html

No entities shown in HA 

### Solution description

have MVP Entities 

    Get room temperature
    Get/Set setpoint temperature
    Get/Set global on/off status
    Get heating status
    Get/Set mode (manual/schedule)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.10.1",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "bosch_shc": {
      "version": "0.4.63",
      "requirements": [
        "boschshcpy==0.2.72"
      ]
    },
    "dwains_dashboard": {
      "version": "3.4.3",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "bosch": {
      "version": "0.23.4",
      "requirements": [
        "bosch-thermostat-client==0.23.2"
      ]
    },
    "sonoff": {
      "version": "3.5.2",
      "requirements": [
        "pycryptodome>=3.6.6"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp",
      "universal_silabs_flasher"
    ],
    "requirements": [
      "bellows==0.36.5",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.104",
      "zigpy-deconz==0.21.1",
      "zigpy==0.57.2",
      "zigpy-xbee==0.18.3",
      "zigpy-zigate==0.11.0",
      "zigpy-znp==0.11.5",
      "universal-silabs-flasher==0.0.14",
      "pyserial-asyncio-fast==0.11"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*tubeszb*",
        "known_devices": [
          "TubesZB Coordinator"
        ]
      },
      {
        "vid": "1A86",
        "pid": "7523",
        "description": "*zigstar*",
        "known_devices": [
          "ZigStar Coordinators"
        ]
      },
      {
        "vid": "1CF1",
        "pid": "0030",
        "description": "*conbee*",
        "known_devices": [
          "Conbee II"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8A2A",
        "description": "*zigbee*",
        "known_devices": [
          "Nortek HUSBZB-1"
        ]
      },
      {
        "vid": "0403",
        "pid": "6015",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate+"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*zigate*",
        "known_devices": [
          "ZiGate"
        ]
      },
      {
        "vid": "10C4",
        "pid": "8B34",
        "description": "*bv 2010/10*",
        "known_devices": [
          "Bitron Video AV2010/10"
        ]
      }
    ],
    "zeroconf": [
      {
        "type": "_esphomelib._tcp.local.",
        "name": "tube*"
      },
      {
        "type": "_zigate-zigbee-gateway._tcp.local.",
        "name": "*zigate*"
      },
      {
        "type": "_zigstar_gw._tcp.local.",
        "name": "*zigstar*"
      },
      {
        "type": "_uzg-01._tcp.local.",
        "name": "uzg-01*"
      },
      {
        "type": "_slzb-06._tcp.local.",
        "name": "slzb-06*"
      }
    ],
    "is_built_in": true
  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 60159,
    "manufacturer": "_TZE200_rtrmfadk",
    "model": "TS0601",
    "name": "_TZE200_rtrmfadk TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": 255,
    "rssi": -29,
    "last_seen": "2023-10-20T16:02:44",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_rtrmfadk",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "Heating Office",
    "device_reg_id": "ae113f29e712d4a2299ec06f781b38c2",
    "area_id": "office",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 72
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE200_rtrmfadk"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

a previous version seems to have been added to zigbee2mqtt https://github.com/Koenkk/zigbee2mqtt/issues/18892
Which got its infos from this issue https://github.com/Koenkk/zigbee2mqtt/issues/9265 
if you need more info or help developing/testing please reach out anytime 
thanks for your awesome work

Tiggx, 2023-10-20:
in my case this worked: https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_rtitek.py

dassiegfried, 2023-10-20:
wow thanks for your fast awnser I'll check it out and report back.

Tiggx, 2023-10-20:
only found this quirk and got over ur request. I have ordered the same valve so we are in the same boat. Could load the quirk and got some infos from the valve. Dont really know if this perfect works but is from now the best result i got!

dassiegfried, 2023-10-20:
that totally worked thanks so much for providing me with that link. I spent hours googling and found nothing.

Tiggx, 2023-10-20:
glad it worked. ur welcome. Spent already hours. Have a nice evening.

AngeloDellaMorte, 2024-01-09:
Thanks a lot.
I was searching days for a solution an now found this and it's just working and so easy.

phulstaert, 2024-01-15:
I have the same _TZE200_rtrmfadk and am using the same "https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_rtitek.py".
I see all the attributes and values, but am running into other issues. I was wondering if you ran into the same issues.

If I change the temperature on the device, I see the temp changing in HA, but when I want to change it in HA, nothing happens.
The thermostat is not locked or anything. I have 3 of these thermostats, but none of them work as they should.
I tried if for over 11 days now and only got it to change the temperature twice... I don't know what I am doing wrong...

phulstaert, 2024-01-15:
[home-assistant_zha_2024-01-15T17-05-11.626Z.log](https://github.com/zigpy/zha-device-handlers/files/13941437/home-assistant_zha_2024-01-15T17-05-11.626Z.log)

I have turned on debugging, restarted HA and I could control the thermostat a few times, but then it stopped working. 
I have included the logging file... I really hope someone can help me.

lpzubero, 2024-11-20:
I have applied this quirk to the device TS0601_TZE204_rtrmfadk, just adding a new signature to the code. Everything works OK. Thanks HA team :-)

pr8x, 2024-12-23:
@lpzubero I got it working with your approach, but it seems that turning the thermostat off does not work correctly. It stop the heating, but the state is then stuck in "Idle". Do you experience the same?

lpzubero, 2024-12-26:
The main problem I have found with this quirk is that in some cases the quirk puts the thermostat in the "schedule" mode (for example, after changing the programed temperature in HA). I think working mode "schedule" is not interesting when the thermostat is conected to HA, because it is supposed that "schedule" is configured in HA and not in the thermostat memory. In this case, the right mode should be is mode "manual". 

I am not a good programmer, so what I have done in my HA is to change following lines:

        if attribute == "system_mode":
            if value == self.SystemMode.Off:
                return {RTI_MODE_ATTR: 2}
            else:
                # LPZ 4-12-2024
                #return {RTI_MODE_ATTR: 0}
                return {RTI_MODE_ATTR: 1}

............

        else:
            operation_preset = self.Preset.Manual
            prog_mode = self.ProgrammingOperationMode.Simple
            occupancy = self.Occupancy.Occupied
# LPZ 4-12-2024            
#            operation_preset = self.Preset.Schedule
#            prog_mode = self.ProgrammingOperationMode.Schedule_programming_mode
#            occupancy = self.Occupancy.Occupied

May be there are better solutions, I would apreciate if someone with a better programing knowledge improve this code

pr8x, 2025-01-07:
@lpzubero That sounds exactly like the issue I am having. The TRV sometimes turns off at 12pm or 10pm, might be due to schedule mode. 
Could you perhaps attach a vcersion of your modified ts0601_trv_rtitek.py file? The formtting of your comment got messed up. Thank you so much!