zigpy/zha-device-handlers #2665: [Device Support Request] Add for Tuya TS130F curtain module calibration time for up and down

saschachina, 2023-10-20:
### Problem description

Opening and closing of the shutter takes different time, therefore the position of the shutter does not work correctly.

### Solution description

The possibility to calibrate the shutter separately for open and close would be great.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

sanchosk, 2023-11-30:
To make this issue clear - in Z2M there is 2 more values for each curtain - one is the calibration on/off and the other one is reverse motor on/off.
The calibration allows to turn on calibration mode and this will measure the time to go from one position to another. This time is then used for calculation of the position.
See the screenshot attached to (a wrong) [request on ZHA core](https://github.com/home-assistant/core/issues/104780) git.
The problem currently is that in ZHA, I cannot use the device - it comes from the factory calibrated to 10 seconds and this is WAY too short for my curtains to go down. I can connect the module to Z2M, calibrate to 63 seconds and then connect it back to ZHA, but this is a temporary workaround only...

sanchosk, 2023-11-30:
Thanks a lot! This helped :) Now I don't need to install Z2M temporary ;)

closed 2024-06-04 by github-actions[bot]

kebibg, 2024-12-15:
Hi,
The issue is not resolved yet.
On my module I don't have attribute calibration time but
in Z2M have these additional values.
Any news about this problem?
module is:TS130F
by _TZ3000_niezp8zq

saschachina, 2024-12-15:
> Hi, The issue is not resolved yet. On my module I don't have attribute calibration time but in Z2M have these additional values. Any news about this problem? module is:TS130F by _TZ3000_niezp8zq

Did you saw this poste? https://github.com/zigpy/zha-device-handlers/issues/2665#issuecomment-1834401370

You can already set a calibration time in ZHA, but not separately for up or down. I finally switched to the Shelly 2PM Gen. 3 and it was the best decision.

kebibg, 2024-12-15:
Hi,
On my  lift down and lift up don't work.
Work only and just only set lift to percentage.
Calibration time for up and down for me is fine but 
I can't do in developer menu too.

Regards