zigpy/zha-device-handlers #2666: [Device Support Request] TS0601 _TZE200_7p8ugv8d

cnru, 2023-10-20:
### Problem description

The Device mentioned in the Topic is Not supported Form ZHA.

https://www.aldi-onlineshop.de/p/smarte-heizkoerperthermostate-2er-set-1025848/?nsctrid=v01MTQyMTExMjExMTExMTExMTEwMTQyMTAwMDAwMDAwMDAxNjk3ODMwNzY2Y3JlYWRkZTMyMTg5MzgyNTk&cid=AF;retailAds;RG;textlink;Generisch;;AEC;42;;CG2;POE1;;e

### Solution description

Add Support for the Device in ZHA

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_7p8ugv8d",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

[zha-72f5503216062f1787dc8df9b6d9fdb5-_TZE200_7p8ugv8d TS0601-de3531c23becaf5acc1d2df265183a48.json.txt](https://github.com/zigpy/zha-device-handlers/files/13058090/zha-72f5503216062f1787dc8df9b6d9fdb5-_TZE200_7p8ugv8d.TS0601-de3531c23becaf5acc1d2df265183a48.json.txt)

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

cnru, 2023-10-24:
Maybe thus helps Whatsapp was eine die Zigbee2MQTT: https://github.com/Koenkk/zigbee2mqtt/issues/18980#issuecomment-1752101824

gbrd, 2023-10-29:
signature is the same than  #2589 ?

closed 2024-05-03 by github-actions[bot]