zigpy/zha-device-handlers #2672: [Device Support Request] Aqara Roller Shade Driver E1 (ZNJLBL01LM) - Missing Light sensor & Inversion toggle

codyc1515, 2023-10-21:
### Problem description

Unable to see the Light sensor or Inversion toggle when using ZHA with Aqara Roller Shade Driver E1 (model: ZNJLBL01LM, Zigbee ID: lumi.curtain.acn002).

### Solution description

Should be able to see the Light sensor value and able to toggle the Inversion of the direction of the cover.

While I cannot be certain, I suspect that the Rolled Shade Driver may be similar "quirk"-wise to the Curtain Driver. So, some of the changes performed in #2629 may translate across (with any luck)!

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0009",
        "0x000d",
        "0x0013",
        "0x0102",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.curtain.acn002",
  "class": "zhaquirks.xiaomi.aqara.roller_curtain_e1.RollerE1AQ_3"
}
```

### Diagnostic information

[zha-f035768bd02b002ee9b22933a6a5788a-LUMI lumi.curtain.acn002-05d4acfd8766bd55ee5233a620e72482.json.txt](https://github.com/zigpy/zha-device-handlers/files/13059921/zha-f035768bd02b002ee9b22933a6a5788a-LUMI.lumi.curtain.acn002-05d4acfd8766bd55ee5233a620e72482.json.txt)

### Logs

There is not anything showing in the logs as this functionality does not currently exist.

### Custom quirk

I am (I think?) running the latest quirk from #2620.

TheJulianJES, 2023-11-08:
Are you sure there's a light sensor on your model?

Also, can you use the clusters UI (device page -> manage zigbee device  -> select `XiaomiAqaraRollerE1` cluster -> select `speed` attribute) to check your speed (and see if you can change it)?

codyc1515, 2023-11-08:
Sorry, the documentation was unclear - it looks like the M1S Hub would instead use its illumination sensor. I have removed that from the ticket accordingly.

codyc1515, 2023-11-08:
Yep, I can confirm that the speed setting works with values 0 (low), 1 (medium), 2 (high - mine was set to this).

codyc1515, 2023-11-12:
Charging state is also working (with 1 being charging).

[image]

closed 2024-06-28 by github-actions[bot]