zigpy/zha-device-handlers #2675: [Device Support Request] IDLock 150 Zigbee module

prilly-dev, 2023-10-21:
### Problem description

IDlock 150 is a door lock that has a optional zigbee module, this module with the lock needs a quirk.
the goodpart is that its a full documentation available for the implementation. 

### Solution description

please make a quirk for this device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[User-Manual-Zigbee-module_EN_v0.4.pdf](https://github.com/zigpy/zha-device-handlers/files/13061791/User-Manual-Zigbee-module_EN_v0.4.pdf)

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4919, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x000a",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0101"
      ],
      "output_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Datek",
  "model": "ID Lock 150",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

prilly-dev, 2023-11-30:
any chance we can get support for this lock? Spec sheet contains the information needed to make a device handler

jorlarsen, 2024-04-23:
Would love for this device to get support.

Full documentation available here https://idlock.no/wp-content/uploads/2023/07/User-Manual-Zigbee-150_EN.pdf