zigpy/zha-device-handlers #2676: [BUG] cover _TZ3210_dwytrmda lift percentage is not updated

tsensei781, 2023-10-22:
### Bug description

with the device identified as below:
  "manufacturer": "_TZ3210_dwytrmda",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130FTOGP"

It seems that the current_position_lift_percentage (id 0x0008) is not updated correctly leading to an unexpected behavior within HA.
In my situation this variable is always set to 100 whatever the position of the cover is.

### Steps to reproduce

1. When the cover is open, request in HA to close the cover
2. The covers closed, immediately after HA is showing the correct position and associated commands.
3. Some times after (or if we read the current_position_lift_percentage value), HA is showing the cover as open
4. Up command is not accessible, activating the down command forces a refresh and HA shows the cover as closed

### Expected behavior

In the step 3 above, we should expect to always read the same position within HA.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_dwytrmda",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130FTOGP"
}
```

</details>

### Diagnostic information

[... 440 lines cut ...]
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

ssorgatem, 2024-02-13:
Same issue here withHA 2024.2 and the following signature (different manufacturer):

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_ol1uhvza",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130FTOGP"
}
```

the-dreamer, 2024-02-13:
Same here :-)

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0202",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0102"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_dwytrmda",
  "model": "TS130F",
  "class": "zhaquirks.tuya.ts130f.TuyaTS130FTOGP"
}

Any workaround known?

2024.2.1

```

the-dreamer, 2024-02-13:
I need to correct me: first power on of Device No. 1 paring starts instantly and i did pair the device in HA. device No. 2 did not want to pair and i need to start paring. After Pairing the state, percentage etc. was working. 

I did a force pair with device No. 1 and now also this device is showing the state and is working as expected.

ssorgatem, 2024-02-13:
> I need to correct me: first power on of Device No. 1 paring starts instantly and i did pair the device in HA. device No. 2 did not want to pair and i need to start paring. After Pairing the state, [...]

what do you mean with "force pair"? Making it enter pair mode manually?

the-dreamer, 2024-02-14:
yes

the-dreamer, 2024-02-15:
Some update: 

after some minutes, HA is loosing the values: 

[image]

tsensei781, 2024-02-15:
The only workaround I found was to use several automation scripts to force the position attribute for a given range.
But that's not very straightforward and needs to be re-done for every cover ...

the-dreamer, 2024-02-15:
This is also something i am considering now. The value which can be read is 0. HA is getting the position from the device when changed and after some time HA seems to read out the device again and get the 0. So for my understanding the value needs to be write back to device or even don't update from device. which could be the better solution.

tsensei781, 2024-02-15:
you are right, there is the need to write the value back on the device through "Set zigbee cluster attribute"

the-dreamer, 2024-02-15:
is this possible through yaml change for that device type? any hint?

MichalLytek, 2024-04-05:
I have a similar issue. The status always go back to "open" after a few minutes:

[image]

So at 21:24 I did full close action, after a few minutes it goes back to open. So I can't open the cover in the morning because it's already open.

I can use the physical switches without any issues. The cover opens or close no matter the status.
It only affects HA.

My workaround is to call "close" for 0.5s before "open" in my automations.
However it's not so simple with the HA buttons in dashboard.

the-dreamer, 2024-04-12:
As mentioned above an automation is solving the issue. The value is write back to the device. This is necessary for all devices. this is working for a while now.

alorle, 2024-06-11:
Samehere in v2024.6.2 with the same device as @ssorgatem: `_TZ3210_ol1uhvza`

> Same issue here withHA 2024.2 and the following signature (different manufacturer):
>
> ```
> {
>   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, [...]

Something curious happens to me: I have 4 identical devices, bought on the same date and with the same firmware versio. Two of them behave well and the others do not

I don't understand it, honestly. I don't know where I have read that it could be related to the fact that some are connected directly to the Zigbee coordinator and others through a router, but I don't know how to check it.

My solution was to implement the following automation for each cover:

```yaml
alias: Persiana cuarto (corregir posición)
description: ""
trigger:
  - platform: state
    entity_id:
      - cover.persiana_cuarto
condition: []
action:
  - variables:
      position: "{{ state_attr('cover.persiana_cuarto', 'current_position') }}"
  - delay:
      seconds: 5
  - repeat:
      while:
        - condition: template
          value_template: >-
            {{ state_attr('cover.persiana_cuarto', 'current_position') }} != position
      sequence:
        - variables:
            position: "{{ state_attr('cover.persiana_cuarto', 'current_position') }}"
        - delay:
            seconds: 5
  - service: zha.set_zigbee_cluster_attribute
    data:
      # Get IEEE from the device info page in HA
      ieee: 00:0d:6f:00:05:7d:2d:34

      # TuyaCoveringCluster (Endpoint id: 1, Id: 0x0102, Type: in)
      endpoint_id: 1
      cluster_id: 258
      cluster_type: in

      # current_position_lift_percentage (id: 0x0008)
      attribute: 8

      # set value from variable
      value: "{{ position }}"

      # do not override manufacturer code
      manufacturer: "-1"
mode: restart
```

tsensei781, 2024-06-12:
Thanks alorle for providing your script. 
much more efficient than mine.

cmos486, 2024-08-03:
The same thing happens to me, I have 5 identical tuya modules both physically and in terms of firmware, manufacturer etc... but only one of them with the problem.

MichalLytek, 2024-08-06:
@alorle your solution works well for HA, when it's closed by HA then it has proper status and can be opened back.
However, when I use the buttons on my wall, the cover goes up but then HA refuses to close it down.

This is because it's still wrong reporting the status as "in move" so "closing" or "opening", thus blocking the buttons in this direction:

[image]

alorle, 2024-08-06:
> @alorle your solution works well for HA, when it's closed by HA then it has proper status and can be opened back.
> However, when I use the buttons on my wall, the cover goes up but then HA refuses to [...]

In my case it's working for HA and phisical buttons. Maybe you have a different hardware?

MichalLytek, 2024-08-06:
I mean the script properly updates the percentage, so no longer I have an issue like reported earlier, that a few minutes after 100% closed, it's reported as 0% opened.

The issue I have now is that when I initiate the move via button and then stop in the middle via button, it still reports as "closing" or "opening", like the HA does not get updated by the zigbee device about the status or something.

Maybe we can find similar zigbee attribute to check and overwrite? 🤔

MichalLytek, 2024-08-06:
Yes, for 0 and 100 it works. Please try pressing the button in the middle of the movement to stop and see then 😉

the-dreamer, 2024-08-06:
thats my automation:
```

alias: Writeback Terrassentür
description: ""
trigger:
  - platform: state
    entity_id:
      - cover.terrassentur_abdeckung
    attribute: current_position
condition: []
action:
  - service: zha.set_zigbee_cluster_attribute
    data:
      ieee: >-
        {{
        (device_attr('cover.terrassentur_abdeckung','identifiers')|list).0.1}}
      endpoint_id: 1
      cluster_id: 258
      attribute: 8
      cluster_type: in
      value: "{{state_attr('cover.terrassentur_abdeckung','current_position')|int(0)}}"
mode: queued
max: 10

```
And it is working for many months now. Sometimes the value is not getting to the max. For example 94 or 13. But all functionallity is working. 

And the value is in my opinion not calculated from HA. It seems to be from the cover device as event. But it is not reflected as status where the script is coming in and writing that value to the device. HA is reading it out after some time.

tsensei781, 2024-08-06:
Hi everyone

otherwise I found another alternative that avoids using scripts : moving to Zigbee2MQTT. 
it recognizes the device without any additional workaround

marlomgirardi, 2024-08-11:
I've built a blueprint to help with my issues given I have a few of those devices and it is a bit painful to manage each automation manually. Feel free to use it as well if helps you:
https://gist.github.com/marlomgirardi/4fca8e6097520727d6cf88056798a8de

My issues with it have been:
- Stuck at 99% or stuck at 1%
- HA state stuck at `opening` or `closing`
- list percentage not updated in the device, making home assistant revert back to the last stored lift position on device.

mkz212, 2024-12-09:
Same here. 

Tuya TS130F TZ3000_eg7awg6a (single roller shutter) not save or wrong read cluster current_position_lift_percentage - it is always 100. When set curtain e.g. to 50% it moves to 50%, but after a while it shows in log that is opening and report in HA wrong position 100% (but in fact it is in right position 50%). Quirk: zhaquirks.tuya.ts130f.TuyaTS130GP.

Exactly same device, but double roller shutter, TS130F _TZ3000_j1xl73iw (Quirk: zhaquirks.tuya.ts130f.TuyaTS130Double_GP_ESTC) works ok.

Using ZHA. This happen for all this devices, not one. Home Assistant core-2024.12.1. No firmware updates.

mkz212, 2024-12-10:
> Samehere in v2024.6.2 with the same device as @ssorgatem: `_TZ3210_ol1uhvza`
> 
> > Same issue here withHA 2024.2 and the following signature (different manufacturer):
> > ```
> > {
> >   "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
> >   "endpoints": {
> >     "1": {
> >       "profile_id": "0x0104",
> >       "device_type": "0x0202",
> >       "input_clusters": [
> >         "0x0000",
> >         "0x0004",
> >         "0x0005",
> >         "0x0102"
> >       ],
> >       "output_clusters": [
> >         "0x000a",
> >         "0x0019"
> >       ]
> >     },
> >     "242": {
> >       "profile_id": "0xa1e0",
> >       "device_type": "0x0061",
> >       "input_clusters": [],
> >       "output_clusters": [
> >         "0x0021"
> >       ]
> >     }
> >   },
> >   "manufacturer": "_TZ3210_ol1uhvza",
> >   "model": "TS130F",
> >   "class": "zhaquirks.tuya.ts130f.TuyaTS130FTOGP"
> > }
> > ```
> 
> Something curious happens to me: I have 4 identical devices, bought on the same date and with the same firmware version. Two of them behave well and the others do not.
> 
> I don't understand it, honestly. I don't know where I have read that it could be related to the fact that some are connected directly to the Zigbee coordinator and others through a router, but I don't know how to check it.
> 
> My solution was to implement the following automation for each malfunctioning cover:
> 
> ```yaml
> alias: Persiana cuarto (corregir posición)
> description: ""
> trigger:
>   - platform: state
>     entity_id:
>       - cover.persiana_cuarto
> condition: []
> action:
>   - variables:
>       position: "{{ state_attr('cover.persiana_cuarto', 'current_position') }}"
>   - delay:
>       seconds: 5
>   - repeat:
>       while:
>         - condition: template
>           value_template: >-
>             {{ state_attr('cover.persiana_cuarto', 'current_position') }} != position
>       sequence:
>         - variables:
>             position: "{{ state_attr('cover.persiana_cuarto', 'current_position') }}"
>         - delay:
>             seconds: 5
>   - service: zha.set_zigbee_cluster_attribute
>     data:
>       # Get IEEE from the device info page in HA
>       ieee: 00:0d:6f:00:05:7d:2d:34
> 
>       # TuyaCoveringCluster (Endpoint id: 1, Id: 0x0102, Type: in)
>       endpoint_id: 1
>       cluster_id: 258
>       cluster_type: in
> 
>       # current_position_lift_percentage (id: 0x0008)
>       attribute: 8
> 
>       # set value from variable
>       value: "{{ position }}"
> 
>       # do not override manufacturer code
>       manufacturer: "-1"
> mode: restart
> ```

Thank you. It is working. So this is proof that something is wrong with ZHA or Home Assistant since automation can save the cluster correctly.

However, my blinds after using your automation show the opening state all the time. I wonder why using repeat if the automation mode is a restart?

the-dreamer, 2025-08-07:
Tested now, bug ist still there. Workaround is an automation to do write back the percantage values to the device

the-dreamer, 2026-02-06:
see edit above.

mkz212, 2026-02-06:
#4725 - Device report wrong position and this cause automation problems.

abmantis, 2026-02-07:
Has anyone who is affected tried using the quirk in https://github.com/zigpy/zha-device-handlers/pull/4688 as a custom quirk?

alorle, 2026-02-07:
> Has anyone who is affected tried using the quirk in [#4688](https://github.com/zigpy/zha-device-handlers/pull/4688) as a custom quirk?

I'm happy to try, but I don't know how. If someone could tell me how, I can do it and provide feedback.

mkz212, 2026-02-07:
It seems to me that the wrong (inverted) value for some covers appears only after saving the cluster, and after some time the value is correct.

abmantis, 2026-02-10:
> I tried it too, but it doesn't work. Here's a screenshot of “zigbee info”
> <img alt="Image" width="271" height="469" src="https://private-user-images.githubusercontent.com/55741502/547877159-cc84ed9a-2dd5-4310-92aa-5a0584b159eb.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NzA3NTA0NTcsIm5iZiI6MTc3MDc1MDE1NywicGF0aCI6Ii81NTc0MTUwMi81NDc4NzcxNTktY2M4NGVkOWEtMmRkNS00MzEwLTkyYWEtNWEwNTg0YjE1OWViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAyMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMjEwVDE5MDIzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRhM2VlOTUxYzRmYmZkYWUzNmUzNTQ1NDlmMjY0NDYzMmNjNDVmZDkyYmViY2JhZjM2NzZmYzk4YmMyMjEzMTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JfciqYeNFaBKCuHvy93uDmzOjvsnVsf6GI_FiXzjv6Y">

Weird 🤔 

Can you elaborate a bit on:
- What steps have you done to test it?
- Manual action from the wall switch or from HA? 
- Did the cover position update while it was moving?

dadoef, 2026-02-11:
> > > Has anyone who is affected tried using the quirk in [#4688](https://github.com/zigpy/zha-device-handlers/pull/4688) as a custom quirk?
> > 
> > I'm happy to try, but I don't know how. If someone could tell me how, I can do it and provide feedback.
> 
> - download the file from https://github.com/abmantis/zha-device-handlers/blob/ts130f_position_fix/zhaquirks/tuya/ts130f.py
> - create a folder named `zha_quirks` on your HA configuration directory, and place the download file in it.
> - open your HA configuration.yaml file and add the following:
> 
>   ```yaml
>   zha:
>     custom_quirks_path: /config/zha_quirks/
>   ```
> - restart your HA instance.
> 
> After that, open/close the cover once (so the value is updated on HA side).

I tried the quirk as well. Still same behavior as before unfortunately. Also for me until the new version, the automation was working fine.
Guess the only thing left is buying better zigbee motor switches.

mkz212, 2026-02-11:
I think a few things have changed:

- saving the cluster triggers the automation trigger (I don't think it happened before). Now, in order for the automation not to fall into the loop, you need to create a helper - at the beginning of the automation, block the helper and, unlock it eg. 30 seconds after the automation.

- some covers reports wrong / inverted positions. The cluster is saved correctly. If you read the cluster, it also forces the reading of the position and it is also correct. So I don't know where and at what point it makes this wrong reading.

What has not changed is the fact that still selected covers do not save the position cluster - automation or custom quirk is needed.

pchenriques, 2026-02-12:
I have tried it again after repairing. It seems to be holding the lift percentage position, but I can't change motor_reversal or calibrate it as I used to before. Also, notice the device is not exactly the one mentioned in the chat subject: my is TZ3210_ol1uhvza instead of TZ3210_dwytrmda. Is it an issue for me?

[image]

mkz212, 2026-02-16:
> > In 2026.2.2 I have such automation. It's not perfect, but I think it works so far. One automation to fix all wrong working covers.
> 
> I can confirm that with the quirk of abmantis and combining it with this automation that so far after a few days the state of the curtains seem to be OK.

I don't have a custom quirk, just automation.

I think about why part of the covers has an inverted state. Maybe quirk was broken when I added part of the covers and then the quirk was fixed and I added new covers with the fixed quirk (if the quirk is loaded only when the device is paired?). Maybe it's enough to remove and pair again each cover that has an inverted state? Or just reconfigure?

simonvic, 2026-02-28:
> Inverted state is not all the time but appears randomly

I've debugged the device for days with a custom quirk of mine that only prints debugging logs; the percentage attribute value is always reported inverted, both when polling the cluster (`read_attributes_raw`) or when the device reported it (i.e. `AttributeReportedEvent`, when the cover is moving)

Were you using some other quirk that was fiddling with the inversion? (the current one does it)
Or perhaps you had an old automation running?
May I ask you what firmware your devices report? Mine say `0x00000042`
Also, what's the value of the "calibrated" (`0xf001`) and "motor reversed" (`0xf002`) attributes of your devices? Maybe the device behaves differently based on that.

> These covers have a problem in the firmware that does not automatically get the position cluster.

To be precise, from my observation, the device simply doesn't automatically update (and save) the `current_position_lift_percentage`, hence home assistant shows stale values when polling (`read_attributes`).
Try to manually write `69`, could you confirm me that home assistant will periodically show `69`?

> it's some kind of bug that appeared in Home Assistant 2026.2.0

what 2026.2.0 update "broke" for me (and many other users from what I've read) is that manually writing to the cluster, hence changing the attribue value of a device, triggers (rightfully) any automation relying on that attribute, causing a loop.

> Automation is enough to fix it.

IMO it is not... why should every user create an error-prone and hard to maintain automation when we could "fix" the issues at quirk-level for all devices for all users?
Also, many suggested automations that I've found in this thread and the ones I've came up with are far from perfect and easily breaks when you have multiple covers opening/closing simultaneusly. And the "hacks" to avoid the loops could easily be replaced with a more performant solution in the quirks

> You can always set calibration time cluster - seconds x 10, eg. 16 sec = 160.

Thanks, yeah that's what I was doing and what I suggested in my previous comment as an alternative to the automatic calibration :)
Anyway, I believe exposing the input entity like I do with my quirk could still be useful to some users