zigpy/zha-device-handlers #2677: [Device Support Request] Tuya TS0601 _TZE204_g2ki0ejr

oxcid3, 2023-10-23:
### Problem description

https://www.aliexpress.com/item/1005005997839493.html

TRV Thermostatic Radiator Valve

### Solution description

Datapoints via Tuya IoT

{"2":"Mode", "4":"Set temperature", "5":"Current temperature", "6":"Battery capacity", "7":"Child lock", "9":"Set temperature ceiling", "10":"The lower limit of temperature", "14":"Window check", "16":"Window temp", "17":"Window time", "18":"Backlight brightness", "19":"Factory data reset", "21":"Holiday temperature", "24":"Home temp", "25":"Leave temp", "28":"Week program", "29":"Week program Tuesday", "30":"Week program Wednesday", "31":"Week program Thursday", "32":"Week program Friday", "33":"Week program Saturday", "34":"Week program Sunday", "35":"Fault alarm", "36":"Frost protection", "37":"Rapid warming", "38":"Rapid heating countdown", "39":"Switch Scale", "47":"Temperature correction", "48":"Valve testing", "49":"State of the valve", "101":"111"}

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

elboletaire, 2023-11-01:
I also purchased this same TRV, and I'm really interested in this quirk.

oxcid3, 2023-11-03:
Added screenshots from tuya app.

Jastreb07, 2023-11-04:
any update? I'm interested too in this quirk.

wormvortex, 2023-11-09:
Got the same device. Adding it via ZHA in home assistant does not expose any of the value nor does it let me control it

Steefph, 2023-11-10:
Here is my first attempt to create an actual working converter :)

``` const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_g2ki0ejr',
        },
    ],
    model: 'BAB-1413 Pro',
    vendor: 'TuYa',
    description: 'Thermostat Radiator Valve',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.child_lock(), e.open_window_temperature().withValueMin(5).withValueMax(30),
        e.comfort_temperature().withValueMin(5).withValueMax(30), e.eco_temperature().withValueMin(5).withValueMax(30), e.holiday_temperature().withValueMin(5).withValueMax(30),
        e.climate().withPreset(['auto', 'manual', 'holiday', 'comfort']).withLocalTemperatureCalibration(-5, 5, 0.1, ea.STATE_SET)
            .withLocalTemperature(ea.STATE).withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
            .withSystemMode(['off', 'heat'], ea.STATE_SET, 'Only for Homeassistant'),
        tuya.exposes.frostProtection('When Anti-Freezing function is activated, the temperature in the house is kept '+
                'at 8 °C, the device display "AF".press the pair button to cancel.'),
        e.numeric('boost_timeset_countdown', ea.STATE_SET).withUnit('s').withDescription('Setting '+
                'minimum 0 - maximum 465 seconds boost time. The boost (â¨) function is activated. The remaining '+
                'time for the function will be counted down in seconds ( 465 to 0 ).').withValueMin(0).withValueMax(465),
        e.text('holiday_start_stop', ea.STATE_SET).withDescription('The holiday mode will automatically start ' +
            'at the set time starting point and run the holiday temperature. Can be defined in the following format: ' +
            '`startYear/startMonth/startDay startHours:startMinutes | endYear/endMonth/endDay endHours:endMinutes`. ' +
            'For example: `2022/10/01 16:30 | 2022/10/21 18:10`. After the end of holiday mode, it switches to "auto" ' +
            'mode and uses schedule.'),
        e.enum('working_day', ea.STATE_SET, ['mon_sun', 'mon_fri+sat+sun', 'separate']).withDescription('`mon_sun` ' +
            '- schedule for Monday used for each day (define it only for Monday). `mon_fri+sat+sun` - schedule for ' +
            'workdays used from Monday (define it only for Monday), Saturday and Sunday are defined separately. `separate` ' +
            '- schedule for each day is defined separately.'),
        e.composite('schedule', 'schedule', ea.SET).withFeature(e.enum('week_day', ea.SET, ['monday', 'tuesday',
            'wednesday', 'thursday', 'friday', 'saturday', 'sunday'])).withFeature(e.text('schedule', ea.SET))
            .withDescription('Schedule will work with "auto" preset. In this mode, the device executes ' +
            'a preset week programming temperature time and temperature. Before using these properties, check `working_day` ' +
            'property. Each day can contain up to 10 segments. At least 1 segment should be defined. Different count of segments ' +
            'can be defined for each day, e.g., 3 segments for Monday, 5 segments for Thursday, etc. It should be defined in the ' +
            'following format: `hours:minutes/temperature`. Minutes can be only tens, i.e., 00, 10, 20, 30, 40, 50. Segments should ' +
            'be divided by space symbol. Each day should end with the last segment of 24:00. Examples: `04:00/20 08:30/22 10:10/18 ' +
            '18:40/24 22:50/19.5`; `06:00/21.5 17:20/26 24:00/18`. The temperature will be set from the beginning/start of one ' +
            'period and until the next period, e.g., `04:00/20 24:00/22` means that from 00:00 to 04:00 temperature will be 20 ' +
            'degrees and from 04:00 to 00:00 temperature will be 22 degrees.'),
        ...tuya.exposes.scheduleAllDays(ea.STATE, 'HH:MM/C'),
        //e.binary('online', ea.STATE_SET, 'ON', 'OFF').withDescription('The current data request from the device.'),
        e.binary('valve', ea.STATE, 'CLOSED', 'OPEN'),
        e.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
        tuya.exposes.errorStatus(),
    ],
    
    meta: {
        tuyaDatapoints: [
            [2, 'preset', tuya.valueConverterBasic.lookup({'comfort': tuya.enum(3),'auto': tuya.enum(0), 'manual': tuya.enum(2), 'holiday': tuya.enum(1)})],
            [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
[... 8 lines cut ...]
            //[18, 'Backlight brightness', tuya.valueConverter.raw],
            [19, 'factory_reset',  tuya.valueConverter.setLimit],
[... cut ...]
            [36, 'frost_protection', tuya.valueConverter.onOff],
            [37, 'boost_heating', tuya.valueConverter.onOff],
            [38, 'boost_time', tuya.valueConverter.countdown],
            [39, 'Switch Scale', tuya.valueConverter.raw],
            [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
            [48, 'valve_testing', tuya.valueConverter.raw],
            [49, 'valve', tuya.valueConverterBasic.lookup({'OPEN': 1, 'CLOSE': 0})],
        ],
    },
};

module.exports = definition; ```

oxcid3, 2023-11-10:
Amazing, thanks for this. - I'll run some testing on my side.
Just to be clear, should this work as a custom quirk, or do I need something else to get this to work. (I've not come across the concept of a converter before)

Steefph, 2023-11-10:
I looked for the "_TZE204_g2ki0ejr" thinking I would end up at zigbee2mqtt git. But apparently this is ZHA (never used it). The converter is for Z2M

enricocirignaco, 2023-11-18:
I have the same problem too. Any changes someone want to write a quirk for ZHA?
Would be really appreciated!

frederick9372, 2023-11-26:
How do you integrate the converter?

mleimbach-beeclever, 2023-12-02:
@frederick9372 unforgenatly the converter was a miss-post to the wrong issue/project.
(hopefully it might be helpful in a quirk development ?)
as @Steefph wrote. the contribution was planned togo into the zigbee2mqtt project but got the wrong issue/project.

For zha we would need a custom quirk or a native integration, 
but currently there does not seem to be a one for that product in the wild.

I also a couple of those valves here an would appriecieatea support (a working quirk) or a starting point for altering an existing quirk to be able to get access to the endpoints.

enricocirignaco, 2023-12-04:
Anyone interested in helping writing a quirk for this valve?
I'm a programmer but I never wrote quirks for HA. I have a valve so I could also test the code myself

mleimbach-beeclever, 2023-12-10:
Okay since nobody had a quirk running, i just tried out using the default trv quirk from tuya.
It originally did not sepcify the model we talk about here, but i just added it anyway.

I might need additional testing here, but for now i can give positive feedback using the official tuya trv quirk and just adding the devices id in the models info array.
This seems to give me the basic functionality that i need.

What you might want to try to also get it running:
- create a custom quirks folder as described [here](https://community.home-assistant.io/t/how-to-setup-local-zha-quirks/341226)
- get the official tuya [ts601 _tze204 trv quirk](https://github.com/Teka101/zha-device-handlers/blob/support_tze204_pcdmj88b/zhaquirks/tuya/ts0601_trv_tze204.py) and save the file in your custom quirks folder.

- modify the "models_info" to add the devices model name we have here

`
        MODELS_INFO: [
            ("_TZE204_pcdmj88b", "TS0601"),
            ("_TZE204_g2ki0ejr", "TS0601"),
        ]
,
       `
- restart HA
- deleted and reconnected the device

works for me 👍

FrankM70, 2023-12-10:
> Okay since nobody had a quirk running, i just tried out using the default trv quirk from tuya. It originally did not sepcify the model we talk about here, but i just added it anyway.
>
> I might need [...]

Thank you, it worked for me too. I am so impressed.
I think I learned a lot to solve other problems.

Ppatou, 2023-12-20:
Hello
I think I have discovered a small bug in the converter file.
I am using z2m as an add on in HA. So I think I have not easy access to the converter files.
In the climate integration 3 attributes concern the detected ambiance temp.
```
- local_temperature: 22
- local_temperature_calibration: 0.5
- current_temperature: 22
```
My understanding is that current_temp = local_temp + local_temp_calibration.
But current and local temp are identical.
I do not see the local_temp number in the expose file of the device
[image]

Can one of you correct the converter file and update the device data's
Thanks in advance
Patou

Teka101, 2024-01-13:
Support added in PR: https://github.com/zigpy/zha-device-handlers/pull/2873

agnus777, 2024-02-05:
> Thank you, it worked for me too. I am so impressed. I think I learned a lot to solve other problems.

I tried this and it works for basic function.
what I noticed is that switch 1 is the related command of "force heat" while switch 2 is the "window detection".

Someone know how to replace functions?

altopais, 2024-02-06:
After the HA update i get 

"Failed to call service switch/turn_on. Failed to send request: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>" 

for every function.

As i am a new to HA and ZHA, looking for help to resolve the issue.
Thanks.

closed 2024-08-11 by github-actions[bot]