zigpy/zha-device-handlers #2680: [Device Support Request] Existing device _TZE200_yjjdcqsq send unknown values ZTH01

jfdzar, 2023-10-25:
### Problem description

I have a TUYA temperature humidity sensor Model: ZTH01
I would like to connect it to Home Assistant through ZHA and Sonoff Donlge P

The interview and the recognition works as expected but in the sensor I only see unknown values

Here the device signature

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_yjjdcqsq",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_sensor.TuyaTempHumiditySensorVar03"
}`

### Solution description

It would be great to correct the quirk or have the correct quirk for this sensor

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-10-25:
Is the coordinator on latest firmware also is is connected to on good router and not to the coordinator (3 or more routes is needed for getting the mesh network working O ) ?

jfdzar, 2023-10-25:
@MattWestb thanks for the tip, I have 4 routers at the moment so the mesh should work correctly. After some times and requesting temperature and humidity looks like it is working now.
I will keep monitoring but for the moment I will close the issue

closed 2023-10-25 by jfdzar