zigpy/zha-device-handlers #2682: [BUG] MOES Zigbee thermostat (TZE200_aoclfnxz) Not syncing time

jfparis, 2023-10-26:
### Bug description

I have installed some of the MOES Zigbee thermostats. The device has an internal clock and displays the time of day on the screen. The thermostat seems to request a time sync over Zigbee but those requests fail

### Steps to reproduce

1. Pair the thermostat with ZHA
2. The thermostat internal clock does not sync 

### Expected behavior

The thermostat internal clock should sync to the time provided by ZHA 

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_aoclfnxz",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_electric_heating.MoesBHT"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.10.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "tfl_livearrivals": {
      "version": "0.0.1",
      "requirements": [
[... 678 lines cut ...]
```python
2023-10-24 20:44:41.958 DEBUG (MainThread) [zigpy.zcl] [0x84D1:1:0xef00] Received ZCL frame: b'\x19t$\x00\x00'
2023-10-24 20:44:41.958 DEBUG (MainThread) [zigpy.zcl] [0x84D1:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufact
urer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False), tsn=116, command_id=36, *direction=<Direction.Client
_to_Server: 1>)
2023-10-24 20:44:41.960 DEBUG (MainThread) [zigpy.zcl] [0x84D1:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_time_request(param=[0, 0])
2023-10-24 20:44:41.961 DEBUG (MainThread) [zigpy.zcl] [0x84D1:1:0xef00] Received command 0x24 (TSN 116): set_time_request(param=[0, 0])
2023-10-24 20:44:41.962 DEBUG (MainThread) [zigpy.zcl] [0x84D1:1:0xef00] No explicit handler for cluster command 0x24: set_time_request(param=[0, 0])

```

</details>

### Additional information

Ultimately the fix seems to be like documented elsewhere: Adding one line ``set_time_offset = 1970`` in https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_electric_heating.py

```
class MoesBHTManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some electric heating thermostats."""

    set_time_offset = 1970
```

Happy to push it however unclear if this can break other's devices

jacekk015, 2023-10-31:
In case you would like to get working quirk with TempCalibration and etc. you can use:
ts0601_thermostat_avatto.py
from:
https://github.com/jacekk015/zha_quirks

goodlucknow, 2024-04-06:
@jfparis were you able to implement a fix for this?
I copied the quirk file and edited with the fix mentioned, and enabled custom quirks in ZHA.
Initially nothing but as I type I think the thermostat has just successfully updated the time...
Would love to get the temperature calibration working also, will need to look through what @jacekk015 linked above.

closed 2024-10-11 by github-actions[bot]