zigpy/zha-device-handlers #2686: [Device Support Request] Sonoff SNZB-01P (new style button)

kc2zgu, 2023-10-29:
### Problem description

Sonoff has a new button called SNZB-01P that uses a different housing design. It looks generally software compatible with the old SNZB-01 so the existing quirk code might work with it.

### Solution description

Support for "toggle", "on", and "off" events from this device.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "SNZB-01P",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.10.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.10.0-20-amd64",
    "supervisor": "2023.10.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
[... 366 lines cut ...]
  params: {}
origin: LOCAL
time_fired: "2023-10-29T15:18:21.179318+00:00"
context:
  id: 01HDY07EQV876Y5V613KK6CH84
  parent_id: null
  user_id: null

event_type: zha_event
data:
  device_ieee: 84:ba:20:ff:fe:dc:2e:9a
  unique_id: 84:ba:20:ff:fe:dc:2e:9a:1:0x0006
  device_id: 96ad04d2cc5fa845b01eb53e221010f0
  endpoint_id: 1
  cluster_id: 6
  command: "off"
  args: []
  params: {}
origin: LOCAL
time_fired: "2023-10-29T15:18:34.012668+00:00"
context:
  id: 01HDY07V8WVTYR1YV5G690K7EQ
  parent_id: null
  user_id: null
```

blakadder, 2023-11-01:
It already is supported by ZHA. I've had it working for a few monts now

[image]

kc2zgu, 2023-11-01:
In HA 2023.10.4 I'm not able to use device triggers in automations as there is apparently no quirk detected for it. The only way to do anything is with a manual event trigger. Can you expand the Zigbee info tab on your system?

blakadder, 2023-11-01:
I am using zha_events for automation triggers

kc2zgu, 2023-11-09:
Would it be possible to add a quirk for supporting the different triggers more easily? To do anything I currently have to use a manual zha_event trigger.

PR #2833 (2023-12-20): Add device triggers for Sonoff SNZB-01P smart button
## Proposed change
This adds support for Sonoff smart button SNZB-01P. 
Basically provides 'normal' action triggers: short click, double click and a long press

Fixes #2686

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2023-12-20 by TheJulianJES