zigpy/zha-device-handlers #2687: [Device Support Request] aqara Smart Radiator Thermostat E1, external sensor

MarkL1969, 2023-10-29:
### Problem description

The TRV has the option to use an external sensor for temperature. (well, the temperature from an external sensor can be past over the the TRV).
It's working for a while using Z2M and there was a quirk offering that for ZHA too:
[jymdman](https://github.com/zigpy/zha-device-handlers/issues/1757#issuecomment-1379267908)

Nwk: 0x3633
Device Type: EndDevice
Eigenart: zhaquirks.xiaomi.aqara.thermostat_agl001.AGL001 

### Solution description

It would be perfect, if it was possible to offer switching to external sensor and passing over temperature to the TRV where integrated into the "official release".

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Sorry if some information or "protocol" is missing - I'm a newbie to this...

hraschan, 2023-10-29:
Implementation of this would be highly appreciated!

sroettering, 2023-11-28:
What exactly does it mean for the E1 to use an external sensor? Does it query the sensor itself or is it just passed a temperature value that it can use?
I'm also very interested in this feature and could try to add it to the current quirk. But I need to understand it first

MarkL1969, 2023-11-28:
The E1 accepts temperature values coming from external sensors, so the algorithms of the E1 will use those passed on values instead of the values coming from its internal sensor.
This means that the temperature values have to be constantly written to the E1 - otherwise it will switch back to using its internal sensor.
There ist a solution on how to solve this using a blueprint (as soon as the E1 quirk enables the use of external sensor values):
https://community.home-assistant.io/t/z2m-aqara-trv-e1-link-external-temperature-sensor/609689/3

mb1248, 2023-11-29:
In z2m it works the following way:

The sensor attrib is set to external with send two buffers to the attrib 0xfff2. The buffers holds a command. In this case activate 'internal' or 'external' temperature value. The buffer is a straight forward calculation https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/xiaomi.ts#L424
I would assume, that the sensor attrib via 0x027e will then return 1 when set to 'external' mode.

A temperature value from an external source could then send to sensor_temp, again via 0xfff2. The buffer containing again a custom command and the temperature value  is here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/xiaomi.ts#L483

When we have a quirk that calculate the command buffer. It should be easy to use zha toolkit to send the temperature value from any source to the thermostat.

hraschan, 2023-12-04:
@sroettering Can you get a quirk running with the current information?

sroettering, 2023-12-05:
I will try maybe around christmas, thanks for the information

Lopicl, 2024-03-31:
Hi, has there been any update on this? I was going to open an issue, but found this one which is exactly what I am looking for. I am available for testing in case.

sroettering, 2024-04-02:
Sadly I did not have time for this yet :( If someone wants to give it a try, please go ahead. You are probably faster than me right now.

Lopicl, 2024-04-02:
> Sadly I did not have time for this yet :( If someone wants to give it a try, please go ahead. You are probably faster than me right now.

Don't worry! :)
I found out that in PR  #2802 there is a working quirk and an automation to set an external sensor, I have already tried it out and it is working.

closed 2024-10-06 by github-actions[bot]

Esp32-zapper, 2026-02-01:
This feature lacking is the main reason I think I will move away from the Aqara thermostats.
In so many cases the curtain hangs down and covers the thermostat, so in winter, if you set the thermostat to 20⁰C the thermostat sensor will rather quickly register 20⁰C but the room temperature sensor reads 17⁰C.
Doesn't help to turn up the setting to 25⁰C as the sensor will register 25⁰C very quickly and turn the heating off.
The technicians at Aqara has never seen or heard about a thing called "wifes". If the fashion calls for long curtains, there is no use trying to explain that it will mess with the thermostat. You and the thermostat are wrong and that's the end of it.