zigpy/zha-device-handlers #2689: [Device Support Request] Aqara Motion Sensor (lumi.sensor_motion.aq2) configure Sensitivity

codyc1515, 2023-10-29:
### Problem description

Some of my PIR sensors receive a lot of sunlight during the day which can cause them to report false motion detections.

### Solution description

Ability to adjust the Sensitivity should reduce the frequency that the false detection occurs.

### Screenshots/Video

[image]

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4151, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=0, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0107",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0002",
        "0x0003",
        "0x0400",
        "0x0406",
        "0x0500",
        "0xffff"
      ],
      "output_clusters": [
        "0x0000",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.sensor_motion.aq2",
  "class": "zhaquirks.xiaomi.aqara.motion_aq2.MotionAQ2"
}

### Diagnostic information

[zha-f035768bd02b002ee9b22933a6a5788a-LUMI lumi.sensor_motion.aq2-29e8d4dfd51496e84413acd2c0e8285b.json.txt](https://github.com/zigpy/zha-device-handlers/files/13199313/zha-f035768bd02b002ee9b22933a6a5788a-LUMI.lumi.sensor_motion.aq2-29e8d4dfd51496e84413acd2c0e8285b.json.txt)

### Logs

Not applicable

TheJulianJES, 2023-10-29:
Those older Aqara motion sensors do not support changing the motion sensitivity.

codyc1515, 2023-10-30:
Hmm okay that's frustrating. I will check one in the Aqara app and confirm but you may be right.

codyc1515, 2023-10-30:
Yep only P1 sensors support this :(

closed 2023-10-30 by codyc1515

21-Jailbraik-21, 2023-12-08:
YAMA V10