zigpy/zha-device-handlers #2690: [Device Support Request] AQARA Switch E1 double - lumi.switch.b2nc01

albalaing, 2023-10-30:
### Problem description

This switch uses the OppleCluster for its configuration which is not implemented

### Solution description

I have added the following attributes to OppleCluster
0x0200 Bdecoupled mode
0x0201 Power outage memory
0x00F0 Flip indicator light

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
"""Xiaomi aqara single key wall switch devices."""

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster
from zigpy.zcl.clusters.general import (
    AnalogInput,
    Basic,
    BinaryOutput,
    DeviceTemperature,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
    GreenPowerProxy,
)

from zhaquirks import Bus, EventableCluster
from zhaquirks.const import (
    ARGS,
    ATTRIBUTE_ID,
    ATTRIBUTE_NAME,
    BUTTON_1,
    BUTTON_2,
    BUTTON_3,
    CLUSTER_ID,
    COMMAND,
    COMMAND_ATTRIBUTE_UPDATED,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    SKIP_CONFIGURATION,
    VALUE,
)
from zhaquirks.xiaomi import (
    LUMI,
    AnalogInputCluster,
    BasicCluster,
    OnOffCluster,
    XiaomiCustomDevice,
    XiaomiPowerConfiguration,
    XiaomiAqaraE1Cluster,
)

ATTRIBUTE_PRESENT_VALUE = "present_value"

class QBKG41LM(XiaomiCustomDevice):
    """Aqara double key switch device."""

    class OppleCluster(XiaomiAqaraE1Cluster):
        """Add Opple cluster for power outage memory attribute."""

        attributes = {
            0x200: ("decoupled_mode", t.uint8_t, True),
            0x201: ("power_outage_memory", t.uint8_t, True),
            0x0F0: ("flip_indicator_light", t.uint8_t, True),
        }

    class WallSwitchMultistateInputCluster(EventableCluster, MultistateInput):
        """WallSwitchMultistateInputCluster: fire events corresponding to press type."""

    def __init__(self, *args, **kwargs):
        """Init."""
        self.power_bus = Bus()
        super().__init__(*args, **kwargs)

    signature = {
        MODELS_INFO: [(LUMI, "lumi.switch.b2nc01")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    OppleCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id,Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    OppleCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            41: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [MultistateInput.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
            42: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [MultistateInput.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
            51: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [MultistateInput.cluster_id],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xa1e0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffCluster,
                    OppleCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOffCluster,
                    OppleCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            41: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [WallSwitchMultistateInputCluster],
                OUTPUT_CLUSTERS: [],
            },
            42: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [WallSwitchMultistateInputCluster],
                OUTPUT_CLUSTERS: [],
            },
            51: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [WallSwitchMultistateInputCluster],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {
            ENDPOINT_ID: 41,
            CLUSTER_ID: 18,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 85, ATTRIBUTE_NAME: ATTRIBUTE_PRESENT_VALUE, VALUE: 1},
        },
        (DOUBLE_PRESS, BUTTON_1): {
            ENDPOINT_ID: 41,
            CLUSTER_ID: 18,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 85, ATTRIBUTE_NAME: ATTRIBUTE_PRESENT_VALUE, VALUE: 2},
        },
        (SHORT_PRESS, BUTTON_2): {
            ENDPOINT_ID: 42,
            CLUSTER_ID: 18,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 85, ATTRIBUTE_NAME: ATTRIBUTE_PRESENT_VALUE, VALUE: 1},
        },
        (DOUBLE_PRESS, BUTTON_2): {
            ENDPOINT_ID: 42,
            CLUSTER_ID: 18,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 85, ATTRIBUTE_NAME: ATTRIBUTE_PRESENT_VALUE, VALUE: 2},
        },
        (SHORT_PRESS, BUTTON_3): {
            ENDPOINT_ID: 51,
            CLUSTER_ID: 18,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 85, ATTRIBUTE_NAME: ATTRIBUTE_PRESENT_VALUE, VALUE: 1},
        },
        (DOUBLE_PRESS, BUTTON_3): {
            ENDPOINT_ID: 51,
            CLUSTER_ID: 18,
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            ARGS: {ATTRIBUTE_ID: 85, ATTRIBUTE_NAME: ATTRIBUTE_PRESENT_VALUE, VALUE: 2},
        },
    }
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Muxa1L, 2024-01-11:
@albalaing why this quirk may not be working? Or it's not supposed to? 
Got following log messages trying to add device again
[0xC1C7:3:0x0006]: 'async_configure' stage failed: 'uint8_t' object is not subscriptable
[0x5B12:3:0x0006]: 'async_configure' stage failed: 'uint8_t' object is not subscriptable
[0x7A37:3:0x0006]: 'async_configure' stage failed: 'uint8_t' object is not subscriptable

But these may be from another source (dunno)

albalaing, 2024-01-11:
I have a switch that works without problem. This switch has the endpoints [1,2,41,42,51,242] and version 22

Since I had an unconfigured switch, I added it and the quirks did not work because it was identified with the enpoints [1,2,3,41,42,51,242]. I bought this switch later and it has version 25. Although I did not use the Quirks, it worked without problems. But it didn't give me that error.

Tell me the signature of your switch and the version to see if we can add it.

Note: The version is seen in:
Basic Clusters (Endpoint id: 1, Id: 0x0000, Type: in)
app_version attributes (id: 0x0001)

scotie, 2024-02-02:
@albalaing  For me your quirk doesn't seem to work, but I didn't have to look into it yet.
I do have an issue open already with the signature if it helps. [#2514](https://github.com/zigpy/zha-device-handlers/issues/2514)

I'll try again a bit later and report back

FerrumLogic, 2024-07-15:
Current quirk doesn't applied to my lumi.switch.b2nc01

closed 2025-01-18 by github-actions[bot]

benjamin-bergia, 2025-01-26:
I tried this quirk on my switches and it doesn't work either. 
The quirk signature contains the endpoints: 1, 2, 41, 42, 51, 242 but my device signature contains the following endpoints:
```yaml
"endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0002",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0012",
            "0xfcc0"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "2": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0012",
            "0xfcc0"
          ],
          "output_clusters": []
        },
        "3": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006"
          ],
          "output_clusters": []
        },
        "41": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0012"
          ],
          "output_clusters": []
        },
        "42": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0012"
          ],
          "output_clusters": []
        },
        "51": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0012"
          ],
          "output_clusters": []
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
```
No idea if it's related but at the moment it doesn't seem to work.

benjamin-bergia, 2025-01-26:
So I copied the H1 quirk and played with it a bit. The following seems to work:
```python
"""Aqara E1 double rocker switch quirks."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Alarms,
    Basic,
    DeviceTemperature,
    GreenPowerProxy,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.xiaomi import LUMI
from zhaquirks.xiaomi.aqara.opple_switch import (
    XiaomiOpple2ButtonSwitchBase,
)

class AqaraE1DoubleRockerSwitchWithNeutral(XiaomiOpple2ButtonSwitchBase):
    """Aqara H1 Double Rocker Switch (with neutral)."""

    signature = {
        MODELS_INFO: [(LUMI, "lumi.switch.b2nc01")],
        ENDPOINTS: {
            # input_clusters=[0, 2, 3, 4, 5, 6, 9], output_clusters=[10, 25]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Alarms.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            # input_clusters=[0, 3, 4, 5, 6], output_clusters=[]
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # input_clusters=[0, 3, 4, 5, 6], output_clusters=[]
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
```

benjamin-bergia, 2025-01-27:
Power metering doesn't seem to work :/

MeatHunter95, 2025-02-05:
Hi there,

First of all: Thank you all for spending your time to get our things to work! :)

I am also trying to get my Aqara E1s to work.
I have the EU version with single and double rocker.

The quirk posted by @benjamin-bergia is not loaded as can be seen by the diagnostics log of my Double Rocker E1:
`"data": {
    "ieee": "**REDACTED**",
    "nwk": 59801,
    "manufacturer": "LUMI",
    "model": "lumi.switch.b2nc01",
    "name": "LUMI lumi.switch.b2nc01",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4447,
    "power_source": "Mains",
    "lqi": 200,
    "rssi": -50,
    "last_seen": "2025-02-05T15:42:21",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": {
        "logical_type": 1,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 142,
        "manufacturer_code": 4447,
        "maximum_buffer_size": 127,
        "maximum_incoming_transfer_size": 100,
        "server_mask": 11264,
        "maximum_outgoing_transfer_size": 100,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0002",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0009"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "2": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006"
          ],
          "output_clusters": []
        },
        "3": {
          "profile_id": "0x0104",
          "device_type": "0x0100",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006"
          ],
          "output_clusters": []
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "LUMI",
      "model": "lumi.switch.b2nc01"
    },
`

I named the quirk "lumi.switch.b2nc01.py".
How can I make sure the quirk is loaded?
I will then be able to test it out and give feedback :)

benjamin-bergia, 2025-02-06:
Hey,

I have a new version of the quirk since the signature seems to change between version:
```python
"""Aqara E1 double rocker switch quirks."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import (
    Alarms,
    Basic,
    DeviceTemperature,
    GreenPowerProxy,
    Groups,
    Identify,
    MultistateInput,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.xiaomi import LUMI
from zhaquirks.xiaomi.aqara.opple_switch import (
    OppleSwitchCluster,
    XiaomiOpple2ButtonSwitchBase,
)

class AqaraE1DoubleRockerSwitchWithNeutral(XiaomiOpple2ButtonSwitchBase):
    """Aqara E1 Double Rocker Switch (with neutral)."""

    signature = {
        MODELS_INFO: [(LUMI, "lumi.switch.b2nc01")],
        ENDPOINTS: {
            # input_clusters=[0, 2, 3, 4, 5, 6, 18, 64704], output_clusters=[10, 25]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    OppleSwitchCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            # input_clusters=[0, 3, 4, 5, 6, 18, 64704], output_clusters=[]
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
[... 105 lines cut ...]
                OUTPUT_CLUSTERS: [],
            },
            # input_clusters=[], output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
```

You can see if the quirk is being loaded on the device page in the "Device info" card under "Zigbee info" at the bottom of the list, if the device use your quirk you should see the name of your quirk in there. If you don't see it then your quirk is not used.
Also you need to do the following:
- Remove delete all the E1 switches from HA.
- Update the config yaml file to enable quirks and add the directory with your quirk file
- Save the file
- Restart HA
- Add your E1 switches

Every time you change the quirk, you have to remove the switches from HA, restart HA and add them back. 
You need to restart HA not quick reload it.

MeatHunter95, 2025-02-08:
Hi Ben,

Thank you so much for your help!

I created the file lumi.switch.b2nc01.py in the same folder where I already
have a Quirk for some Beok Thermostats.
So this should be correct, right?

I then paired the switch and I have this:
[image: image.png]

This is the same as I have seen without quirk - I can't use multi click,
decoupled mode, ...

What did I do wrong?

Best regards