zigpy/zha-device-handlers #2691: Loratap SC400ZB-EU-V5S (Ts130f) percentage position not working[Device Support Request]

jopeyyyy, 2023-10-30:
### Problem description

Percentage Positioning is not working if state 0 and put for example 10%, it will go to 100%. And same in the other direction.
Calibration was made and calibration time is correctly set. Percentage state is also correctly get back in HA if actuate via physical button or via HA.
Quirk is loaded: zhaquirks.tuya.ts130f.TuyaTS130ESTC

### Solution description

Would like to get the percentage positioning working. 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

closed 2023-11-09 by jopeyyyy

ciltocruz, 2023-12-08:
How did you fix it?

TobiasB91, 2023-12-24:
I would be interested in a fix as well.

jopeyyyy, 2023-12-24:
Hello,

check here.
https://github.com/zigpy/zha-device-handlers/issues/2724

 I didn't solve the problem but I found a way to use thoses switched.
It would still be finde to get thoses switches working correctly with the percentage position. Or at least to integrate the actual percentage value in the cover rf time based so everything works even when usingthe hardware switch...