zigpy/zha-device-handlers #2694: [Device Support Request] TS0601 _TZE204_upagmta9 Tuya outdoor temperature and humidity sensor

raven4is, 2023-11-01:
### Problem description

No values recognized 

Device info
TS0601
by _TZE204_upagmta9
Zigbee info
IEEE: a4:c1:38:74:06:b0:2c:89
Nwk: 0x2aff
Device Type: EndDevice
LQI: 192
RSSI: -52
Last Seen: 2023-11-01T13:11:52
Power Source: Battery or Unknown

### Solution description

How can i get or make a custom quirk for this device?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_upagmta9",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.5",
    "dev": false,
    "hassio": true,
[... 320 lines cut ...]
```python
Logger: zigpy.zcl
Source: runner.py:186
First occurred: 13:05:38 (5 occurrences)
Last logged: 13:13:56

[0x2AFF:1:0xef00] Unknown cluster command 17 b'\x00\x03@'
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I tried some quirks but with no help.
Can someone help me?

benoitm974, 2023-11-02:
It seems not the be working either on zigbeeMQTT, there might be some aspect of the protocol missing at pairing, trying to get element from snifing protocol here : https://github.com/Koenkk/zigbee2mqtt/issues/19137

raven4is, 2023-11-03:
@benoitm974 Seems lit a batch that beside the problems with zigbee information transmission, they have huge problems with battery time, or maybe because it tries to send information to gateway to often, the battery just goes dead in 12h.

benoitm974, 2023-11-03:
No in fact the device uses tuya specific protocol, the device works well when use with their gateway and Tuya app and don't have battery issue. The problem exists in both zigbee2mqtt and ZHA where the current code doesn't follow the tuya protocol after pairing the device stay "locked" in a waiting for answer state where it doesn't update the screen info nor send any data and consume all the battery in 12 to 24h...
In the zigbee2mqtt forum I attached a capture of the tuya sequence, where we can see:
 - the tuya send a "reset" command at pairing
 - also send a command 3 and 10 to the device
 - and lastly reply wuccess to device command 11, which seems to be the expected command for the sensors to go to sleep and start updateing screen and sending zigbee values

darknessblade, 2023-11-10:
I have a similar issue with a device that has the same specification/ID

_TZE204_upagmta9 TS0601
Aliexpress Item page [ordered the item trough the 3=5.99$ sale]
https://aliexpress.com/item/1005006030373016.html?

For me it does not show ANY entities at all

raven4is, 2023-11-10:
Hello! For this sensors as a temporary sollution i added a wifi zigbee gateway and used tuya integration for them.

They do eat batteries if not communicating good with ha integration.

I wait for news when they cand be integrated locally. :(

darknessblade, 2023-11-10:
> Hello! For this sensors as a temporary sollution i added a wifi zigbee gateway and used tuya integration for them.
> 
> They do eat batteries if not communicating good with ha integration.
> 
> I wait for news when they cand be integrated locally. :(

Soo they are worse than "for example" 
https://aliexpress.com/item/1005005352101380.html?
If all you would need is the temperature+ humidity

I bought one as a test, but might return/see if I can get a refund for it, if there is no probable fix in the next few days.

Since its a choice store getting a refund is quite easy.

raven4is, 2023-11-11:
I wanted with display, i suppose will be a while until this model will be added to the fully supported devices,
I saw some people had luck with zigbee2mqtt. So they are not worse, they are not yet supported but they have the capability to be as good.

thubalek, 2023-11-15:
It looks like devs from zigbee2mqtt made it working (https://github.com/Koenkk/zigbee2mqtt/issues/19137#issuecomment-1811953885)

benoitm974, 2023-11-15:
Indeed after sniffing zigbee traffic and debuging @koenkk found the race condition preventi the 0xef00 / 0x03 command sent just after the sensors join the network which was keeping the device in a loop mode... If any one with experience on zha code can point in the right directtion for sending command when sensor joint I'm open to try/look into it.

dhruvinsh, 2023-11-16:
I manage to get some data at least,

> If any one with experience on zha code can point in the right directtion for sending command when sensor joint I'm open to try/look into it.

correct file path would be, `zhaquirks/tuya/ts0601_sensor.py`, more specifically line [248](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py#L248) code. Not entirely sure, everything is fairly new for me.

[image]

benoitm974, 2023-11-18:
The question is not which definition to use, since either V03 or V04 would work, the issue is how to have ZHA to send a custom tuya command on joining to get the sensor to start sending data on regular interval and to sleep in between.

phormix, 2023-11-22:
Sounds like I should just pull the battery from mine until this is resolved.

Can anyone recommend a temperature sensor that does with properly?

benoitm974, 2023-11-22:
I use these working ok but no screen, and you need to provide the 2 AAA batteries.
https://fr.aliexpress.com/item/1005006022905558.html?spm=a2g0o.order_list.order_list_main.97.21ef5e5bIqwGzC&gatewayAdapt=glo2fra

tokkeherfst, 2023-11-29:
I also have a couple of these that are quite useless in ZHA... is anyone working on a fix please?

slimninja, 2023-12-15:
Adding my support to merge a similar fix from:
https://github.com/Koenkk/zigbee2mqtt/issues/19137#issuecomment-1811953885

JordanAco1, 2024-01-08:
Anyone please? Would be nice to have them in ZHA because they are cheap, have a display and natively have zigbee. The people who buy these are mostly the ones who have bricked  a LYWSD03MMC and are looking for something with native zigbee.

eddietheeagle7, 2024-01-19:
Same for me, can't get it working.

marcoaureliovaz, 2024-01-21:
Hello! The correct model for this device seems to be ZTH05 
[image]

I'm having same issues. Identified as a different device and no entities.

[image]

RoboKaren, 2024-01-21:
> Hello! The correct model for this device seems to be ZTH05 ![317738231-751844634](https://private-user-images.githubusercontent.com/155663270/298409359-0496305e-a0f4-4929-a4e5-56b59fc7d64f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU4NzIxNzksIm5iZiI6MTcwNTg3MTg3OSwicGF0aCI6Ii8xNTU2NjMyNzAvMjk4NDA5MzU5LTA0OTYzMDVlLWEwZjQtNDkyOS1hNGU1LTU2YjU5ZmM3ZDY0Zi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyMVQyMTE3NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNmM0MjYxMTkzNDA1YjgxY2JjMWU1OTViMmExZmFmMTE1ODk4YzAyMDEzM2ExOTAzNWExNTkwNjg5MTcyNWQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5msBjxe1CiDJMtoA31zF_amaMx8oBGJ74Xl99Rmd9LI)
> 
> I'm having same issues. Identified as a different device and no entities.
> 
> ![Screenshot_20240121-212037](https://private-user-images.githubusercontent.com/155663270/298409436-2064ea37-b525-4155-bbd8-09b98342e694.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU4NzIxNzksIm5iZiI6MTcwNTg3MTg3OSwicGF0aCI6Ii8xNTU2NjMyNzAvMjk4NDA5NDM2LTIwNjRlYTM3LWI1MjUtNDE1NS1iYmQ4LTA5Yjk4MzQyZTY5NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyMVQyMTE3NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kYzE4MjNhZTZkOWI5YTRjZTkyNzU4MDI3NDVmYzJjZGRjNzkzZjFkYWE3MmViNzIzNTFhMWM2ZTNkNTZhNTg3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.wjrewxqlHJncaFV3IfukRzvqeTg8Zu4_DzMGyqZ9bFA)

I suppose the question is whether there is a way to force the real identity onto devices that list a fake identity.

sweetpants, 2024-01-25:
@marcoaureliovaz, i'm seeinig the exact same on my ZHA installation

marcoaureliovaz, 2024-01-25:
[image]

I tried some time with ZHA but it didn't work and the "pairing" seemed not to conclude... (After pairing, the logs were always bringing new messages and at the device, low battery icon active)

ZigBee2MQTT
Then I flashed my Sonoff Dongle-E to MultiPAN with that Webflasher, installed Mosquitto, Z2M Edge and now it's correctly recognized and working great.
Frequent updates during increase/decrease of temperature/humidity. Battery info is reliable.

TheJulianJES, 2024-01-30:
With the next HA version 2024.2.0 beta (releasing 2024/01/31), you can try to send the "Tuya query data" command through the clusters menu UI if the manufacturer cluster is added using a quirk.

Copying my answer from Discord:
https://github.com/zigpy/zha-device-handlers/pull/2940 and https://github.com/zigpy/zha-device-handlers/pull/2888 added support for the "query data" command (0x03).
So, with the next HA version, it should be relatively simple to make a quirk for that device. You can likely use [zigpy/quirk-generator](https://github.com/zigpy/quirk-generator) to generate a stub quirk, replace `CustomDevice` with the Tuya `EnchantedDevice` implementation.
That'll send Tuya specific commands during pairing. But you'll also need to set tuya_spell_data_query = True to send the new 0x03 / query data command (like seen in #2940).

That way, no manual command sending would be involved.

benoitm974, 2024-01-31:
> With the next HA version 2024.2.0 beta (releasing 2024/01/31), you can try to send the "Tuya query data" command through the clusters menu UI if the manufacturer cluster is added using a quirk.
> 
> Copying my answer from Discord: #2940 and #2888 added support for the "query data" command (0x03). So, with the next HA version, it should be relatively simple to make a quirk for that device. You can likely use [zigpy/quirk-generator](https://github.com/zigpy/quirk-generator) to generate a stub quirk, replace `CustomDevice` with the Tuya `EnchantedDevice` implementation. That'll send Tuya specific commands during pairing. But you'll also need to set `tuya_spell_data_query = True` to send the new 0x03 / query data command (like seen in #2940) during pairing/reconfiguration (and then pair/reconfigure).
> 
> That way, no manual command sending would be involved.

Thanks @TheJulianJES so much!
So it means from the existing quirks https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py

We could on create a new 05 variation and then

A Add the import "EnchantedDevice"
-> `from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster, EnchantedDevice`

B Add the variation this way`

`
class TuyaTempHumiditySensorVar05(EnchantedDevice):
    """Tuya temp and humidity sensor (variation 05)."""
    tuya_spell_data_query = True

    signature = {
        # "profile_id": 260,
        # "device_type": "0x0051",
        # "in_clusters": ["0x0000","0x0004","0x0005","0xef00"],
        # "out_clusters": ["0x000a","0x0019"]
        MODELS_INFO: [
            ("_TZE204_upagmta9", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityBatteryStatesManufCluster,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }`

benoitm974, 2024-01-31:
> > > With the next HA version 2024.2.0 beta (releasing 2024/01/31), you can try to send the "Tuya query data" command through the clusters menu UI if the manufacturer cluster is added using a quirk.
> > > Copying my answer from Discord: #2940 and #2888 added support for the "query data" command (0x03). So, with the next HA version, it should be relatively simple to make a quirk for that device. You can likely use [zigpy/quirk-generator](https://github.com/zigpy/quirk-generator) to generate a stub quirk, replace `CustomDevice` with the Tuya `EnchantedDevice` implementation. That'll send Tuya specific commands during pairing. But you'll also need to set `tuya_spell_data_query = True` to send the new 0x03 / query data command (like seen in #2940) during pairing/reconfiguration (and then pair/reconfigure).
> > > That way, no manual command sending would be involved.
> > 
> > 
> > Thanks @TheJulianJES so much! So it means from the existing quirks https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_sensor.py
> > We could on create a new 05 variation and then
> > A Add the import "EnchantedDevice" -> `from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster, EnchantedDevice`
> > B Add the variation this way`
> 
> Sorry.... But I have the same issue and after looking at this thread I got the quirk indicated and made the suggested changes. It picked up that quirk and I get the identities for humidity and temperature which I didn't have before. But they have no values. Am I missing some step or will this just work in the upcoming versions?
> 
> Thank you.

Only changing the quirk will not be enough until this is relaeased or you ensure to get the beta version install and test the new quirk...

domusonline, 2024-01-31:
> Only changing the quirk will not be enough until this is relaeased or you ensure to get the beta version install and test the new quirk...

Thank you very much for the promptly reply. I'll wait until this is changed... I don't usually install beta versions of HA because I don't have a "test" system. But I'll keep an eye on this.
Regards.

benoitm974, 2024-02-01:
> So new beta is out with EnchantedDevice type.
> 
> Install brand new beta home assistant in docker connected to clean sonOff zigbee 3.0v2 Added zha quirks, they get picked up by homeassistant, tried 2 quirks variation ... But seems the spell 'get data 0x0003 command doesn't work
> 
> var 05: `class TuyaTempHumiditySensorVar05(EnchantedDevice): """Tuya temp and humidity sensor (variation 05).""" tuya_spell_data_query = True
> 

Also tried to add a inheritance from TuyaEnchentableCluster as mention in the documentation (if I understood it correctly), and added it to the Device definition ... But no luck so far

class TemperatureHumidityBatteryStatesManufClusterSpell(TuyaEnchantableCluster, TemperatureHumidityBatteryStatesManufCluster):
    """Tuya which needs spell query_data """

atus42, 2024-02-04:
I didn't read this thread lately, so I made my dirty hack in the weekend to finaly get my devices working.
As you all noticed, this didn't solve the problem completelly. :(
without the 0x03 command sent to the 0xef00 endpoint the commands arriving from it was unknown to the ZCL handler:

`Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 1, 27, 14, 20, 45, 394485, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x47D8), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tE\x11\x00\x06@'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=99, rssi=None)`
`[0x47D8:1:0xef00] Received ZCL frame: b'\tE\x11\x00\x06@'`
`[0x47D8:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=69, command_id=17, *direction=<Direction.Client_to_Server: 1>)`
`[0x47D8:1:0xef00] Unknown cluster command 17 b'\x00\x06@'`
`[0x47D8:1:0xef00] Received command 0x11 (TSN 69): b'\x00\x06@'`
`[0x47D8:1:0xef00] No explicit handler for cluster command 0x11: b'\x00\x06@'`

However after sending the magic packet I got these answers:

`Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 2, 4, 21, 5, 18, 154777, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6853), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\t\xd0\x11\x00\x06@'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=55, rssi=None)`
`[0x6853:1:0xef00] Received ZCL frame: b'\t\xd0\x11\x00\x06@'`
`[0x6853:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=208, command_id=17, *direction=<Direction.Client_to_Server: 1>)`
`[0x6853:1:0xef00] Decoded ZCL frame: TemperatureHumidityBatteryStatesManufCluster:mcu_version_response(version=MCUVersion(status=0, tsn=6, version_raw=64, *version='1.0.0'))`
`[0x6853:1:0xef00] Received command 0x11 (TSN 208): mcu_version_response(version=MCUVersion(status=0, tsn=6, version_raw=64, *version='1.0.0'))`
`[0x6853:1:0xef00] MCU version: 1.0.0`
`[0x6853:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=208, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)`
`[0x6853:1:0xef00] Sending reply: Default_Response(command_id=17, status=<Status.SUCCESS: 0>)`

Unfortunatelly no measurement data arrives from the node besides this 'TemperatureHumidityBatteryStatesManufCluster' self report.

Anyone has any idea what shoud we do next?

rodvlopes, 2024-02-10:
I am having the same issue here. Just bought two of these ZTH05 and they pair but can't send any temp ou humidity to ZHA. The display of the device never updates.

michelgre, 2024-02-14:
Hello
I am confused by all this thread but I understand that a change in HA would make it possible to configure a workaround (quirk) for this protocol issue. I installed the latest version now, 2024.2.1, and I wonder if:
1. This version includes the necessary change for that ?
2. How/where to install this workaround, and which one (var 06 ?) ?
It would be helpful for many people who got these sensors and who will come here but never installed ZHA quirks to have a small guide to explain that... 
Thank you !

Talixme, 2024-02-21:
Same here, i have 5 of them. They work great on zigbee2mqtt but nothing on ZHA

FranzTen, 2024-03-07:
have the same problem with them, i hope they will work some day with HA out of the box

jacekk015, 2024-03-13:
Quirk code is already in ZHA. Only device ID was missing:
_TZE204_upagmta9
https://github.com/zigpy/zha-device-handlers/pull/3036

jacekk015, 2024-03-13:
@slimninja go to

https://github.com/zigpy/zha-device-handlers/blob/6a1638b0e55811dff1f5061758b9836a7cc22b93/zhaquirks/tuya/ts0601_sensor.py

Use 3-dot menu to download file
Enable custom quirks settings in HA

https://github.com/jacekk015/zha_quirks?tab=readme-ov-file#usage

Test it and let us all know.
You will use my patched quirk. Later, after update, you just disable Custom Quirks or delete that file.

domusonline, 2024-03-13:
> @slimninja go to
> 
> https://github.com/zigpy/zha-device-handlers/blob/6a1638b0e55811dff1f5061758b9836a7cc22b93/zhaquirks/tuya/ts0601_sensor.py
> 
> Use 3-dot menu to download RAW file Enable custom quirks settings in HA
> 
> https://github.com/jacekk015/zha_quirks?tab=readme-ov-file#usage
> 
> Test it and let us all know. You will use my patched quirk. Later, after update, you just disable Custom Quirks or delete that file.

I tried it without success. It recognizes the device and it uses the quirk but I have no entities...
I have downloaded the diagnostic data. Will attach it here.

[zha-08a1cb49e83dff14adee605b1b9800f1-_TZE204_upagmta9 TS0601-94a3e23b5027bb5c9b04c2c7d6a8f882.json](https://github.com/zigpy/zha-device-handlers/files/14591376/zha-08a1cb49e83dff14adee605b1b9800f1-_TZE204_upagmta9.TS0601-94a3e23b5027bb5c9b04c2c7d6a8f882.json)

domusonline, 2024-03-13:
> Diagnostic data doesn't contain any usable info. Debug level LOG is needed.
> 
>     1. Enable debug log level
> 
> 
> ```
> logger:
>   default: info
>   logs:
>     homeassistant.components.zha: debug
>     zigpy: debug
>     zhaquirks: debug
> ```
> 
>     2. Remove one device from HA
> 
>     3. Restart Home Assistant!!!
> 
>     4. Pair device like a brand new - only then there's a full initialization process
> 
>     5. Wait for another, minimum, 5 minutes!!!
> 
>     6. Download Home Assistant log [Setting ->System->Log] - attach it here as a file

Here it is. I hope you can understand something from it.
Feel free to suggest any test you may think is useful.
Regards.
[home-assistant_2024-03-13T22-00-01.396Z.log](https://github.com/zigpy/zha-device-handlers/files/14594323/home-assistant_2024-03-13T22-00-01.396Z.log)

domusonline, 2024-03-13:
I see a few messages in the log mentioning "FOREIGN KEY constraint failed"... But I am not able to understand what that means in this context.... seems like it interacts well with the device but can't save something into a database?...

domusonline, 2024-03-13:
Another update.... if I restart HA I do get the entities, although with no values.... weird

Kerber0ss, 2024-03-17:
Same issue, with quirks - sensors adds to HA but not displaying data from it.

jacekk015, 2024-03-28:
Bought one from the Ali link above. 9 days delivery.
I will try to debug it deeper after delivery

go-pro-v, 2024-04-26:
Hello, I have the same problem.
I try again every time there's an update from HA, but it still doesn't work. And I can confirm that the sensor battery drains quickly.

CaptainMalu, 2024-05-16:
Any news on this topic. I got one of this sensors today.
I'm some shocked that it isn't working out of the box like every other zigbee device I got.

I use ZHA with my home assistant but never heard of quirks. Do I have to use them or is a solution on its way?
And if I have to use them: could someone help me find a step by step guide?

Dexter0007, 2024-07-08:
Hi, I also bought these sensors, but it turned out that after integration there is no temperature and humidity entity.
Only after a few hours of searching, I found this thread.
Is any working solution?

srohr87, 2024-09-14:
Same Problem. 3 ZTH05 without any entity.
Are there any news about a solution?

SarcasticShrimp, 2024-09-17:
Looking for a solution as well.

Peketr, 2024-10-05:
I've spent the last 2 days comparing how z2m and zha work.
Now I've got an (as far as I can tell) working custom quirk for it, but I had to override a lot of stuff from parent classes.
Anyways, if anyone wants to try here it is: [https://github.com/Peketr/zha-device-handlers/blob/Peketr-patch-zht05/zhaquirks/tuya/ts0601_sensor.py](https://github.com/Peketr/zha-device-handlers/blob/Peketr-patch-zht05/zhaquirks/tuya/ts0601_sensor.py)

P.S.: I don't know if this one will run the battery down or not, and there is no support for changing °C/°F.

LennardPlay, 2024-10-06:
@Peketr 
Thank you for your hard work. Sadly in my case it still doesn't show any entities. Maybe My device is on another firmware or maybe I just did something wrong.

[image]

[image]

[image]

It doesn't seem like its using your quirk even after resetting and repairing the device.

Peketr, 2024-10-06:
@LennardPlay I changed the file a bit so the upagmta9 is only in the new Var05. Maybe that helps you. Just make sure to restart HA after adding the file, and check if the logs show something like "exception while importing quirk"

LennardPlay, 2024-10-06:
I found a few lines in my zha log containing the word "quirk". In the time window I did delete the device and paired it again. 

`2024-10-06 11:40:46.882 DEBUG (MainThread) [zha] Emitting event device_removed with data DeviceRemovedEvent(device_info=ExtendedDeviceInfo(ieee=a4:c1:38:d4:d5:09:7a:fc, nwk=0x1A1D, manufacturer='_TZE204_upagmta9', model='TS0601', name='_TZE204_upagmta9 TS0601', quirk_applied=False, quirk_class='zigpy.device.Device', quirk_id=None, manufacturer_code=4417, power_source='Battery or Unknown', lqi=176, rssi=-56, last_seen='2024-10-06T11:39:25', available=True, device_type='EndDevice', signature={'node_descriptor': NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False), 'endpoints': {1: {'profile_id': '0x0104', 'device_type': '0x0051', 'input_clusters': ['0x0000', '0x0004', '0x0005', '0xef00'], 'output_clusters': ['0x000a', '0x0019']}}, 'manufacturer': '_TZE204_upagmta9', 'model': 'TS0601'}, active_coordinator=False, entities={'a4:c1:38:d4:d5:09:7a:fc-1-0-rssi': SensorEntityInfo(fallback_name=None, unique_id='a4:c1:38:d4:d5:09:7a:fc-1-0-rssi', platform=<Platform.SENSOR: 'sensor'>, class_name='RSSISensor', translation_key='rssi', device_class=<SensorDeviceClass.SIGNAL_STRENGTH: 'signal_strength'>, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, entity_registry_enabled_default=False, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='BasicClusterHandler', generic_id='cluster_handler_0x0000', endpoint_id=1, cluster=ClusterInfo(id=0, name='Basic', type='server', commands=[ZCLCommandDef(id=0x00, name='reset_fact_default', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.reset_fact_default'>, is_manufacturer_specific=None)]), id='1:0x0000', unique_id='a4:c1:38:d4:d5:09:7a:fc:1:0x0000', status='CREATED', value_attribute=None)], device_ieee=a4:c1:38:d4:d5:09:7a:fc, endpoint_id=1, available=True, group_id=None, attribute=None, decimals=1, divisor=1, multiplier=1, unit='dBm'), 'a4:c1:38:d4:d5:09:7a:fc-1-0-lqi': SensorEntityInfo(fallback_name=None, unique_id='a4:c1:38:d4:d5:09:7a:fc-1-0-lqi', platform=<Platform.SENSOR: 'sensor'>, class_name='LQISensor', translation_key='lqi', device_class=None, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, entity_category=<EntityCategory.DIAGNOSTIC: 'diagnostic'>, entity_registry_enabled_default=False, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='BasicClusterHandler', generic_id='cluster_handler_0x0000', endpoint_id=1, cluster=ClusterInfo(id=0, name='Basic', type='server', commands=[ZCLCommandDef(id=0x00, name='reset_fact_default', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.reset_fact_default'>, is_manufacturer_specific=None)]), id='1:0x0000', unique_id='a4:c1:38:d4:d5:09:7a:fc:1:0x0000', status='CREATED', value_attribute=None)], device_ieee=a4:c1:38:d4:d5:09:7a:fc, endpoint_id=1, available=True, group_id=None, attribute=None, decimals=1, divisor=1, multiplier=1, unit=None), 'a4:c1:38:d4:d5:09:7a:fc-1-25-firmware_update': UpdateEntityInfo(fallback_name=None, unique_id='a4:c1:38:d4:d5:09:7a:fc-1-25-firmware_update', platform=<Platform.UPDATE: 'update'>, class_name='FirmwareUpdateEntity', translation_key=None, device_class=<UpdateDeviceClass.FIRMWARE: 'firmware'>, state_class=None, entity_category=<EntityCategory.CONFIG: 'config'>, entity_registry_enabled_default=True, enabled=True, cluster_handlers=[ClusterHandlerInfo(class_name='OtaClientClusterHandler', generic_id='cluster_handler_0x0019', endpoint_id=1, cluster=ClusterInfo(id=25, name='Ota', type='client', commands=[ZCLCommandDef(id=0x03, name='image_block', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.clusters.general.ImageBlockCommand'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x04, name='image_page', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.clusters.general.ImagePageCommand'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x01, name='query_next_image', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.clusters.general.QueryNextImageCommand'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x08, name='query_specific_file', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.query_specific_file'>, is_manufacturer_specific=None), ZCLCommandDef(id=0x06, name='upgrade_end', direction=<Direction.Client_to_Server: 0>, schema=<class 'zigpy.zcl.foundation.upgrade_end'>, is_manufacturer_specific=None)]), id='1:0x0019', unique_id='a4:c1:38:d4:d5:09:7a:fc:1:0x0019', status='CREATED', value_attribute=None)], device_ieee=a4:c1:38:d4:d5:09:7a:fc, endpoint_id=1, available=True, group_id=None, supported_features=<UpdateEntityFeature.INSTALL|SPECIFIC_VERSION|PROGRESS: 7>)}, neighbors=[], routes=[], endpoint_na [... cut ...] DevicePairingStatus.INITIALIZED: 4>), new_join=False, event_type='zha_gateway_message', event='device_fully_initialized')
2024-10-06 11:41:29.613 DEBUG (MainThread) [bellows.ezsp.protocol] Received command sendUnicast: {'status': <EmberStatus.SUCCESS: 0>, 'sequence': 150}

2024-10-06 11:42:23.114 DEBUG (MainThread) [zhaquirks.tuya] [0xf1d5:1:0xef00] Received value [0, 0, 0, 184] for attribute 0x0218 (command 0x0001)
`

I know this is a bit off topic, but can you help me with applying this quirk?

Kerber0ss, 2024-10-06:
In my variant - is unk_model
unk_manufacturer

:(

[image]

OZub, 2024-10-07:
> check

[image]
Great job! You are the hero!

stevh22, 2024-10-07:
For me it's also working fine, well done!

[image]

Peketr, 2024-10-07:
@LennardPlay in the yaml custom_quirks_path should be under zha, not zigpy_config
[image]

LennardPlay, 2024-10-09:
> @LennardPlay in the yaml custom_quirks_path should be under zha, not zigpy_config [...]

Thank you very much. I wonder why it never gave me a warning that my config was broken all these years.

This quirk seems to be working for my device, but sadly it only shows temperature and humidity. (Maybe it just hasn't send a humidity update yet)

atus42, 2024-10-09:
> I've spent the last 2 days comparing how z2m and zha work. Now I've got an (as far as I can tell) working custom quirk for it, but I had to override a lot of stuff from parent classes. Anyways, if [...]

Unfortunatelly it only worked for 1 device out of three for me.
I think one key part is still missing from here.

mathieumd, 2024-10-12:
Thanks @Peketr! I confirm your [ts0601_sensor.py](https://github.com/Peketr/zha-device-handlers/blob/Peketr-patch-zht05/zhaquirks/tuya/ts0601_sensor.py) is working well for mine too.

Didn't even had to remove the device (after restarting to load the new quirk): just to reconfigure the device from ZHA.

Although brand new, battery is reporting only 50%. Will see if it'll got a decent life duration...

atus42, 2024-10-13:
> Although brand new, battery is reporting only 50%. Will see if it'll got a decent life duration...

My batteries were also reported to 50%, and one of the device that didin't reported temperature and humidity data fully drained the battery since then.

Peketr, 2024-11-06:
> As this quirk seems to work, @Peketr do you want to create a pull request? Then this fix could find its way into HA by default. Over in #2854 it also seems to work.

Well I was thinking about cleaning it up for a PR but 2024.11.0 changes how the ts0601_sensors are handled(YAY), and I have no idea how to hack it into that implementation.
For anyone looking I have updated [the file ](https://github.com/Peketr/zha-device-handlers/blob/Peketr-patch-zht05/zhaquirks/tuya/ts0601_sensor.py) so the quirk file works again, but I'm not sure I have the capacity to make it "mergeable"

WeterPeter, 2024-11-11:
> Well I was thinking about cleaning it up for a PR but 2024.11.0 changes how the ts0601_sensors are handled(YAY), and I have no idea how to hack it into that implementation. For anyone looking I have [...]

And this got them working again after my HA 11.0 update. 😀

Thank you! 👍🏻

delyriand, 2024-11-14:
Hi @Peketr, 

Your script for Tuya works for temperature sensor but not humidity sensor on my side.
Do you know why?

Anyway, thanks for the work!

Bhoft, 2024-11-21:
@Peketr the quirk works perfectly on my ZTH05.
All sensors can be read out (temperature, pressure and battery).

[24 more replies]

closed 2025-01-26 by TheJulianJES