zigpy/zha-device-handlers #2701: [Device Support Request] TY0201 - Temperature + Humdity Sensor, round. Humidity not shown

senna1992, 2023-11-02:
### Problem description

It works for temperature and battery reporting.
Humidity is not shown. 

### Solution description

A quirk supporting humidity, temp and battery!

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_zl1kmjqx",
  "model": "TY0201",
  "class": "zigpy.device.Device"
}```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.10.2.dev3001",
    "host_os": "Home Assistant OS 11.1",
    "docker_version": "24.0.6",
    "chassis": "embedded",
    "run_as_root": true
[... 549 lines cut ...]
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidityX10,
                    TuyaTemperatureHumidityAlarmCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

</details>

### Additional information

Source: https://de.aliexpress.com/item/1005005454107609.html

alexsoh, 2023-11-09:
The humidity sensor seems to be marked as unsupported. Hopefully this can be supported.

`       "0x0405": {
            "endpoint_attribute": "humidity",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 5097
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "measured_value"
              }
            }
          }`

nikospall, 2023-11-15:
Not experienced in Python so apologies for the naive question in advance. This seem to be a quirk written already for this device, does it means it works if I add it to the respective directory in my HAOS? (no idea if I have one yet but I'll dig into it)

alexsoh, 2023-11-15:
Yah, you can try but the quirks doesn't work for me unfortunately.

yuraant, 2023-11-22:
> I'll see if I can figure it out in the weekend, and update if anything positive comes out.

Hey nikospall,
Have you had a chance to take a look?  Any chance to get Humidity values in ZHA? 
thanks

alexsoh, 2023-11-25:
i gave up waiting and just move this sensor to z2m. Will start migrating my sensors over to z2m and keep ZHA as a secondary going forward.

batista16, 2024-01-02:
same issue here

waverunner1962, 2024-01-02:
Hi and Happy new year 2024.

Same issue here, work like a charm with Tuya gateway, no humidity with Zigbee

Tnanks

mpactdev, 2024-01-31:
I can't get it working on mine either with the quirk provided (though the quirk does show as applied), and I'm curious why anyone else can... the quirk seems to make the output template identical to the input. Let me be clear, this is not criticism but a question, because I don't know enough to understand why that would or wouldn't help.

mpactdev, 2024-01-31:
There appears to be a more lively (but still yet unresolved) discussion on this device over here:
https://github.com/zigpy/zha-device-handlers/issues/2862

PR #3201 (2024-06-26): Add Tuya TY0201 temperature/humidity sensor
This seems to be a pretty common sensor, for which several models seem to exist : _TZ3000_bjawzodf and _TZ3000_zl1kmjqx

## Proposed change

Add support for  another humidity/temperature sensor

## Additional information
First time contributing here. And I am not sure it is completely correct, but 
this quirk is confirmed to work for a wide number of people. I welcome any comments
to improve this.

Fixes #2862, fixes #2851, fixes #2701

 

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been [... cut]
closed 2024-06-26 by TheJulianJES